zs6buj/mav2pt #2: Telemetrie transfer stop after a few minutes

master-of-desaster-88, 2018-07-09:
Hello Eric,
my name is Markus and I am one of the ArduPilot neerds like to fly planes which a lot of telemetry :-)
At current I have a FrSky Horus X12S and a FrSky Taranis x9d+. For the telemetry I use the yaapy lua script from Alex. He was so kind to extend the script that I be able to display also external sensors from the S.Bus.

So far so nice.

Now I like to transfer Mavlink inforamtions to external sensors on the FrSky.
For example the rc in and out values or the air- and groundspeed from the Mission Planer HUD:
I try it with the MavLink_FrSkySPort from Keen (althertop). But there are to many sensors and and it work together with the yaapu. So I ask him how to switch the unneded off. He suggest your new release as a good solution.

I downloaded it in a Teensy 3.2 and select the Air_Mode.
Together with the yaapu script on the x9d+ it runs well a few minutes and then the telemetrie stop.
The receiver is running well, also the S.Bus for the external devices like rpm sensor or batt sensor.
After a restart same issue again.
Also I have a second question.
Is it possible to transfer the rc in, rc out and HUD informations to the external sensors?
Like Alex it did with a few of the sensors in his script.

Cheers, Markus

zs6buj, 2018-07-10:
Hey Markus

 I will try to help you with pleasure. From the above I understand two main issues:

1)  When you use the Teensy in Air Mode, together with additional sensors and extended yaapu script, it hangs after a few minutes.
2)  Is it possible to transfer the rc in, rc out and HUD information to the external sensors? By this I understand can we transfer the information to the "DIY" sensor ID 0x5000 range?

It will be difficult for me to test your circumstances, since I don't have your hardware setup. Are you in a position to compile and flash the firmware, and run in debug mode? That means enabling some of the debug #define statements, and observing displayed debug data on the screen.

As for  question 2), yes indeed the information is  available already from Mavlink, we just need to decide on which 0x5000 ids to use, and define some lean data frames. We should do this in conjunction with the Ardupilot community so that the work is not reinvented again by others.

Cheers
Eric

yaapu, 2018-07-10:
Hi,
there might be another option to send custom values without interfering with current 0x5000 packets. It should be possible to do as for GPS info and use different data frames.
Just an idea

tadam777, 2018-07-10:
Hello Eric and Markus,

Can I suggest to discuss this in the rcgroups ULRS thread ? You'll get more
feedback from users.

It's here :
https://www.rcgroups.com/forums/showthread.php?2037442-Ultimate-LRS-Hassle-Free-Full-Mavlink-16-Channels-Fly-More/page916

Thanks in advance,

Ben

zs6buj, 2018-07-10:
Hi Alex

I would say that a record type like RC in or RC out deserves an ID type of its own. It could get rather bandwidth hungry, with maybe 16 PWM values @ 11bits each (max 2048).

Hi Ben

There is merit in what you say, but when we get into detail it becomes rather off-topic in amongst all the ULRS discussions.

We do have a thread for the Teensy/Passthrough project. It's here: 
[](url)

https://www.rcgroups.com/forums/showthread.php?3089648-Mavlink-To-FrSky-Passthrough-Converter

Perhaps we could simply let the ULRS thread know, and then check back periodically.

yaapu, 2018-07-10:
Hi Eric, 
If the purpose is telemetry feedback than having full +-1024 is overkill, maybe +-100% 7bits plus sign would be enough. With 8 bits would be 4 channels per frame.
It would be hard to go faster than maybe 1 or 2Hz, I would even use 6 bits plus sign and scale 64 to 100, this would give some spare bits for each frame

yaapu, 2018-07-10:
Markus are you planning on using the teensy in air mode alone or were you thinking of connecting it in parallel to the frsky from the pixhawk?

zs6buj, 2018-07-10:
Ok, I see. You want to display them graphically.

yaapu, 2018-07-10:
Well the whole frsky diy protocol is more toward visual and vocal telemetry feedback than a real data transmission protocol, as you know most of the values are lossy encoded, this are just 2 cents

zs6buj, 2018-07-10:
Sounds good to me

zs6buj, 2018-07-10:
I've coded MavToPassthroughPlus for you to test, but run out of time to test it myself first. I'll post it tomorrow. Basically 0x5009, 32bit frames as usual, first 4 bits marked with 0xF, then channel count in next 4 bits. Then 3 x int8_t = pwm %

Then multiple 4 x int8_t = pwm%, with empty ones = 0x00

Ok, just posted it on my GitHub, but can't guarantee anything :)

master-of-desaster-88, 2018-07-10:
Hello Eric,
Hello Alex,
this days I am on a bussines trip so I could not answer closer.
@Eric, from my technical background I am a electric engieer. The hardware is closer to me as the software.
But I have a basic understanding of C and Assembler. Also I could compile the files with visual basic.
Compile and flash is no problem. For the debug a little bit help would be needed.

@Alex, at current I use the teensy in airmode alone. I am not sure if it would work in a parallel mode with the frsky passtrough protocl. A teensy airmode alone would have the advantage to spare the converter from the pixhawk to the frsky receiver.
I like to feed the swap screen (Horus X12S) from your telemetrie script with aditional sensors like groundspeed, airspeed, throttle or rc outputs. 
This would help me to see for example how much throttle is needed to keep the plane in a balance position and how well is the trimming.
Of course I could see that on the log, but in flight it's faster to do the adjustment.
Cheers, Markus

master-of-desaster-88, 2018-07-12:
Hello Eric,
I have load the version plusv3 to the Teensy.
At the telemetrie page from the Horus and/or Taranis fist only the GPS position would be shown.
After I start the lua script from Alex the normal sensors will be add on.
So I think I need a new / extended lua from Alex for the additional senors.
Am I right so far?
Would it possible to show the additional senors direkt - like the GPS position?
Also I have observed that with the v03 version no stop in the transmission happen.
The telemetrie keep stable over 15 minutes on the Horus and Taranis.
For my better understanding:
- MavlinkToPassthru.ino convert the Mavlink stream to segments.
-FrSkyPortPassthru.ino convert the segmenst to a FrSky S.Bus stream.
AmI right?

Cheers,
Markus

yaapu, 2018-07-12:
Markus on x9d to show GPS add a telemetry screen of type values from the model telemetry setup page, choose GPS as source and also add your extra sensors. 
Then when the yaapu script is running press page to toggle between yaapu display and extra screen.

On Horus GPS is shown on the main screen but you have to discover it first.

Eric will answer your other questions :)

zs6buj, 2018-07-12:
Hi Markus, yes you are correct. MavlinkToPassthru.ino  reads and decodes mavlink telemetry,  and the other one encodes and sends S.Port telemetry.

master-of-desaster-88, 2018-07-12:
Hi Eric,
I Alex,
Hopefully not working to hard on your patience :-)
How to add sensors is clear for me. Problem is that the sensors are not show at the OpenTX page.
So I could not select them.
[image]
That's the reason why I have asking if there is a need to edit the lua script from Alex.
Because if the lua script is not running only the GPS is vissible.

Cheers,
Markus

master-of-desaster-88, 2018-07-12:
One additional information.
On the Taranis x9d I have to start once the lua script then the sensors in the picture above get copy to the TX telemetrie.

yaapu, 2018-07-12:
Markus you are right, you need a custom version of the Lua script, it's not ready yet.

master-of-desaster-88, 2018-07-12:
Alex thanks for the short info.
So I could take a beer instead of coffee :-)

yaapu, 2018-07-12:
Ya Markus, aber eine ist keine, zwei ist besser :)

master-of-desaster-88, 2018-07-13:
Wie immer - das Erste und das Letzte - zwei Bier :-)

master-of-desaster-88, 2018-07-16:
Hello Eric,
after some testing - and some beer :-)
I think it would be helpfull to aktivate the LED from the Teensy if the Link have startet.
Like it is done on the MavLink_FrSPort-2.01 programm.
Light keeps people quite and save :-)
Do you think its possible?
Cheers, Markus

zs6buj, 2018-07-17:
Hey Markus

Lol   :).   Yes I will certainly look at that and let you know.

zs6buj, 2018-07-17:
Done. On plus v0.3 dated today.

zs6buj, 2018-07-19:
Good to see: )

closed 2019-10-07 by zs6buj