zs6buj/mav2pt #3: Teensy LC

simjman, 2018-07-14:
Does this support Teensy-LC? If not, is it possible to make it supported?

zs6buj, 2018-07-14:
Hi simian

I just compared them and the LC looks pretty close to the 3.2 spec. The lack of FIFO buffers on the serial ports is different, and it is likely to be slightly slower, but that should not be a problem. The pins we use also appear to be identical. If you have an LC, then try it, but I cannot guarantee it will work. I don't have one and can't test it here right now, but might just buy one to try..

simjman, 2018-07-18:
That sounds great. I have already heavily invested in the Teensy-LC on my airframes with Athertop's Mavlink-FrSky code - and this was working well. However I have spent the last few days trying with your code on the same setup and Yapuu's telemetry script with no luck. When I change the teensy code from Athertop's to yours, the sensor discovery returns nothing (and yes I am deleting and restarting the discovery). What should the SERIAL#_PASSTHROUGH be set to for your code? I have tried 1 & 10 with no luck - and yes I rebooted between changes. This particular airframe is with a pixhawk 1 & X8R. I am also using a Taranis 9XD with Opentx 2.2.

I have also set the parameters in your code to Air_Mode & Target_Teensy3x.

The wiring is the same as your airside diagram, except I use +5v from the Rx s.port to Teensy Vin.

I have not tried your most recent change with the LED feedback, but will try it tonight. 

Btw, the link to the "Ground side with Orange 1W and HC-06 Bluetooth" image is broken.

Appreciate your help

zs6buj, 2018-07-18:
Hi simjman

SERIAL#_PASSTHROUGH

You mean inside MP. It must be set to 10, and the telemetry speed to 57, obviously for the Pixhawk port you are using.

Then make sure in the source code this line looks like this "#define Use_Serial1_For_SPort   // The default, else use Serial3", then S.Port is on serial1, which means pin 1, which is the same on the LC according to Google. :)

Thanks, I'll check the link.

Not sure what

simjman, 2018-07-18:
Hi Eric,

Thanks for the help.

I've tried the new code with passthrough set to 10 and have a continuously blinking LED.

The source code is as you have suggested.

Does this link change your understanding of how the LC works? https://www.pjrc.com/teensy/td_uart.html

zs6buj, 2018-07-19:
Hey simjman

Have you made any progress?

simjman, 2018-07-23:
Hi Eric,

Sorry about the delay. Been busy. After your previous comment I finally twigged that the teensy wasn't a replacement for the FUL-1 and passthrough, which is what I was hoping for so I didn't have to do any hardware mods. However I just so happened to have a few FUL-1's sitting around, so I wired it up and got the passthrough working as per the wiki. Now I understand what these different systems are for, whereas previously it wasn't 100% clear. Either way I never did get the teensy LC working. I do have another airframe still running an apm 2.5 so I might be back here trying to get it working soon anyway. Appreciate your help.

choyado, 2019-01-02:
I have a Teency LC and am willing to do some testing. However, I am struggling to get the code to compile right now. Is there a list of libraries that are required to compile?

zs6buj, 2019-01-03:
Hi choyand

Be sure to clone the libraries folder together with the appropriate version source. Then copy/merge the library folder into the default sketchbook location. The usual library path rules apply.

[image]

This is the default library folder on my machine. You are likely to have many more libraries than this, as this is a fresh installation.

[image]

If you want to use Mavlink version  2 (recommended), then the source is set up to use the c_library-v2 library, otherwise for Mavlink 1 change back to the GCS_Mavlink folder.

zs6buj, 2019-01-03:
Oh, and I have not tested the Teensy LC yet.

zs6buj, 2019-01-03:
You can safely ignore the warning. It comes complements of the Mavlink 2 library.
Upload the sketch to the LC and immediately switch to the serial monitor. You should see 'Starting...' and some other information.  If you hook up good Mavlink telemetry the LED (not the power led) should come on.

Thanks for testing the LC for potential users.

closed 2019-10-07 by zs6buj