zs6buj/mav2pt #5: Not sending all the data to FrSky Rx

LTMNO, 2018-07-18:
Hello, Great solution... and I know the solution works as I had tried it a few days ago with the help of Yaapu and then today, I finally got around to doing two test flights and no data was coming thru... other than VFAS, Fuel @99% and RSSI and RxBt.

Here is the Radio Log...

Thoughts?

just remove the .xlsx from the file below to load.
[DART450-2018-07-18.csv.xlsx](https://github.com/zs6buj/MavlinkToPassthru/files/2207243/DART450-2018-07-18.csv.xlsx)

zs6buj, 2018-07-19:
Hey Pino

I use the Mavlink 1 libraries. Mavlink 2 adds nothing the this solution.

Oh, and I have "fastest" ticked in the Arduino IDE.

From the log it seems that no Frsky telemetry is reaching the Taranis input pin, so it is likely not reaching the receiver in the air. If remember correctly you are using Air Mode together and a Pixhawk. Could you expand on the detail a bit? Pixhawk running APM PX4 firmware, or something else? I assume you know if you are running APM 3.5 or later, you can get S.Port Passthru telemetry straight out of the Pixhawk without this converter.

Can you do a desk check with the solution again using some of the debug features ?

LTMNO, 2018-07-19:
Hello Eric. 
Thx for the response.  
Running 3.2.1 APM 2.6
Have to use Teensy.  
I will use debug when I get back home late tonight. Don’t know why I didn’t just do that and send to ya. 😁
I originally did so and everything was working.  
I will report back. Thx.

LTMNO, 2018-07-20:
Hello Eric, I have found what the problem is...
I am not sure how the solution is setup or where the code is for the polling but the APM doesn't send the Telem Data automatically from what I understand until requested.  Once and a while I get data sending to the Converter by chance and most times it does not.
I was monitoring the Debug Console via USB to Teensy and noticed that when I turn the system on and that is including the Quad(power by Batt) --> APM 2.6--> Teensy --> FrSky Rx (X8R) --> Air --> QX7S . I would get no data sent consistently.  But when I introduce the Telemetry Radio and establish Link to other Telemetry Radio, the data is flowing and the yaapu/zs5buj solution is working nicely.

What is the trick to have the Teensy poll or pull data from the APM to Initiate Data Stream?

Many thanks...

Pino.

zs6buj, 2018-07-21:
Hi Pino

You must setup the SRx fields in Mission Planner, where x is the serial port you are using. This tells your APM flight controller to send out the appropriate telemetry types regularly.

https://imgur.com/0HIKh8g
[](url)

http://ardupilot.org/copter/docs/common-telemetry-port-setup-for-apm-px4-and-pixhawk.html
[](url)

If all else fails you can un-comment this line in the source, which will cause the Teensy to request the data streams every so often. Should you go this route, you must wire the Teensy tx pin to the APM rx pin (this connection is not needed otherwise).

//#define Data_Streams_Enabled // Rather set SRn in Mission Planner

Let me know how it goes. :)
Eric

LTMNO, 2018-07-21:
Good day Eric, 
I have implemented the SR1 changes and happy to report that data is flowing.
I am going to take it out for a test flight and record the data log.
At a quick glance, it all seems to be coming down the pipe. ;-)
Just to note, I have the Tx/Rx/5v/Gnd from the Telemetry Port connected to my Teensy and then have the Teensy Signal Wire and Ground connected to the X8R. (see below)

X8R SPort S <–- TX1 Pin 1 
APM Tx (Mavlink) --> RX2 Pin 9 
APM Rx (Mavlink) <-- TX2 Pin 10 
APM Vcc 5V (not 3.3V) <--> Vin
APM GND <--> Gnd

LTMNO, 2018-07-21:
Well, clearly i am getting the data.  The Fuel, CURR, GAlt, seem to not be working... i will start to test them with debug logging.
The battery voltage is 11.6V Full Charge was 12.6.
The Altimeter was acting up... very windy outside this morning here in Toronto Area.  Alt-Hold wasn't working well until later in the flight. ;-)
Either way, making progress here. ;-)

[DART450-2018-07-21.csv.xlsx](https://github.com/zs6buj/MavlinkToPassthru/files/2216174/DART450-2018-07-21.csv.xlsx)

zs6buj, 2018-07-21:
In yaapu`s LAU app you need to go into the menu, space down to voltage source and select the FC or your sensor

LTMNO, 2018-07-21:
Hi Eric, that option is enabled... ;-)

zs6buj, 2018-07-21:
Did you click on it and select the correct source?  Yry the available sources to fix your low battery reading.

LTMNO, 2018-07-21:
Hello Again, the only source that gives data is FC.  I connected my FrSky Voltage Sensor that connects to Balance Charger, did the discovery and then it added another sensor reading called CELLS.  Now I see the correct voltage at 11.6v and 3.88v/cell under the VS option.
[image]

What doesn't change is the GAlt, Hdop and the Fuel @99% or the mAh Consumed.

The rest seems to be working...

[DART450-2018-07-21.csv.xlsx](https://github.com/zs6buj/MavlinkToPassthru/files/2216550/DART450-2018-07-21.csv.xlsx)

master-of-desaster-88, 2018-07-22:
Hi togehter,
I have observed one issue wiht could be helpfull.
I also testing the long range R9 receifer from FrSky with the FCC 180329 firmware.
After a minute the mavlink will be get interupted and no trafic between receifer and Pixhawk is possible.
The sensor values from the hard wired sensors will be still working.
This never happen with the X8R receifer on 2,4 GHz.
Maybe with a new firmware from FrSky this issue will be solved.
Cheers,
Markus

LTMNO, 2018-07-22:
Hello Markus, I will check if there is a new firmware for my X8R Rx.
Thanks for the post...

master-of-desaster-88, 2018-07-22:
With the X8R RX everthing is fine.
The interrupt happen with the R9 RX and use of the Mavlink protokoll.
If I use the FrSky S.Bus pass trough also everything working well with the R9 RX.
But then is no Teensy involved.

LTMNO, 2018-07-22:
Ah, sorry for the misunderstanding.  There seems to be one update for Horus feature on the X8R.. not relevant for me. ;-)

I think there are some APM SR1 Settings that need to be tweaked... 

Here are some logs of testing with Debug Mode Enabled...

#define Use_Local_Battery_mAh @5400
[BatteryLogc.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2217134/BatteryLogc.txt)

//#define Use_Local_Battery_mAh 
[Batt_Commented_Out.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2217129/Batt_Commented_Out.txt)

Strangely, GPS showing Lock but nothing else
[GPSLog.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2217136/GPSLog.txt)

Detailed GPS Data
[GPS_Detail_Log.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2217137/GPS_Detail_Log.txt)

zs6buj, 2018-07-23:
Good morning Gentlemen

I see you had a busy weekend.  

R9:

I'm afraid I know nothing about the R9 RX and will need to do some reading. It may or may not pass the 0x5000 range of telemetry. What you can do is un-comment  //#define Frs_Debug_All and let's look at how the Teensy is interacting with the R9 firmware.

Battery - volts, amps and mAh :  

From the log you posted it is clear that the output routine for 0x5003 Battery 1 is not receiving any data. This data comes from Mavlink message # 1  SYS_STATUS. So our next step is un-comment   //#define Debug_Batteries. I expect you might then confirm that the record is arriving from your APM with zero in those fields. If that is so, then we need to determine why APM is not configured to send them. Have you 
set up your power module properly?  http://ardupilot.org/copter/docs/common-power-module-configuration-in-mission-planner.html[](url) If you run Mission Planner live with a usb connection to your APM, can you see the voltage and current, and are they correct?

GPS Status: 

Similar to above, we appear to not be receiving GPS data from APM. You can check if it is coming in by un-commenting these two lines:  

//#define Mav_Debug_GPS_Raw     // #24
//#define Mav_Debug_GPS_Int     // #33

Heartbeat:  ID #0

The heartbeat is very important for several reasons, one of which is it tells us what type of craft (or GCS or Tracker etc) is sending the mavlink telemetry. Flight modes and other things are determined according to ap_type. Please could you un-comment //#define Mav_Debug_Heartbeat

Regards
Eric

LTMNO, 2018-07-23:
Good morning. I agree with all your assessments. I will test asap. 
Un/fortunately family life gets in the way of the hobbies.  Off for a 7 day vacation. Back following Monday.  Will test then. 😁
Thank you all. Enjoy your week.

LTMNO, 2018-07-31:
Hello Eric, back again... ;-)
Well, I plugged everything in and started the logs.... but I have also decided to try to connect my Telemetry Radios (900Mhz) system and see what happens.
I have deducted that when these radios are connected, that all information is pushed thru to the system from Quad to Radio.
When this is not connected, I miss some information.

So I think I need to work on the SR1 settings... Thoughts?
I think your code and yapuu is working fine...even for my setup.

When connected you can see it works here...
Radio Screen
[image]
DroidPlanner Screen
[image]
Can also see that I switched the HDOP to 3D Lock Display
[image]

Also, the FC is Off compared to the VS...
[image]
[image]
[image]
[image]

LTMNO, 2018-08-01:
p.s. #define Data_Streams_Enabled   // Rather set SRn in Mission Planner

Won't Compile...?

LTMNO, 2018-08-01:
[Debug_Heartbeat.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2247507/Debug_Heartbeat.txt)
[Debug_Batteries.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2247510/Debug_Batteries.txt)
[Debug_GPS.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2247509/Debug_GPS.txt)
[Debug_GPS_Cont'd.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2247508/Debug_GPS_Cont.d.txt)

zs6buj, 2018-08-01:
Hi LTMNO

The Srx settings work for everyone else, so I don't think you need to #define Data_Streams_Enabled. However, I will make sure it compiles in future in case.

zs6buj, 2018-08-01:
Hi again Markus

If you want to patch , could you make this change: Note missing comma!

WAS:
const uint16_t mavRates[] = { 0x04, 0x0a, 0x04, 0x0a, 0x04, 0x04 0x04};

MUST BE:

const uint16_t mavRates[] = { 0x04, 0x0a, 0x04, 0x0a, 0x04, 0x04, 0x04};

I added in MAV_DATA_STREAM_EXTRA3 but missed the comma. :(

I'm working with Alex on v0.05 of Plus, so I'll patch that one also

master-of-desaster-88, 2018-08-01:
Hello Eric,
Hot weeks in Germany - feeling like South Africa.
Kind of body barbeque.

Alex send me the new version from his program witch is now working as a widget.
Very convinient and also easy to enter the other sensors in additional widgets.

He also wrote that he will add the transport from the senors to the Horus telemtrie.
But he need also a modfied version from You.
I think v0.05 Plus will be the solution.
So I have to wait until the new script from Alex is finish before testing.

Cheers,
Markus

zs6buj, 2018-08-01:
Oops. Hi Markus. It's 23 C here today. Mid winter :) 

I meant to address the previous post to LTMNO.

Sorry guys.

LTMNO, 2018-08-01:
Hello... Thanks for the reply... I will continue to play with the SRx values until I get a favourable result.
I wonder if I can trap packets that are being sent when I connect the Telem Radios... this could perhaps tell us what we need in order to get the full messaging?  As I mentioned before, that seems to be the only way to get all the messaging... but it would defeat the purpose to have my tablet and my radio receiving the same information. ;-)

Thanks again, 

Pino (LTMNO)

zs6buj, 2018-08-01:
Hi Pino

I patched Plus v0.04   Try   #define Data_Streams_Enabled   again

master-of-desaster-88, 2018-08-01:
Hi Eric,
I have tested the new version Plus_v04 - Plus_v06 on my Teensy 3.2 board.
I set the variables to select the Teensy board and for airmode.
The "mavlinkgood" LED is constant, but only the hard sensors like RSSI, Cels or RxBt get copy to the telemetrie on the Horus. No GPS for the start of the YAAPU telemetrie.
If I use Your Plus_v03 version all working well.
Some smal bug?

Cheers,
Markus

zs6buj, 2018-08-02:
BHi Marcus

Thanks for the 'heads up'.  Plus is very much a work-in-progress I'm afraid.  I re-jigged the options from  v0.04 to support the Maple Mini. It's a lovely little STM32 board, affordable, boot-loader works properly through USB, and it has 3 additional serial ports. Anyway, please carefully check the options you have un-commented. Don't use auxilliary (bluetooth) option for now as I'm working on it. If you could check v04 again carefully I would appreciate the help.

Thanks and regards
Eric

master-of-desaster-88, 2018-08-02:
Hi Eric,
Yes the Maple Mini is nice :-)
I have check the version v04 & v05 release from today - carefully.
But no change.
Still same behavior.
Seems the mavlink get running but no valid information on the FrSky side.
I am not used to debug modes. But if you have a small how to I could try.
For the upload to the Teensy 3.2 I use the Visual Studio 2017 with the vMicro Arduino.

Cheers,
Markus

LTMNO, 2018-08-02:
Hello Eric, I tried to use the : "Plus v0.04 Try #define Data_Streams_Enabled again" version, it compiled for me but I was getting no data TELEMETRY on the radio side.. I am not sure what is the diff between the Plus and the v1.01 as I have been using the v1.0.1 this whole time.
I tried to load Plus_v05 that would install but wouldn't compile for me in Arduino.

[image]

I have put the v1.0.1 back on and I am back to where we started.. i didn't have time today to test different Mhz rates in APM for SRx.  I will tomorrow.

99, 2018-08-02:
Hey @LTMNO, can you please remove "@ 99" from your initial message? Tired receiving that many emails for the tread I don't really care. Thx. 
Please note that @ + username in the GitHub will mention users and automatically will subscribe them to all the related messages.

yaapu, 2018-08-02:
I found the bug, in ver 0.5 line 286 of FrskySPortPassthru.ino

```
void FrSkySPort_SendDataFrame(uint8_t Instance, uint16_t Id, uint32_t value) {

 #if defined Target_Teensy3x
  setSPortMode(TX); 
 #endif
```

should be

```
void FrSkySPort_SendDataFrame(uint8_t Instance, uint16_t Id, uint32_t value) {

  #if (Target_Board == 0)
  setSPortMode(TX); 
  #endif
```

it works but messages are often corrupted, did you change the message processing logic?

LTMNO, 2018-08-02:
Hello All, I would like to share that I have added the change stated above, 

void FrSkySPort_SendDataFrame(uint8_t Instance, uint16_t Id, uint32_t value) {

  #if (Target_Board == 0)
  setSPortMode(TX); 
  #endif

I have also added the ',' to the following line which was missing.

const uint16_t mavRates[] = { 0x04, 0x0a, 0x04, 0x0a, 0x04, 0x04, 0x04 };

I compiled with "#define Data_Streams_Enabled // Rather set SRn in Mission Planner" enabled.

And now I am getting all the information including the HDOP, etc... which i wasn't before, right from the start.  I will have to take it up flying to see if the Mah Calculations are working, etc to show battery usage and that should be it.

Great work guys...  I will report back after a flight.

Cheers!

master-of-desaster-88, 2018-08-03:
Hi togehter,
I have changed the script according to the points from Alex.
Now it works.
The messages looking also fine on my Horus12.
Cheers,
Markus

zs6buj, 2018-08-03:
Gut gemacht Markus!

Great news Pino! 

Thanks again to Alex for finding the bug. I'll patch Plus to v0.06 this morning

LTMNO, 2018-08-21:
Hello Eric,
It seems that everything is working but the Ctot property is missing and Fuel is not updating... therefore I don't get any updates on my Mah Consumption from the 5400Mah battery used.

see https://github.com/yaapu/FrskyTelemetryScript/issues/7#issuecomment-414499415

I am getting the Curr, Volts and CurrTot in the mission planner from the APM.

Thoughts?

zs6buj, 2018-08-21:
Hello Markus

Yes total mAh is sent back to the Taranis/Horus. I suggest that we trace the data through the system. You could un-comment in the Teensy

`//#define Debug_Batteries`

and monitor the output. Monitor for a while to see the mAh increasing over time. Once we know it is transmitted, we can look further.

Regards

LTMNO, 2018-08-21:
[Debug_Batteries.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2307230/Debug_Batteries.txt)
Here is the debug log with Debug_Batteries enabled.

zs6buj, 2018-08-21:
Ok Markus, the record type BATTERY_STATUS ( #147 ) is not coming in from Mavlink. I will post a quick alternative for you, where I accumulate mAh by addition of di/dt. i.e. sum of all the instantaneous current measurements over time.

I'll look more closely tomorrow,. In the meantime please try v1.0.3b. Be sure to set all the #defines as per your requirements.

LTMNO, 2018-08-21:
Hey Thank you Eric... I will test now...

Pino. ;-)

LTMNO, 2018-08-21:
Okay, had to fix this in the code to compile... was missing a comma
```
//const uint16_t mavRates[] = { 0x02, 0x05, 0x02, 0x05, 0x02, 0x02};
const uint16_t mavRates[] = { 0x04, 0x0a, 0x04, 0x0a, 0x04, 0x04, 0x04};
 // req_message_rate The requested interval between two messages of this type
```

here is the output...
[Debug_Batteries_v1.0.3b.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2307596/Debug_Batteries_v1.0.3b.txt)

Here is the screenshot of the Radio... Ah is showing up now and Fuel has changed...
[image]

This great... the question now is it accurate?

master-of-desaster-88, 2018-08-21:
Hello Eric,
was not my issue :-)
But nice that you are thinking on me.
Cheers,
Markus

LTMNO, 2018-08-21:
Yup, was me Pino aka LTMNO... I just didn't want to correct you... all good! ;-)

zs6buj, 2018-08-21:
Oops, very sorry Pino

zs6buj, 2018-08-21:
Pino, I would appreciate if you could give me feedback on its accuracy. The theory is sound, but when you add up an infinite number of small measurements, I'm not sure. We could tweak it if necessary. I'll also check up tomorrow to see why Mavlink did not pass down the APM measurement.

LTMNO, 2018-08-21:
All good.  😁
I will use my amp meter and measure tomorrow am. Thank you.  
Pino

zs6buj, 2018-08-23:
It seems there was a short burst of activity on Battery Status #147, but then trickled out and I can't see where to activate it.

https://github.com/ArduPilot/ardupilot/issues/3133

LTMNO, 2018-08-23:
> ArduPilot/ardupilot#3133 <https://github.com/ArduPilot/ardupilot/issues/3133>

Interesting…I am away on business… unfortunately, I can't test till tomorrow afternoon upon my return… sorry for the delay.

LTMNO, 2018-08-25:
Alright.. here are the test results of this morning... 
Full Charge Battery at 12.6V @ 5400mAh
[image]
[image]

Then I ran the motors with inverted props to draw some amps.  These are the readings...
Actual draw in AMPS is 5.37A vs Script reading at 4.8A and Volts a 12.3V vs 12.0V respectively...
[image]
[image]

Then another reading....
1.58A/12.34V  vs  0.8A/12.2V . (Granted the sensors don't work well under 1A)
Please note that we are at 2360mAh Consumed at this point at @ 56% Fuel (5min in)
[image]
[image]

Then another... 
2610mAh Consumed at 51% Fuel
[image]
[image]

Then the last one.. at 3000mAh @ 44% Fuel
[image]
[image]

The Actuals from the Battery via recharging and reading the graph is...
419mAh Consumed
[image]

[image]
[image]
[image]

In conclusion the mAh Consumed calculation is not accurate. ;-)

I hope this testing information helps.

yaapu, 2018-08-25:
Hi Pino ideally  to  get all data we need the current reading from mission
planner in same setup to determine if the power module is well calibrated
or not

zs6buj, 2018-08-26:
Hi Pino

Thank you for all this information. It helps fill in the picture, but the critical numbers for me are the average current during the 7 odd minutes that you did the test. If the average current was 3.591A over 7 minutes, then the mAh over that period = 3.591 * 7 / 60 *1000 =  419 mAh, which correlates with the energy you had to put back. So the 3.59A looks like it could be roughly correct, and the approximately 3000mAh on the Taranis can't be correct, especially after only a few minutes at 5A.

Let me go back and do some more work. I want to look harder at how to get the APM mAh out of the FC remotely, second I'll do some current consumption tests of my own and see if I can calibrate the calculation with a correction-factor. Third, I want to look at the Mission Planner code.

As Alex points out, everything is based on the current reading determined by the Pixhawk, in conjunction with the chosen power module, and its calibration.

LTMNO, 2018-08-26:
Hello Eric/Alex...
I have configured/calibrated the APM 2.6 Power Module on the APM as well.
Last night around 11pm (wife wasn't too happy) decided to run the quad indoors with inverted props and run the motors at about 12-15Amps.  As I was doing that, i was getting my output reading from the Amp Meter and then entering the parameters to the APM Battery Advanced Configuration screen.  The higher the Amps  Reading for Calibrations, the better Calibrated it is... so i have read.
That beings said, i did that last night. So this morning I am going to charge the batts.. do a simple draw again and see how calibrated it is. On that APM side.. excluding the PassThru Code.  

Stay tuned...

Thanks again for your feedback and help.

Pino

zs6buj, 2018-08-26:
So please un-comment this line if you are not getting the #147 record type through

`  //fr_bat1_mAh = Total_mAh1();  // If record type #147 is not sent by FC`

in this function

`void Send_Bat1_5003() `

I'll make this automatic in future

LTMNO, 2018-08-27:
Hello Eric, 
I put in the update...
I changed to : 
```
APM 
Teensy
Air_Mode
bat1_Capacity = 5400
```

I then ran and didn't see message #147 `fr_bat1_mAh=0`
```
> Frsky out Bat1 0x5003:  fr_bat1_volts=124 fr_bat1_amps=1 fr_bat1_mAh=0
> Frsky out Params 0x5007:  fr_param_id=4 fr_bat1_capacity=5400
```
then I un-commented `fr_bat1_mAh = Total_mAh1();`
and it started to come down the wire.

```
> Frsky out Bat1 0x5003:  fr_bat1_volts=123 fr_bat1_amps=1 fr_bat1_mAh=73
> Frsky out Params 0x5007:  fr_param_id=4 fr_bat1_capacity=5400
```
Question?
would it be possible to increase the fr_bat1_volts to 2 decimal points?
And for fr_bat1_mAh to increase to 2 digits?
I know that in the Debug that there are 1 decimal point and 2 decimal points recorded respectively.
I know that in the Yaapu display that only 1 decimal digits come thru on the qx7 display on CURR and 2 for VFAS.

I have done some serious attempts to calibrate my Power Module in APM... and it seems that based off of the Turnigy Amp Meter that I am off by 0.40 amps from the reading being sent down by FrSky.

Perhaps putting a parameter in place to increase the value might be an easy fix?
I know MikeB does that for er9x.

LTMNO, 2018-08-27:
Hello Again, I made a change in the debug to output this with the Battery Output.
```
Frsky out Bat1 0x5003:  fr_bat1_volts=121 fr_bat1_amps=0 fr_bat1_mAh=41
Mavlink in #1 Sys_Status:  Bat volts=12.192  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
Mavlink in #1 Sys_Status:  Bat volts=12.202  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
Mavlink in #1 Sys_Status:  Bat volts=12.202  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
Mavlink in #1 Sys_Status:  Bat volts=12.200  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
Mavlink in #1 Sys_Status:  Bat volts=12.197  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
Mavlink in #1 Sys_Status:  Bat volts=12.205  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
Mavlink in #1 Sys_Status:  Bat volts=12.191  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
Mavlink in #1 Sys_Status:  Bat volts=12.192  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
Mavlink in #1 Sys_Status:  Bat volts=12.178  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
Mavlink in #1 Sys_Status:  Bat volts=12.188  Bat amps=0.0  mAh=0.000000  Total mAh=41.351  Bat1 cell count= 3
```
With this I was able to record the starting and finishing point of the Amp meter output from a start and stop of consumption.
For a period of time like 30 seconds... 
I recorded a delta of 7.0 mAh
29 mAh consumed o the Amp Meter vs 22 mAh Consumed on the Mavlink Status # 1 log.

now this is a short period of time... i will do a longer test tomorrow AM...

zs6buj, 2018-08-27:
Good sleuthing Pino! Yes you should try to emulate real-life conditions, higher currents and longer times. I have not built in a conversion factor, just increased float to double for the timing calc. So the method is still "pure", if you know what I mean? For you next FC consider getting a Pixhawk mini. They are now around the price of an APM 2.6 board, but support the latest APM firmware. The one catch is that the firmware (and Mission Planner) is modified by RadioLink (the vendor) first.

LTMNO, 2018-08-27:
[DART450-2018-08-27.xlsx](https://github.com/zs6buj/MavlinkToPassthru/files/2325902/DART450-2018-08-27.xlsx)

Hello Eric, after extensive testing and calibration at higher amps... ~10A-14A 
I believe that I have the CURR set correctly. I completed a test and the battery recorded at 76% remaining on the Amp Meter... the APM recorded 77%.  
With the 1 Decimal point of margin... I believe this to be good! ;-)

77% is 1242mAh used @ 5400 Batt
76% is 1296mAh used @ 5400 Batt

I charged the Batt and put back in...1320mAh

I have also noticed that in APM config Param there is an Offset for Amps... I didn't use it, but its there ;-)

p.s I have a PixHawk Full Size board that I could put in this quad.. but that is for another build. ;-)

zs6buj, 2018-08-28:
[image]

Your log reflects the numbers you mentioned above. That's as close as you need Pino! Is the same number appearing on your Taranis now?

LTMNO, 2018-08-28:
Morning Eric.  Yes sir. We are in business 😜👍
I believe to be whole now on the radio front.  
Many thx!

closed 2018-08-28 by LTMNO