zs6buj/mav2pt #7: Frsky Horus X10 with Crossfire

heliprop, 2018-08-02:
Hi,
I have a Horus X10 with a TBS Crossfire TX and want to run the amazing yaapu script. So I tried to connect the teensy as in your description of your orange modul and configured the crossfire to mavlink transmission.  Inside of the crossfire I found a TX and a RX pin, but with no luck at all - no sensors found. Has anybody tried to connect a TBS crossfire  to a Taranis or Horus and have a tip for me, please?
I have no idea, how I can find the problem.

master-of-desaster-88, 2018-08-03:
Hi,
I think the problem is that the script from Alex is using the FrSky S.Port Passthrough.
The crossfire is not supporting that protocoll. So you will not see the sensors.
The crossfire could only transport RC and Mavlink.
That was the  reason why I switch to FrSky.
FrSky also have a long range transmitter and receiver.
But if I use the long range version with the Teensy I got interrupts. Seems that is a problem from the long range receiver. So still waiting for a update and use the standard.
Or You use the cable from Craft&Theory for the FrSky Passtrough. This is working stable with both receivers. 
Cheers, Markus

heliprop, 2018-08-03:
Sorry, now I'm confused.
My plan is, to send mavlink from the crossfire micro receiver to the transmitter on the back of the horus and use between the transmitter and the Horus the teensy to convert mavlink to Frsky passthrough. 
Is this not possible?
As I understood, is this way the idea of this MavlinkToPassthrough script.

zs6buj, 2018-08-03:
Hi Jürgen and Markus

Yes, if you can get Mavlink to the back of the Horus, then the Teensy will convert the telemetry to Frsky Passthrough. Make sure that you are using Mavlink 1 (not 2), that your tx/rx pins cross over (rx to tx, tx to rx) from the Crossfire to the Teensy, and that the serial port of the Crossfire is set to 57600 b/s or the Horus side.

closed 2019-10-07 by zs6buj