zs6buj/mav2pt #8: Always "Stabilised Flightmode"

Seeelefant, 2018-08-17:
Hi Eric,
so far everything is working fine. The only problem I currently have, is that I always get "Stabilised Flightmode" independent from the Mode I choose.

[image]

zs6buj, 2018-08-17:
Hi Seeelephant

Could you remind me what your setup is?  Which version of the converter are you using?

Please un-comment Mav_Debug_Heartbeat and let's see what is coming from Mavlink?

zs6buj, 2018-08-17:
#define Mav_Debug_Heartbeat        :)

Seeelefant, 2018-08-17:
Hi Eric,
I have version 1.01 running on a Teensy 3.2  in Air Mode. As debug messages I get 
[image]

Seeelefant, 2018-08-17:
Re-Open :-)

Seeelefant, 2018-08-17:
Re-Open :-)

Seeelefant, 2018-08-18:
Hi Eric,
sure: 
[LOG.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2299242/LOG.txt)

Seeelefant, 2018-08-18:
Hey Eric,
after looking at it and seeing that only ap_custom_mode changed, the problems seems to be related to https://github.com/athertop/MavLink_FrSkySPort/issues/27.
Thanks
--Seeelefant

zs6buj, 2018-08-19:
Hi Seeelephant

Yes, flight mode is derived from ap_custom_mode on the mavlink heartbeat record, and also depends on the type of frame - ap-type. Obviously a quadcopter does not have the same flight modes as a fixed wing aircraft, an so on.

The LUA script determines the correct flight mode name and enunciation.

If there are other "sensors" on the bus, the Taranis and the LUA script need to decide which one to use for flight-mode. Alex (yaapu) can probably help with that.

Good luck
Eric

zs6buj, 2018-08-19:
Hi Seeelephant

Yes, Alex did talk to me. I'm working on it. I'll let you know.

Seeelefant, 2018-08-19:
Thanks Eric,

it´s highly appreciated :-) ....Thinking beyond it: It would be very nice to have all information in one place, sort of QGroundcontrol integrated within the sender....or..?

Best 
--Seelefant

Seeelefant, 2018-10-12:
Hi Eric,
may I steel a some time again?  I´ve seen that you made some PX4 correction, thank you. However currently PX4 has quite some problems with CAN buses. Therefore I switched to ArduCopter, but unfortunately this time things are reversed. I get perfectly the flightmodes and MAV-messages, but the sensors are empty:
[image]
Please advice.
Best
--Seelefant

zs6buj, 2018-10-13:
Hey Seelefant
Have you set up the SRx parameters?

[image]

Seeelefant, 2018-10-13:
Hi Eric,
looks much better now, thank you 🥇 👍 
[image]
Maybe you can add these settings to your Wiki, even Google does not know much about that.

May I ask another question: 
Here https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.-Setup there is a section on "Combining Teensy with FrSky Sensors". Is this still possible with your solution? 
Here https://github.com/Clooney82/MavLink_FrSkySPort/wiki/2.1.-Lipo-Singe-Cell-Monitor there are some remarks how to (mis-) use the Teensy for voltage measurement of the single cells. Could that be implemented within your environment?  Which solution would you recommend?

Best
--Seelefant

Seeelefant, 2018-10-13:
Hey Eric,
well, there is life even beyond Quadrocopters. Maybe I can ask Alex on that topic and do some experiments myself. Drop me a line once your are back, we can have further discussions on that topic  👍 .
Best
--Seeelefant

closed 2019-10-07 by zs6buj