zs6buj/mav2pt #9: Teensy Crashes After Initial Yaapu Success

GregCovey, 2018-09-17:
I have not been successful at getting the MavlinkToPassthru firmware to run on my setup. It starts up ok on the Yaapu telemetry, making it past the flight mode announcement and "GPS 3D fix Lock", but then I hear "Telemetry Lost" from my Taranis X9D and the Teensy 3.1 LED stays on instead of slow flashing. I have tried .hex files from MavlinkToPassthru firmware v0.11, v1.0.4 and v1.0.5. They all crash the same way.

I am using a Taranis X9D with OpenTX 2.2.1 and Yaapu v1.5.0. Any ideas are appreciated...

Here is my code setup for v1.0.4:

```
#include <CircularBuffer.h>
#include <GCS_MAVLink.h>

//************************************* Please select your options here before compiling **************************
//#define PX4_Flight_stack // If your flight stack is PX4 and not APM, un-comment this line
// Choose one (only) of these target boards
#define Target_Board 0 // Teensy 3.x Un-comment this line if you are using a Teensy 3.x
//#define Target_Board 1 // Blue Pill STM32F103C OR un-comment this line if you are using a Blue Pill STM32F103C
//#define Target_Board 2 // Maple_Mini STM32F103C OR un-comment this line if you are using a Maple_Mini STM32F103C

// Choose one (only) of these three modes
#define Ground_Mode // Converter between Taranis and LRS tranceiver (like Orange)
//#define Air_Mode // Converter between FrSky receiver (like XRS) and Flight Controller (like Pixhawk)
//#define Relay_Mode // Converter between LRS tranceiver (like Orange) and FrSky receiver (like XRS) in relay box on the ground

//#define Use_Local_Battery_mAh // Un-comment this if you want to define battery mAhs here, no FC tx line needed. Alternatively
// enter battery capacities into yaapu's LUA script menu

const uint16_t bat1_capacity = 5200; // These are ignored if the above #define is commented out
const uint16_t bat2_capacity = 0;

#define SPort_Serial 1 // The default is Serial 1, but 3 is possible if we don't want aux port

//#define Aux_Port_Enabled // For BlueTooth or other auxilliary serial passthrough 
```

zs6buj, 2018-09-18:
Hi Greg

Are you using the Ardupilot flight stack with a Pixhawk? Have you set the SRx  parameters in Mission Planner to request regular Mavlink telemetry. Then try #define Debug_All,  paste the debug text into a .txt file and post it here.

Let's see if we can get to the bottom of the problem.

Eric

GregCovey, 2018-09-18:
Hi Eric,
I am using APM Plane v3.9.1 and I can connect to Mission Planner via a WiFi module or SiK Telemetry module. I am also using a ULRS link between the Pixhawk and Taranis X9D with OrangeRx module. So the Teensy 3.1 seems to be wired up to eavesdrop on the MavLink comms coming out of the ULRS OrangeRx module and send FrSky telemetry to the Taranis S.Port pin. Very cool!

My assumption is that normal MavlinkToPassthru firmware operation has the LED flashing slowly about 500msec. When the LED stays on, something bad happened and I hear the Taranis announce, "Telemetry Lost". Does this make sense?

I believe the default SR1 setting is for MavLink1. I will post the parameters for the Telem1 port here. Also, I will try the Debug_All but I am uncertain where to find the debug text?

Thank you!

zs6buj, 2018-09-18:
Hi Greg

All ok, just make sure you enter parameters for the appropriate serial port you are using to feed ULRS. 

Something like this, but feel free to change the frequency:

[image]

 The SR0 corresponds with the USB UART, SR1 with Telem1 and so on.  The debug messages come out of the Serial (0), the USB port, and are displayed from the Arduino IDE. Navigate to Tools/Serial Monitor. When the firmware hags/stops, copy and paste the entire debug output from the serial monitor display window into a .txt file.

"Telemetry Lost" occurs when the Taranis does not get a valid (and strong enough) RSSI reading through the telemetry. So it could be caused by quite a few possible problems. You should delete all sensors and and discover new sensors in the Taranis. With ULRS, I assume you are using an Orange or similar receiver?

zs6buj, 2018-09-18:
The LED should be solid when everything is working properly.

GregCovey, 2018-09-18:
Eric,

The Serial Monitor appears to be a simple terminal with no way to manage the logging. The debug information simply scrolls off the screen with no way to capture it.

I had Mission Planner running all the time so only the Teensy seems to get stuck after reporting "Telemetry Lost". I think it is a false report since I have full control from the Taranis to the Pixhawk.

I'm not sure what is going on here...

GregCovey, 2018-09-18:
I tried my good dipole antennas and got the same result (with a good real RSSI) so for some reason my RSSI is not making it to your code but the Frs_Dummy_rssi works fine.

GregCovey, 2018-09-18:
Eric,

I wonder if the issue is related to my bench setup not having the analog RSSI line from the ULRS connected or not having RSSI assigned to a PWM channel? In other words, there is currently no RSSI signal from the URLS connected to my Pixhawk. Previously, the analog RSSI line was connected to an old Arduflyer when it was in my Skywalker.

Cheers!

zs6buj, 2018-09-19:
Hi Greg
Yes , well done, you have located the problem! ::)

You need to set up rssi to use channel 16 on the ULRS and it should be fine.

GregCovey, 2018-09-19:
Issue resolved.

closed 2018-09-19 by GregCovey