zs6buj/mav2pt #10: Frsky HORUS X10 with Dragonlink v3

kgoltsov, 2018-10-24:
Hello Eric,
I have a Horus X10 with a Dragonlink v3 configured to mavlink transmission. 
My setup is Pixhack v3 (Ardupilot 3.9.2)  ---> Dragonlink microRX ---> Dragonlink TX (with MavLink Decoding) ---> Frsky R-XSR ---> Frsky HORUS X10, between Dragonlink TX and R-XSR script v.1.05 running on a Teensy 3.2 in Relay Mode.
When I run the yaapu lua script all indicators start to display incorrect information. Any, on the wrong.
 Here is the log [Debug_All.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2512552/Debug_All.txt)
I don't know why. I need your help.

Best regards, Kirill.

zs6buj, 2018-10-25:
Could you show me how you wired the Teensy.

zs6buj, 2018-10-25:
Relay_Mode works like Air_Mode, with the exception that RSSI is relayed from the Pixhawk rather than taken from the Frsky-XSR. 

Please could you also un-comment Frs_Debug_All and run again

latitude=101.8881989  longitude=102.0312653   looks very strange.  Latitude can't be more than 90 deg North, right? 

Bat volts=7.420 .   Is your system 2S?

kgoltsov, 2018-10-25:
> Hi Kirill,
> 
> It's not clear to me where the Teensy is. Is it between the Dragonlink TX and Frsky R-XSR? The Mavlink is coming in fine, but nothing going out. I'm a bit pressed for time, but will look more closely later today.
> 
> You could try the Teensy in Ground Mode then into the back of the Horus, but I guess you want to be free to walk around.
> 
> regards
> Eric

Hi Eric,
Yes, teensy Is between the Dragonlink TX and Frsky R-XSR. I tried the Teensy in Ground Mode then into the back of the Horus, but this setup doesn't work at all.

kgoltsov, 2018-10-25:
> Could you show me how you wired the Teensy.

[image]
[image]

kgoltsov, 2018-10-25:
> Relay_Mode works like Air_Mode, with the exception that RSSI is relayed from the Pixhawk rather than taken from the Frsky-XSR.
>
> Please could you also un-comment Frs_Debug_All and run again [...]

Un-comment Frs_Debug_All and run again. Nothing has changed.
I fixed a bug in Dragonlink RX settings, now Horus screen readings are stable but not all data is displayed.
[image]
Latitude and longitude are no longer displayed, Bat volts is correct, i use 4S, the compass is not working and flight modes are displayed for quad, and should be for Plane.
[Debug_All Relay Mode.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2517110/Debug_All.Relay.Mode.txt)

zs6buj, 2018-10-26:
Well, we are making a little progress. At least you have something on the screen now. :)

It would be very helpful if you could activate these two defines:

#define Mav_Debug_All
#define Frs_Debug_All

I need to see what is coming in and going out of the Teensy. In the short .txt you just sent, we did not yet have a GPS lock, so lat/lon are zero.

From your debug.txt however, looking at the heartbeat record, your system is sending Mav_Type (ap_type) = 2

See the table below, that means the vehicle type is a Quad. So from the start we have bad Mavlink telemetry entering the Teensy. :(

Hope this helps.

`Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=3  ap_base_mode=81 ap_custom_mode=12584961  ap_system_status=4  ap_mavlink_version=3`

Value | Field Name | Description
-- | -- | --
0 | MAV_TYPE_GENERIC | Generic micro air vehicle.
1 | MAV_TYPE_FIXED_WING | Fixed wing aircraft.
2 | MAV_TYPE_QUADROTOR | Quadrotor
3 | MAV_TYPE_COAXIAL | Coaxial helicopter
4 | MAV_TYPE_HELICOPTER | Normal helicopter with tail rotor.
5 | MAV_TYPE_ANTENNA_TRACKER | Ground installation
6 | MAV_TYPE_GCS | Operator control unit / ground control station
7 | MAV_TYPE_AIRSHIP | Airship, controlled
8 | MAV_TYPE_FREE_BALLOON | Free balloon, uncontrolled
9 | MAV_TYPE_ROCKET | Rocket
10 | MAV_TYPE_GROUND_ROVER | Ground rover
11 | MAV_TYPE_SURFACE_BOAT | Surface vessel, boat, ship
12 | MAV_TYPE_SUBMARINE | Submarine
13 | MAV_TYPE_HEXAROTOR | Hexarotor
14 | MAV_TYPE_OCTOROTOR | Octorotor
15 | MAV_TYPE_TRICOPTER | Tricopter
16 | MAV_TYPE_FLAPPING_WING | Flapping wing
17 | MAV_TYPE_KITE | Kite
18 | MAV_TYPE_ONBOARD_CONTROLLER | Onboard companion controller
19 | MAV_TYPE_VTOL_DUOROTOR | Two-rotor VTOL using control surfaces in vertical operation in addition. Tailsitter.
20 | MAV_TYPE_VTOL_QUADROTOR | Quad-rotor VTOL using a V-shaped quad config in vertical operation. Tailsitter.
21 | MAV_TYPE_VTOL_TILTROTOR | Tiltrotor VTOL
22 | MAV_TYPE_VTOL_RESERVED2 | VTOL reserved 2
23 | MAV_TYPE_VTOL_RESERVED3 | VTOL reserved 3
24 | MAV_TYPE_VTOL_RESERVED4 | VTOL reserved 4
25 | MAV_TYPE_VTOL_RESERVED5 | VTOL reserved 5
26 | MAV_TYPE_GIMBAL | Onboard gimbal
27 | MAV_TYPE_ADSB | Onboard ADSB peripheral
28 | MAV_TYPE_PARAFOIL | Steerable, nonrigid airfoil
29 | MAV_TYPE_DODECAROTOR | Dodecarotor
30 | MAV_TYPE_CAMERA | Camera
31 | MAV_TYPE_CHARGING_STATION | Charging station
32 | MAV_TYPE_FLARM | Onboard FLARM collision avoidance system

kgoltsov, 2018-10-26:
[Debug_Mav and FRS Relay Mode.txt](https://github.com/zs6buj/MavlinkToPassthru/files/2520530/Debug_Mav.and.FRS.Relay.Mode.txt)

kgoltsov, 2018-10-27:
Eric, I tried to run the Yaapu script without Dragonlink (I used r-xsr and LRS R9M) and everything works fine. I think the problem in Dragonlink. I will check with JettPilot on rcgroups how to get Mavlink telemetry on UEXP of DL tx. Thanks for the help!

closed 2019-10-07 by zs6buj