zs6buj/mav2pt #11: MavLink Passthrough no telemetry data on X9D+

KadaverJoe, 2019-01-03:
Hi Eric,
I copy the text of the issue I have set up in the yaapu github (https://github.com/yaapu/FrskyTelemetryScript/issues/17).
Maybe you can help me with that?

I connected my ULRS 1W module to a teensy 3.2 like shown in the ground side picture. If I connect the teensy to the Arduino Serial monitor, I can see Mavlink messages.
I set up the MavlinkToPassthru_v1.0.8 script as follows:
#define Target_Board 0
#define Ground_Mode
#define Battery_mAh_Source 2

My FC:
Pixhawk 4 with ArduPlane 3.9.4

These are the messages, which I can see in the serial monitor:
Mavlink in #30 Attitude: ap_roll degs=-0.8 ap_pitch degs=-0.3 ap_yaw degs=-86.3
Mavlink in #74 VFR_HUD: Airspeed= 1.35 Groundspeed= 0.10 Heading= 273 Throttle %= 0 Baro alt= 36 Climb rate= 0.03
Mavlink in #147 Battery Status: bat id= 0 bat current mA= 270 ap_current_consumed mAh= 54 my di/dt mAh= 1
Mavlink in #27 Raw_IMU: accX=-0.04 accY=0.01 accZ=-1.00
Mavlink in #29 Scaled_Pressure: press_abs=1029.4hPa press_diff=0.000hPa temperature=35.1C
Mavlink in #30 Attitude: ap_roll degs=-0.7 ap_pitch degs=-0.3 aMavlink in #30 Attitude: ap_roll degs=-0.7 ap_pitch degs=-0.2 ap_yaw degs=-86.3
Mavlink in #27 Raw_IMU: accX=-0.04 accY=0.01 accZ=-1.00
Mavlink in #29 Scaled_Pressure: press_abs=1029.5hPa press_diff=0.000hPa temperature=35.1C
Mavlink in #1 Sys_Status: Bat volts=18.851 Bat amps=0.3 mAh=0.037125 Total mAh=1.188 Bat1 cell count= 5
Mavlink in #125 Power Status: Vcc= 5019 Vservo= 5097 flags= 1
Mavlink in #24 GPS_RAW_INT: ap_fixtype=3 3D Fix sats visible=15 latitude=54.1732979 longitude=12.3405924 gps alt amsl=25.4 eph (hdop)=77 epv (vdop)=127 vel=0.130 cog=346.6
Mavlink in #65 RC_Channels: Channel count= 16 values: 1=1493 2=1494 3=982 4=1494 5=1494 6=1494 7=1494 8=1817 9=1494 10=1494 11=1494 12=1494 13=1494 14=1494 15=1494 16=982 Receive RSSI=49.21
Mavlink in #74 VFR_HUD: Airspeed= 1.32 Groundspeed= 0.13 Heading= 273 Throttle %= 0 Baro alt= 36 Climb rate= -0.17
Mavlink in #30 Attitude: ap_roll degs=-0.7 ap_pitch degs=-0.2 ap_yaw degs=-86.3
Mavlink in #147 Battery Status: bat id= 0 bat current mA= 270 ap_current_consumed mAh= 55 my di/dt mAh= 1
Mavlink in #30 Attitude: ap_roll degs=-0.7 ap_pitch degs=-0.3 ap_yaw degs=-86.3
Mavlink in #74 VFR_HUD: Airspeed= 1.31 Groundspeed= 0.13 Heading= 273 Throttle %= 0 Baro alt= 36 Climb rate= 0.45
Mavlink in #0 Heartbeat: ap_type=1 ap_autopilot=3 ap_base_mode=89 ap_custom_mode=11 ap_system_status=5 ap_mavlink_version=3
Mavlink in #0 Heartbeat: ap_type=27 ap_autopilot=8 ap_base_mode=4 ap_custom_mode=0 ap_system_status=1 ap_mavlink_version=3
Mavlink in #33 GPS Int: ap_lat=54.xxxxxx ap_lon=12.xxxxxx ap_amsl=36 ap_alt_ag=-0.0 ap_vx=0.00 ap_vy=0.00 ap_vz=0.00 ap_gps_hdg=273.7
Mavlink in #27 Raw_IMU: accX=-0.05 accY=0.01 accZ=-1.00
Mavlink in #29 Scaled_Pressure: press_abs=1029.4hPa press_diff=0.000hPa temperature=35.1C
Mavlink in #1 Sys_Status: Bat volts=18.852 Bat amps=0.3 mAh=0.045344 Total mAh=1.233 Bat1 cell count= 5
Mavlink in #125 Power Status: Vcc= 5027 Vservo= 5102 flags= 1
Mavlink in #24 GPS_RAW_INT: ap_fixtype=3 3D Fix sats visible=15 latitude=54.xxxxxxx longitude=12.xxxxxxx gps alt amsl=25.2 eph (hdop)=77 epv (vdop)=127 vel=0.180 cog=346.6
Mavlink in #65 RC_Channels: Channel count= 16 values: 1=1493 2=1494 3=982 4=1494 5=1494 6=1494 7=1494 8=1817 9=1494 10=1494 11=1494 12=1494 13=1494 14=1494 15=1494 16=982 Receive RSSI=49.21

But if I start my Taranis, the yaapu telemetry page tells me "no telemetry data". Is there anything I did wrong?

zs6buj, 2019-01-03:
Hi OK

I'm not at home right now, so can't check the detail, but the LUA scripts depend on RSSI being good. If I remember correctly, channel 16 for rssi didn't work for me, and I ended up using channel 12. There is a #define you can try to force rssi good, for testing purposes only.

KadaverJoe, 2019-01-03:
Hi Eric,
thanks for the info. I also tried that dummy rssi but that did not help either. I also cannot discover any new sensors on my taranis or in other words: they don't appear.

zs6buj, 2019-01-03:
#define Frs_Debug_All will show all the Frs output, and if you have  #define SPort_Serial   1  , then you should be getting Frs telemetry out of pin 1, correctly inverted.

zs6buj, 2019-01-03:
I did spot something strange:

Mavlink in #0 Heartbeat: ap_type=1 ap_autopilot=3 ap_base_mode=89 ap_custom_mode=11 ap_system_status=5 ap_mavlink_version=3

Mavlink in #0 Heartbeat: ap_type=27 ap_autopilot=8 ap_base_mode=4 ap_custom_mode=0 ap_system_status=1 ap_mavlink_version=3

We are getting a heartbeat from the plane, but also from ap_type = 27, which is enumerated to 

27 | MAV_TYPE_ADSB | ADSB system (standalone) - collision avoidance? 
-- | -- | --

KadaverJoe, 2019-01-03:
Okay, yeah I use an ADSB RX which is connected to Pixhawk! Is this the problem?

KadaverJoe, 2019-01-04:
I have the ULRS, which is plugged in the back of the taranis configured as TX. If I change it to RX, the teensy LED only blinks in operation. If I change it to TX again, it is solid.
I unplugged the ADS B receiver and disabled ADSB in MP, but it seems to be still no connection to the yaapu script.
Do you want me to post the serial monitor messages agin?

Thanks for your help, Eric!

zs6buj, 2019-01-04:
Yes, please post the serial output and I'll look for clues. I'll check out the second heartbeat, as it should be handled explicitly. Do you think you could post a telemetry log?

KadaverJoe, 2019-01-04:
Yes, of course I could post a telemetry log. I guess, you want it with ADSB connected and enabled? Is it enough to post a log, where the plane is just sitting in my room?

KadaverJoe, 2019-01-04:
Eric,
I got GPS lock in tlog and I uploaded it. No hurry with the debugging! Do you want me to post the serial monitor of teensy as well? If so, with or without ADSB RX connected?

Thanks for your help!

[TLOG-ADSB.zip](https://github.com/zs6buj/MavlinkToPassthru/files/2726350/TLOG-ADSB.zip)

zs6buj, 2019-01-04:
The TLog looks fine KJ. I'll get back to you. :)

KadaverJoe, 2019-01-05:
Hi Eric,
thank you very much for the info! I have no idea what to change then! Is it okay to set the ULRS in the back of the taranis to "TX" mode? As I mentioned before, in RX mode it seems not to work as the Teensy LED is blinking. On what channel do I need to set up the Pixhawk RSSI? Somebody mentioned APM Plane can handle only 12 channels?
These are the serial monitor messages, if I activate MAV_Debug_Rssi and Mav_Debug_RI also activated Frs_Dummy_rssi:
Mavlink in #65 RC_Channels: Channel count= 16 values:  1=1493 2=1494 3=982 4=1494 5=1494 6=1494 7=1494 8=1817 9=1494 10=1494 11=1494 12=1494 13=1494 14=1494 15=1494 16=982  Receive RSSI=49.21
 Dummy rssi=87
Frsky out:  rssi=87

KadaverJoe, 2019-01-05:
It still don't see any telemetry even with the Frs_Dummy_rssi uncommented. Am I too blind to see where the problem is?

zs6buj, 2019-01-05:
What did you change that you now get rssi=87 and the tlog has rssi =0 ?
This is rather confusing.

The transceiver in the back of the Taranis is designated the "tx", because it was called tx before two-way telemetry came about. So you must set the firmware into tx mode and ensure all the fields are completed properly.  I would use ch  12 for rssi. The transceiver module on the aircraft is designated rx, and set up as such.

Ensure that the Teensy firmware is set up in Ground mode, and if you are receiving good Mavlink telemetry at the the Taranis, then good Frsky telemetry should come out of pin 1 of the Teensy to the bottom pin in the back bay of the Taranis. 

You should maybe do some more reading on how to set up the rssi.

zs6buj, 2019-01-06:
When you are sure you are set up correctly, and it still displays "no telemetry data" give me a new tlog and ill test it for you again.

KadaverJoe, 2019-01-06:
Hi Eric,
I have never read somewhere, that the transceiver on the aircraft has to set up as RX. I did that and it solved a couple of problems I had before. But still no telemetry on the taranis...I think I have set everything up correctly.
Transceiver aircraft: RX, RSSI: Send RSSI on RC channel - Channel 12
Pixhawk settings: SERIAL1_BAUD: 19, SERIAL1_PROTOCOL: 1, RSSI_CHANNEL: 12, RSSI_TYPE: 2
Transceiver Taranis: TX, channel setup seems to be correct as I get Mavlink Data in ULRS CC

Teensy Serial monitor:
Starting .... 
Target Board is Teensy 3.x
Ground Mode
Battery_mAh_Source = 3 - Define battery capacities in the LUA script
Using Serial_1 for S.Port
hb_count=1
hb_count=2
hb_count=3
mavgood=true
Frsky out:  rssi=255
Mavlink in #65 RC_Channels: Channel count= 12 values:  1=1497 2=1496 3=1497 4=1496 5=1497 6=1498 7=1496 8=1497 9=1497 10=1496 11=1497 12=1839  Receive RSSI=83.86
Frsky out:  rssi=83
Mavlink in #65 RC_Channels: Channel count= 12 values:  1=1497 2=1496 3=1497 4=1496 5=1497 6=1498 7=1496 8=1497 9=1497 10=1496 11=1497 12=1839  Receive RSSI=83.86
Frsky out:  rssi=83
Mavlink in #65 RC_Channels: Channel count= 12 values:  1=1497 2=1496 3=1497 4=1497 5=1497 6=1496 7=1497 8=1496 9=1497 10=1497 11=1497 12=1853  Receive RSSI=85.43
Frsky out:  rssi=85
Mavlink in #65 RC_Channels: Channel count= 12 values:  1=1497 2=1497 3=1496 4=1497 5=1497 6=1497 7=1496 8=1497 9=1496 10=1497 11=1497 12=1860  Receive RSSI=86.22
Frsky out:  rssi=86
Mavlink in #65 RC_Channels: Channel count= 12 values:  1=1497 2=1497 3=1496 4=1497 5=1497 6=1496 7=1497 8=1501 9=1496 10=1497 11=1496 12=1861  Receive RSSI=86.22
Frsky out:  rssi=86
Mavlink in #65 RC_Channels: Channel count= 12 values:  1=1497 2=1496 3=1497 4=1497 5=1496 6=1497 7=1496 8=1497 9=1497 10=1498 11=1497 12=1860  Receive RSSI=86.22
Frsky out:  rssi=86
Mavlink in #65 RC_Channels: Channel count= 12 values:  1=1497 2=1497 3=1496 4=1497 5=1497 6=1496 7=1497 8=1497 9=1496 10=1497 11=1497 12=1856  Receive RSSI=85.83
Frsky out:  rssi=85
Mavlink in #65 RC_Channels: Channel count= 12 values:  1=1497 2=1497 3=1496 4=1497 5=1497 6=1496 7=1497 8=1497 9=1496 10=1497 11=1497 12=1853  Receive RSSI=85.43
Frsky out:  rssi=85
Mavlink in #65 RC_Channels: Channel count= 12 values:  1=1496 2=1497 3=1497 4=1496 5=1497 6=1497 7=1496 8=1497 9=1497 10=1496 11=1497 12=1864  Receive RSSI=86.61
Frsky out:  rssi=86

I think I have everything connected as it should. I have no idea what is wrong with it. The funny thing is when I activate the dummy rssi, it stil does not work. 
Btw: I did not change anything to get rssi 87.

KadaverJoe, 2019-01-06:
Sure!
One more question: I have three ULRS transceivers. One connected to aircraft, one to the pc and one in the taranis bay.
But I cannot talk with all three together. So when I record the tlog, I shut down my Taranis. Might this be related with the rssi problem?

KadaverJoe, 2019-01-06:
[TLOG.zip](https://github.com/zs6buj/MavlinkToPassthru/files/2730365/TLOG.zip)

zs6buj, 2019-01-06:
Are the pc and taranis units set up identically?

Are these Orange radios?

zs6buj, 2019-01-06:
So pc radio is in tx mode?

zs6buj, 2019-01-06:
Could the S.Port pin (1) on the Teensy be blown?
Maybe you should compile for alternative output pin.

KadaverJoe, 2019-01-06:
Yes, both units are configured the same. If I connect either the taranis module or the the PC module via FTDI to ULRS CC and connect them either one to Pixhawk, both modules have connection. Both as TX with the same channels etc. It works if I have all three powered but it is very very slow! As I shut down the taranis, it is back at normal speed.
All modules are 1W Orange radios.
I soldered the SPort cable from pin (1) to pin (3) and changed the code to:
#define SPort_Serial   3

Do I need to change anything else? No telem so far...very confusing stuff...
Can I somehow check the sport input to the taranis?

zs6buj, 2019-01-06:
You also have a ground wire to Taranis, I'm sure.

zs6buj, 2019-01-06:
Serial 3 is pin 8

KadaverJoe, 2019-01-06:
thank your for the hint! Now it works! But hows it possible that pin (1) is blown? Its a brand new teensy...hmmm.
thank you so much for you patient help!

KadaverJoe, 2019-01-06:
the only thing I need to find out now is how to set up my taranis as the transmitter over Orange RX ;-)

zs6buj, 2019-01-06:
Glad we got it working KJ. I've blown up serial 1 myself before, that's why I had the serial 3 option.

On the Taranis radio set to external tx.

KadaverJoe, 2019-01-06:
okay I will give it a try. Is it normal that there can only be two modules connected to each other?

closed 2019-01-06 by KadaverJoe