zs6buj/mav2pt #19: Navio2 & Jumper T16

fadilrosli, 2019-09-11:
Currently I'm having problem getting the telemetry works on my Jumper T16.
Basically I'm using Navio2 (running arduplane) --> USB to Serial Converter -->Teensy 3.2 (v2.26)(airmode) --->Frsky X6R (Sport) ---> airlink -->Jumper T16 (4in1 module) running Yaapu (horus) script 1.8.0

The only data that I have now are the X6R rssi and battery status.
On the teensy the led is constantly on which mean that it receives the mavlink data from the usb-to-serial converter.

The reason I'm using usb for mavlink is I already used the uart port for OSD and GCS telemetry.

When I run arduino serial monitor i got these messages:
hb_count=1
hb_count=2
hb_count=3
mavgood=true
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.

Please advice. Thanks.

zs6buj, 2019-09-11:
Sensor table exceeded means that nothing is going out. Output buffer gets
full.

*From:* fadilrosli [mailto:notifications@github.com]
*Sent:* Wednesday, 11 September 2019 1:53 PM
*To:* zs6buj/MavlinkToPassthru
*Cc:* Subscribed
*Subject:* [zs6buj/MavlinkToPassthru] Navio2 & Jumper T16 (#19)

Currently I'm having problem getting the telemetry works on my Jumper T16.
Basically I'm using Navio2 (running arduplane) --> USB to Serial Converter
-->Teensy 3.2 (v2.26)(airmode) --->Frsky X6R (Sport) ---> airlink -->Jumper
T16 (4in1 module) running Yaapu (horus) script 1.8.0

The only data that I have now are the X6R rssi and battery status.
On the teensy the led is constantly on which mean that it receives the
mavlink data from the usb-to-serial converter.

The reason I'm using usb for mavlink is I already used the uart port for
OSD and GCS telemetry.

When I run arduino serial monitor i got these messages:
hb_count=1
hb_count=2
hb_count=3
mavgood=true
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.

Please advice. Thanks.

zs6buj, 2019-09-11:
Try changing to uart3 (pin 8)  for output to XR6. Take static precautions when you touch/ solder the pins.

See my answer to you on RC groups PM

fadilrosli, 2019-09-11:
I have changed to uart3 (pin 8) and it works. Thank you brother.

zs6buj, 2019-09-11:
Happy to help :).

closed 2019-09-11 by zs6buj

fadilrosli, 2020-01-18:
Hi,
I try to compile the latest v2.50 but I got an error says "'mavlink_msg_scaled_imu_get_temperature' was not declared in this scope ap26_temp = mavlink_msg_scaled_imu_get_temperature(&R2Gmsg);   " How can I fix this? Thanks

zs6buj, 2020-01-18:
That particular identifier is new in mavlink2 I think. Try cloning the latest mavlink2 libraries and copy to libraries folder.

fadilrosli, 2020-01-18:
You are right bro. Replaced the old lib with the new one and it works. Thank you.

Moksh-2000, 2020-04-21:
How to use this script , can you tell me , this is interesting stuff.

Thank you 🙂

zs6buj, 2020-04-22:
Just take a look at the Wiki above :)