zs6buj/mav2pt #20: relay mode doesn't seem to work

burtgree, 2019-10-03:
Been doing some testing, using a teensy 3.2 board.
found that I can make it work with version 2.14 correctly.
I didn't try every version but 2.14 works, but 2.17 up to 2.32 don't seem to work anymore.
below are the settings I used with 2.14
// ******************************* Please select your options below before compiling *******************************

// Do not enable for FlightDeck
#define PlusVersion  // Added support for 0x5009 Mission WPs, 0x50F1 Servo_Channels, 0x50F2 VFR_Hud

// Choose one only of these three modes
//#define Ground_Mode          // Converter between Taranis and LRS tranceiver (like Orange)
//#define Air_Mode             // Converter between FrSky receiver (like XRS) and Flight Controller (like Pixhawk)
#define Relay_Mode           // Converter between LRS tranceiver (like Orange) and FrSky receiver (like XRS) in relay box on the ground

// Choose one only of these Flight-Controller-side I/O channels 
// How does Mavlink telemetry enter the converter?
#define FC_Mavlink_IO  0    // Serial Port (default)         
//#define FC_Mavlink_IO  1    // BlueTooth Classic - ESP32 only
//#define FC_Mavlink_IO  2    // WiFi - ESP32 only
//#define FC_Mavlink_IO  3    // SD Card / TF - ESP32 only

// Choose one only of these GCS-side I/O channels
// How does Mavlink telemetry leave the converter?
// These are optional, and in addition to the S.Port telemetry output
//#define GCS_Mavlink_IO  9    // NONE (default)
//#define GCS_Mavlink_IO  0    // Serial Port        
//#define GCS_Mavlink_IO  1    // BlueTooth Classic - ESP32 only
//#define GCS_Mavlink_IO  2    // WiFi - ESP32 only

//#define GCS_Mavlink_SD      // SD Card  - for ESP32 only

// Choose one - for ESP32 only
//#define WiFi_Protocol 1    // TCP/IP
//#define WiFi_Protocol 2    // UDP     useful for Ez-WiFiBroadcast in STA mode

// Choose one - AP means advertise as an access point (hotspot). STA means connect to a known host
//#define WiFi_Mode   1  //AP            
//#define WiFi_Mode   2  // STA

//#define Battery_mAh_Source  1  // Get battery mAh from the FC - note both rx and tx lines must be connected      
//#define Battery_mAh_Source  2  // Define bat1_capacity and bat2_capacity below and use those 
const uint16_t bat1_capacity = 5200;       
const uint16_t bat2_capacity = 0;
#define Battery_mAh_Source  3  // Define battery mAh in the LUA script on the Taranis/Horus - Recommended

#define SPort_Serial   1            // The default is Serial 1, but 3 is possible 
#define LRS_RSSI     // Un-comment this line only if you are using a ULRS, QLRS or similar telemetry system

// ****************************** Set your time zone here ******************************************
// Date and time determines the TLog file name
//const float Time_Zone = 10.5;    // Adelaide
const float Time_Zone = 2.0;    // Jo'burg
bool daylightSaving = false;

I tried to use the same settings while testing 2.17, 2.20, 2.25, & 2.31

zs6buj, 2019-10-04:
burtgree, 

I would be happy to help you make Mav2PT work in Relay mode.

Let's take it one step at a time, and let's use the latest version Mav2PT_v2.31.

Can you successfully compile and flash?

zs6buj, 2019-10-04:
Go to confihg.h and config there exactly as you have illustrated above. I assume you are using  a long range system (LRS)?

burtgree, 2019-10-05:
Eric

I checked that my setup was still working with the 2.14, it was.
I made the changes in the config.h file, it did compile and I was able to load it into the teensy.
I deleted the sensors on my transmitter and did a discover, only comes up with RSSI 25 & RxBt 25.
Last question yes I have it setup for a Dragonlink system, but it's all on a bench till i work out everything. for rssi I use this line "mavlink_msg_rc_channels_raw_get_rssi(&R2Gmsg); " and have the channel I use for RSSI on a pot on the transmitter for testing.
I will attach my config.h after this. I also will include the output of the teensy.

Starting .... d:\Users\XXXXX
[config.zip](https://github.com/zs6buj/MavlinkToPassthru/files/3693600/config.zip)
\Documents\Arduino\Mav2PT_v2.31_teensey\Mav2PT_v2.31_teensey
Target Board is Teensy 3.x
Relay Mode
Battery_mAh_Source = 3 - Define battery capacities in the LUA script
Using Serial_1 for S.Port
RSSI from PWM channel
Mavlink Serial In
Waiting for telemetry
hb_count=1
hb_count=2
hb_count=3
mavgood=true
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.
Warning, sensor table exceeded. Push ignored.

zs6buj, 2019-10-07:
The sensor table (output to S.Port) overflows because nothing is being forwarded from it.  In relay mode I read from the ground FrSky receiver to find a sensor slot, and push our new sensor frame into it. That is likely not happening for some reason. Since 2.14 is working for you, and your config looks correct, I should test relay mode here again to make sure it's working for me.

If you find the time you could un-comment //#define Frs_Debug_All and post the serial output here for me.

burtgree, 2019-10-08:
Eric

attached is about a minute of debug info like you asked for.

Like before, I tested the setup with 2.14 first, so I know the wiring is all correct, just uploaded the new bin.

later
burt

[debug_info.zip](https://github.com/zs6buj/MavlinkToPassthru/files/3700202/debug_info.zip)

zs6buj, 2019-10-08:
burtgree

I apologise for doubting your information! A nasty typo affecting relay mode was introduced in v2.17. 

#define BufStatusLed 14 overtyped as BufStatusLed  1 (SPort) !

So the S.Port could not read (it could still write) from the FrSky receiver.

Well spotted, and thank you for you assistance.

Eric

burtgree, 2019-10-09:
Eric

Thanks, just tried it and works.

now if I could only get a ESP32 working. I came to the conclusion it's the libraries for each board.
i can get parts to work but not all, that's why I tried the teensy.

zs6buj, 2019-10-09:
Tell me which board you have. Maybe send a picture and we can work it out.

zs6buj, 2019-10-09:
I guess you know you must insert a bi-directional two-wire to single wire converter when you use the ESP32. The ESP32 can invert, but not convert to single wire.

zs6buj, 2019-10-09:
I use[ this ](https://www.banggood.com/FrSky-Yaapu-Telemetry-Converter-Cable-Pixhawk-to-Horus-X10S-X12S-Taranis-X9DP-QX7X7S-Smart-Port-Receiver-X8R-X4R-XSR-RS8R-p-1535564.html?rmmds=buy&cur_warehouse=CN)inverter/converter.

Since the commercial board **inverts** and converts to single wire, the ESP32 must not invert in Relay Mode. I changed v2.33 accordingly for convenience.

burtgree, 2019-10-14:
Eric
I had time to try my ESP32 again, yes it worked for s-port (was using a converter). I was still having a problem, but found it fast. This line has a typo too, should end with 3.
"#define Battery_mAh_Source  1  // Define battery mAh in the LUA script on the Taranis/Horus - Recommended"

Bluetooth worked, still problems with wireless, but it has to do with the libraries for each board. the board I'm using is this one. https://www.amazon.com/gp/product/B07Q576VWZ/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1

I use these 2 libraries:
https://github.com/espressif/arduino-esp32
https://github.com/nodemcu/nodemcu-firmware/tree/dev-esp32

one has problems with the way IP addrresses are used, the other one doesn't like serial ports defined the way your code does it.

I hard coded some changes & have it working, but losing over 50% of packets. In my own attempt to get wireless working, I created a buffer for the UDP packets being sent out of the ESP32. I can get 3 to 4 packets in one buffer, and send that all at once. makes my packet lose down to about 15%.

Thanks for all the hard work, I don't know how your mind can keep all the if & endif straight.

zs6buj, 2019-10-14:
Hi burtgree, I'm please you got some use out of it.  I wonder if you have a lot of RFI nearby causing the lost packets? I assume you view lost packets in Mission Planner? Good work on the buffering. Perhaps we should incorporate it.

Yeah, the multi-platform thing is a bit of a challenge,  but it became very difficult to keep the versions aligned with separate code for each platform.  You probably noticed I don't like capitalizing macros because readability suffers (in my opinion).

zs6buj, 2019-10-14:
>  I was still having a problem, but found it fast. This line has a typo too, should end with 3.
> "#define Battery_mAh_Source 1 // Define battery mAh in the LUA script on the Taranis/Horus - Recommended"

Correct. Thank you!

burtgree, 2019-10-20:
Eric
Got time to work on it again yesterday, I did get it working a lot better but still have problems.I have attached my code changes to the zip file, also a text file with more info on what i did. I still think the problems are all the types of ESP32 boards, & their lib's.[Mav2PT_v2.33_esp32_wifi2.zip](https://github.com/zs6buj/MavlinkToPassthru/files/3748507/Mav2PT_v2.33_esp32_wifi2.zip)

zs6buj, 2019-10-21:
Thank you for this burtgree. I'm going to try your buffering tomorrow.  :)

closed 2019-12-01 by zs6buj