zs6buj/mav2pt #21: Power for Teensy in air mode

kpihus, 2019-10-05:
Hi,

According to this diagram, power for Teensy 3.2 is taken from Pixhawk switch port, in order to get 3.3V power for Teensy. 

But according to Teensy 3.2 specs, Vin pin can handle voltage from 3.6V to 6.0V.

Is there any good reason why Vin pin is not a good idea to power Teensy from Pixhawk motor output port, or better yet from Receiver module?

--
Kpihus

zs6buj, 2019-10-07:
Hi Kpihus

No good reason at all. By all means use the Vin pin. I originally used an STM32F103 uC board with no 3.3v regulator.

zs6buj, 2019-10-07:
Be careful with the Teensy. It is susceptible to static damage, especially serial port 1. Anti-static precautions are advisable.

closed 2019-10-07 by zs6buj