zs6buj/mav2pt #22: GPS location and Flightmode missing

kpihus, 2019-10-09:
Hello, 
I got this converter working yesterday and i have to admit it is awesome. Two minor things still to solve:

1) There is not GPS location displayed on the telemetry screen. GPS fix is ok. Rhe display can show fix info, but no location. Is it safe to assume that it should appear there automatically and if not then this information is missing from telemetry feed?

2) No flightmode is shown on screen. Same with previous, should it appear there automatically?

I'm using PX4 (v 1.9), And Jumper T16 with FrSky X8R receiver. Yaapu script version 1.8.0

Where and how should I start debugging these issues?

yaapu, 2019-10-09:
Hi,
please do a sensor discovery in OpenTX, you're probably missing the GPS sensor and enable the PX4 flight mode support in the Yaapu Widget Config menu.

Eric excuse me for answering this but my first guess is that is a config issue on the radio side :-)

kpihus, 2019-10-09:
@yaapu funny thing, sensor discovery didn't find any additional sensors (besides default RSSI and RX voltage) but somehow your telemetry script still works. You have added some magic dust there? :)

yaapu, 2019-10-09:
@kpihus haha no magic :-) my script doesn't use sensors, the only needed one is GPS.

If everything but GPS works than it's not being sent by @zs6buj firmware and we'll need his help in the end.

Are flight modes working now?

kpihus, 2019-10-09:
Nope they don't.

zs6buj, 2019-10-09:
Hello kpihus:

Alex does not automatically display GPS co-ordinates, though he does have a version with a real-time Google Maps display with your plane on it. I believe it will be combined into the newest release.

You can display GPS co-ords on one of the display screens yourself if you want.

Flight mode should appear automatically. Please try v2.33. I tested it this afternoon. it should work for  PX4. Let me know.

yaapu, 2019-10-09:
Eric, it's a Jumper T16, GPS coords are there, lower right corner :-)

zs6buj, 2019-10-09:
Ok, thanks for jumping in Alex. Excuse me for jumping around myself, but today has been crazy.

zs6buj, 2019-10-10:
> Nope they don't.
> 
> EDIT: Only thing what I modified before burning it to the chip was commented in Air_Mode and commented out Ground_Mode

 Please try v2.33

kpihus, 2019-10-10:
Thank you. Little better results now. 
1) GPS sensor is now discovered and location shown -all good with that

2) Regarding to flight modes, i can see that information also now, but its little bit incorrect.

On Jumper T16, we have 6pos switch for flight modes. And in my case, it has been set up as:
1) Manual
2) Altitude
3) Position
4) Hold
5) Retrun
6) Land

Now when i press buttons on TX, then all thse flightmodes are switched correctly in FC according to QGC. But telemtry screen displays them as:
1) <empty>
2) Manual
3) AltCtrl
4) (cant tell FC rejects HOLD, when not in air)
5) (cant tell FC rejects RETURN when not in air)
6) Rattitude

So it seems that something is still wrong in there.

yaapu, 2019-10-10:
@kpihus did you enable PX4 flight mode support in the widget configuration menu? after doing it you need to power cycle the radio.

kpihus, 2019-10-10:
@yaapu yes it is enabled

zs6buj, 2019-10-10:
Could you enable debugging of frs parameters and post the serial output please

zs6buj, 2019-10-10:
Also Mavis heartbeats. 

Sorry I'm mobile and can't remember the exact syntax.

#define Debug_......

kpihus, 2019-10-10:
@zs6buj what debug options you want me to enable? #define Frs_Debug_All gives lot of crap

zs6buj, 2019-10-10:
No, mav heartbeat and frs params

kpihus, 2019-10-10:
according to that we can say that heartbeat displays modes correctly?

Debu options used:

`#define Mav_Debug_FC_Heartbeat
#define Mav_Debug_GCS_Heartbeat
#define Mav_Debug_Mav2PT_Heartbeat
#define Frs_Debug_Params`

output:

Starting .... Mav2PT_v2.33
Target Board is Teensy 3.x
Air Mode
Battery_mAh_Source = 1 - Get battery capacities from the FC
Using Serial_1 for S.Port
Default RSSI_Source
Mavlink Serial In
Waiting for telemetry
hb_count=1
Period ms=3376	Frsky out Params 0x5007:  fr_param_id=1 fr_frame_type=2 fr_payload=16777218 
01  00  00  02 
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=81 ap_custom_mode=65536  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=1 px4_sub_mode=0 MANUAL
hb_count=2
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=81 ap_custom_mode=65536  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=1 px4_sub_mode=0 MANUAL
Sending mav2pt hb to FC
hb_count=3
mavgood=true
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=81 ap_custom_mode=65536  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=1 px4_sub_mode=0 MANUAL
Battery capacities requested
Battery params successfully read
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=81 ap_custom_mode=131072  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=2 px4_sub_mode=0 ALTITUDE
Sending mav2pt hb to FC
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=81 ap_custom_mode=131072  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=2 px4_sub_mode=0 ALTITUDE
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=89 ap_custom_mode=196608  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=3 px4_sub_mode=0 POSCTL
Sending mav2pt hb to FC
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=89 ap_custom_mode=196608  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=3 px4_sub_mode=0 POSCTL
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=89 ap_custom_mode=196608  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=3 px4_sub_mode=0 POSCTL
Sending mav2pt hb to FC
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=89 ap_custom_mode=196608  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=3 px4_sub_mode=0 POSCTL
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=89 ap_custom_mode=196608  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=3 px4_sub_mode=0 POSCTL
Sending mav2pt hb to FC
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=89 ap_custom_mode=196608  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=3 px4_sub_mode=0 POSCTL
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=29 ap_custom_mode=100925440  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=4 px4_sub_mode=6 AUTO LAND
Sending mav2pt hb to FC
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=29 ap_custom_mode=100925440  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=4 px4_sub_mode=6 AUTO LAND
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=29 ap_custom_mode=100925440  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=4 px4_sub_mode=6 AUTO LAND
Sending mav2pt hb to FC
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=29 ap_custom_mode=100925440  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=4 px4_sub_mode=6 AUTO LAND
Mavlink in #0 Heartbeat: ap_type=2  ap_autopilot=12  ap_base_mode=29 ap_custom_mode=100925440  ap_system_status=3  ap_mavlink_version=3 px4_main_mode=4 px4_sub_mode=6 AUTO LAND
Sending mav2pt hb to FC

zs6buj, 2019-10-10:
Looks correct. There is also a debug for, I think, 5003 flight modes?. To see what is sent to Taranis.

zs6buj, 2019-10-10:
ap_type 2 means a quad. The flight modes are frame dependent, obviously.

kpihus, 2019-10-10:
@zs6buj 

Period ms=977	Frsky out AP_Status 0x5001:  fr_flight_mode=1 fr_simple=0 fr_land_complete=0 fr_armed=0 fr_bat_fs=0 fr_ekf_fs=0 px4_flight_stack=1 fr_payload=4097 00  00  10  01 

this seem to be the message describing flightmode, but this is 5001 instead, correct?

zs6buj, 2019-10-11:
Hi Kristjan

I'm back at my desk, and yes 0x5001 is the correct outgoing S.Port frame for flight mode, and it looks correct. The strange thing is we have debugged and fixed this issue before, but the system is being constantly developed and I guess demons take the opportunity to creep in. Lol.

Alex: I'm thinking maybe I should add FrSky telemetry logging and playback into Mav2PT.

kpihus, 2019-10-11:
I understand your point, but for me, it seems that the frame type is not relevant. 
If Manual mode is with id==0, for boat then I think it should be the same for boat aswell?
Tho I don't know mavlink protocol so deeply so I might be wrong.

Nevertheless, it seems to me that this converter passes information correctly, right?

yaapu, 2019-10-11:
> Alex: I'm thinking maybe I should add FrSky telemetry logging and playback into Mav2PT.

there sure are cases where this would be very usefull, you could use the same format I use to dump debug info from the radio, one packet per line
`radio_time::int, flight_time::int, data_id::uint16,  value::uint32`

@kpihus can you use this [debug version](https://drive.google.com/open?id=1pDZB1hsbt8N3-AOsd7iALfq7ovv3LgEN) of my widget and post the log file produced in the /LOGS folder, file name should be modelname-datetime.plog

kpihus, 2019-10-24:
Just an update. Sorry but i havent got time to test debug script yet. Busy with other (and more important like lidars and stuff) hardware issues. But i'll get back to it as soon as i get my bird successfully airborne

kpihus, 2019-10-25:
@yaapu i guess this is the log file what you looking for: https://www.upload.ee/files/10641695/birdai-20191025_234424.plog.html

tho i'm not sure how to interpret it.

yaapu, 2019-10-30:
this is the converted file 
[birdai-20191025_234424.plog.zip](https://github.com/zs6buj/MavlinkToPassthru/files/3791476/birdai-20191025_234424.plog.zip)

regarding flight modes, they have been reported to work from several users so I would assume they are right, the culprit should be somewhere else...

kpihus, 2019-10-31:
@yaapu how did you convert this file? I think i'll do some more tests so it would be nice to know how to make it readable.

closed 2019-12-01 by zs6buj