zs6buj/mav2pt #23: Warning, sensor table exceeded. Push ignored

Seeelefant, 2019-11-19:
Trying to set up version 2.45 on Wemos Lolin32 Board I get:
"Warning, sensor table exceeded. Push ignored.", no telemetry data
[ESP32.txt](https://github.com/zs6buj/MavlinkToPassthru/files/3866280/ESP32.txt)

zs6buj, 2019-11-20:
Rather use v2.46. Sensor table overflow simply means messages are queued for SPort but S Port is not being read.

Seeelefant, 2019-11-20:
Hey Eric,
this time I get:
22:25:20.611 -> Period ms=63	Frsky out Attitude 0x5006: fr_roll=897 fr_pitch=315 fr_range=0 Frs_Attitude Payload=646017
22:26:12.979 -> Starting .... D:\Users\Wolfgang\Documents\Arduino\Sketch\MavlinkToPassthru-master\Mav2PT_v2.46\Mav2PT_v2.46
22:26:12.979 -> Target Board is ESP32 / Variant is Dev Module
22:26:12.979 -> Air Mode
22:26:12.979 -> Battery_mAh_Source = 3 - Define battery capacities in the LUA script
22:26:12.979 -> Using Serial_1 for S.Port
22:26:12.979 -> RSSI Automatic Select
22:26:12.979 -> Mavlink Serial In
22:26:13.013 -> Mavlink WiFi Out - Protocol is UDP
22:26:13.013 -> Waiting for telemetry
22:26:13.047 -> ESP32 S.Port pins NOT inverted for Air or Relay Modes. Must use a converter
22:26:13.047 -> AutoBaud - Sensing FC_Mav_rxPin 16 
22:26:13.047 -> Trying to connect to OmegaOffice..........
22:26:19.143 -> Failed to connect in STA mode
22:26:19.143 -> Starting AP instead.
22:26:19.143 -> WiFi-Reset ...
22:26:19.654 -> AP IP address: 192.168.4.1  SSID: Mav2Passthru
22:26:19.688 -> UDP started, listening on IP 192.168.4.1, UDP port 14555
22:26:19.824 -> Telem found at 57600 b/s
22:26:20.583 -> hb_count=1
22:26:21.648 -> hb_count=2
22:26:22.605 -> hb_count=3
22:26:22.605 -> mavgood=true
22:27:01.638 -> Sensor buffer full. Check S.Port link
22:27:36.782 -> Sensor buffer full. Check S.Port link
22:28:11.960 -> Sensor buffer full. Check S.Port link
22:28:47.475 -> Sensor buffer full. Check S.Port link

I am not able to read S.Port data neither I am able to connect via QGroundcontrol via WiFi.

Please advice.

zs6buj, 2019-11-21:
1. The data indicates that you are in Air Mode, so you must have a bi-directional inverter converter from the FC to single wire S Port. Could you provide more details please. Px4 or APM and version?
Did you connect to "Mav2PT" hotspot, or you must change to your ssid in the code.

Seeelefant, 2019-11-22:
Well, currently I have PX4 version 1.10 running-do you think I should try ArduCopter? 
Concerning WiFi: I was able to connect to the "Mav2PT" hotspot, but seemely there was no Mavlink connection. 
Concerning the In-/Converter; I built one of these: https://discuss.ardupilot.org/t/some-soldering-required/27613 , which  are working on PixHawks.
Best 
--Seeelefant

zs6buj, 2019-11-22:
Px4 works fin, and all the status messages look correct. To me it looks like you have an electrical problem with the sport side. With wifi make sure you setup like it shows in the wiki for AP Mode UDP and it should connect. Unfortunately I am not at home and working off my phone.

Seeelefant, 2019-11-22:
Thank you Eric, well I have some Chinese ESP32 board....should I attach an oscilloscope to the pins? Sometimes there are problems with ESP32 serial ports: https://hackaday.com/2017/08/17/secret-serial-port-for-arduinoesp32/

zs6buj, 2019-11-22:
Yes if you have a scope take a look at the s.port pin. Disconnect from the esp32 sport pins and you  should see signals from the frsky receiver on the tx wire that goes to esp32 rx pin. You should also see infrequent inverted pulses on the tx pin. Then connect and you should see fairly frequent traffic as mav2pt responds to packets from frsky receiver. If you know wireshark ie similar tool in serial mode you could look at the packets.

zs6buj, 2019-11-22:
Could I ask why you want to use an esp32 with wifi on the aircraft in air mode?

Seeelefant, 2019-11-22:
Well, my idea was to have FrSky telemetry on my Taranis radio and also a Wifi connection to the groundstation. Would you recommend another setup?

zs6buj, 2019-11-22:
Wifi from the plane to the ground will have limited range, and possibly interfere with your rc control signal.

Seeelefant, 2019-11-22:
Well, I tested the other setup with [ESP8266 ](https://docs.px4.io/v1.9.0/en/telemetry/esp8266_wifi_module.html) already. So far, I did not have any problems with interference, neither with range.
Currently I am also looking at https://github.com/HD-Fpv/Open.HD , I certain situations they reported ranges above 10 kms--even with (modified) WiFi.

zs6buj, 2019-11-22:
Very interesting

Seeelefant, 2019-11-30:
Well, the problem with the S.Port is fixed now . It was a problem with a homebrewn inverter.

Now trying to connect via WiFi the ESP32 reboots:
00:47:12.735 -> Air Mode
00:47:12.769 -> Battery_mAh_Source = 3 - Define battery capacities in the LUA script
00:47:12.769 -> Using Serial_1 for S.Port
00:47:12.769 -> RSSI Automatic Select
00:47:12.769 -> Mavlink Serial In
00:47:12.769 -> Mavlink WiFi Out - Protocol is UDP
00:47:12.769 -> Waiting for telemetry
00:47:12.803 -> ESP32 S.Port pins NOT inverted for Air or Relay Modes. Must use a converter
00:47:12.803 -> AutoBaud - Sensing FC_Mav_rxPin 16 
00:47:12.904 -> AP IP address: 192.168.4.1  SSID: Mav2Passthru
00:47:12.938 -> UDP started, listening on IP 192.168.4.1, UDP port 14555
00:47:13.145 -> Telem found at 57600 b/s
00:47:13.249 -> hb_count=1
00:47:14.246 -> hb_count=2
00:47:15.272 -> hb_count=3
00:47:15.272 -> mavgood=true
00:47:28.885 -> E (21427) wifi: pp.c 3470
00:47:28.885 -> 
00:47:33.916 -> E (26425) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
00:47:33.916 -> E (26425) task_wdt:  - IDLE0 (CPU 0)
00:47:33.916 -> E (26425) task_wdt: Tasks currently running:
00:47:33.916 -> E (26425) task_wdt: CPU 0: wifi
00:47:33.916 -> E (26425) task_wdt: CPU 1: IDLE1
00:47:33.916 -> E (26425) task_wdt: Aborting.
00:47:33.916 -> abort() was called at PC 0x400dcaeb on core 0
00:47:33.916 -> 
00:47:33.916 -> Backtrace: 0x4008cbf8:0x3ffbe170 0x4008ce29:0x3ffbe190 0x400dcaeb:0x3ffbe1b0 0x40084f01:0x3ffbe1d0 0x4008ed2e:0x3ffb5de0 0x4008ff15:0x3ffb5e10 0x4008dcef:0x3ffb5e40 0x4008fd65:0x3ffb5ee0 0x4008930d:0x3ffb5f20
00:47:33.949 -> 
00:47:33.949 -> Rebooting...
00:47:33.949 -> ets Jun  8 2016 00:22:57
00:47:33.949 -> 
00:47:33.949 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:47:33.949 -> configsip: 0, SPIWP:0xee
00:47:33.949 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:47:33.949 -> mode:DIO, clock div:1
00:47:33.949 -> load:0x3fff0018,len:4
00:47:33.949 -> load:0x3fff001c,len:1044
00:47:33.949 -> load:0x40078000,len:8896
00:47:33.949 -> load:0x40080400,len:5816
00:47:33.949 -> entry 0x400806ac
00:47:36.793 -> Starting .... D:\Users\Wolfgang\Documents\Arduino\Sketch\MavlinkToPassthru-master\Mav2PT_v2.46\Mav2PT_v2.46
00:47:36.793 -> Target Board is ESP32 / Variant is Dev Module
00:47:36.793 -> Air Mode
00:47:36.827 -> Battery_mAh_Source = 3 - Define battery capacities in the LUA script
00:47:36.827 -> Using Serial_1 for S.Port
00:47:36.827 -> RSSI Automatic Select
00:47:36.827 -> Mavlink Serial In
00:47:36.827 -> Mavlink WiFi Out - Protocol is UDP
00:47:36.861 -> Waiting for telemetry
00:47:36.861 -> ESP32 S.Port pins NOT inverted for Air or Relay Modes. Must use a converter
00:47:36.861 -> AutoBaud - Sensing FC_Mav_rxPin 16 
00:47:36.999 -> AP IP address: 192.168.4.1  SSID: Mav2Passthru
00:47:36.999 -> UDP started, listening on IP 192.168.4.1, UDP port 14555
00:47:37.205 -> Telem found at 57600 b/s
00:47:37.239 -> hb_count=1
00:47:38.267 -> hb_count=2
00:47:39.257 -> hb_count=3
00:47:39.257 -> mavgood=true

zs6buj, 2019-12-01:
Hi Seeelefant

I am happy to see we are making progress. :)

It's strange that cpu1 is idle when the watchdog steps in, Can you check your clock speed setting,. Are you running at 80MHz, 160 MHz or 240 MHz? I believe that 240 is regarded as overclocking, and lucky if it can run reliably.

Does it crash every time, or only sometimes, and does it crash immediately when you connect, or after a few moments. Does QGC or MP connect first?

Seeelefant, 2019-12-01:
Hi Eric,

same problem again, even on another board. It happens immediately I connect to the WLAN, no Groundstation involved:
15:42:29.755 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:42:29.755 -> configsip: 0, SPIWP:0xee
15:42:29.755 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:42:29.755 -> mode:DIO, clock div:1
15:42:29.755 -> load:0x3fff0018,len:4
15:42:29.755 -> load:0x3fff001c,len:1044
15:42:29.755 -> load:0x40078000,len:8896
15:42:29.755 -> load:0x40080400,len:5816
15:42:29.755 -> entry 0x400806ac
15:42:32.588 -> Starting .... D:\Users\Wolfgang\Documents\Arduino\Sketch\MavlinkToPassthru-master\Mav2PT_v2.46\Mav2PT_v2.46
15:42:32.588 -> Target Board is ESP32 / Variant is Dev Module
15:42:32.588 -> Air Mode
15:42:32.623 -> Battery_mAh_Source = 3 - Define battery capacities in the LUA script
15:42:32.623 -> Using Serial_1 for S.Port
15:42:32.623 -> RSSI Automatic Select
15:42:32.623 -> Mavlink Serial In
15:42:32.623 -> Mavlink WiFi Out - Protocol is UDP
15:42:32.657 -> Waiting for telemetry
15:42:32.657 -> ESP32 S.Port pins NOT inverted for Air or Relay Modes. Must use a converter
15:42:32.657 -> AutoBaud - Sensing FC_Mav_rxPin 16 
15:42:32.760 -> AP IP address: 192.168.4.1  SSID: Mav2Passthru
15:42:32.760 -> UDP started, listening on IP 192.168.4.1, UDP port 14555
15:42:33.138 -> Telem found at 57600 b/s
15:42:33.171 -> hb_count=1
15:42:34.198 -> hb_count=2
15:42:35.194 -> hb_count=3
15:42:35.194 -> mavgood=true
15:43:38.322 -> E (70986) wifi: pp.c 3470
15:43:38.322 -> 
15:43:43.385 -> E (75985) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
15:43:43.385 -> E (75985) task_wdt:  - IDLE0 (CPU 0)
15:43:43.385 -> E (75985) task_wdt: Tasks currently running:
15:43:43.385 -> E (75985) task_wdt: CPU 0: wifi
15:43:43.385 -> E (75985) task_wdt: CPU 1: IDLE1
15:43:43.385 -> E (75985) task_wdt: Aborting.
15:43:43.385 -> abort() was called at PC 0x400dcaeb on core 0
15:43:43.385 -> 
15:43:43.385 -> Backtrace: 0x4008cbf8:0x3ffbe170 0x4008ce29:0x3ffbe190 0x400dcaeb:0x3ffbe1b0 0x40084f01:0x3ffbe1d0 0x4008ed2e:0x3ffb5cf0 0x4008ff15:0x3ffb5d20 0x4008dcef:0x3ffb5d50 0x4008f6a7:0x3ffb5df0 0x4008e033:0x3ffb5e10 0x4008e5df:0x3ffb5e40 0x4015b72d:0x3ffb5e80 0x4015b882:0x3ffb5eb0 0x4008fda3:0x3ffb5ee0 0x4008930d:0x3ffb5f20
15:43:43.385 -> 
15:43:43.385 -> Rebooting...
15:43:43.418 -> ets Jun  8 2016 00:22:57
15:43:43.418 -> 
15:43:43.418 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:43:43.418 -> configsip: 0, SPIWP:0xee
15:43:43.418 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:43:43.418 -> mode:DIO, clock div:1
15:43:43.418 -> load:0x3fff0018,len:4
15:43:43.418 -> load:0x3fff001c,len:1044
15:43:43.418 -> load:0x40078000,len:8896
15:43:43.418 -> load:0x40080400,len:5816
15:43:43.418 -> entry 0x400806ac
15:43:46.115 -> Starting .... D:\Users\Wolfgang\Documents\Arduino\Sketch\MavlinkToPassthru-master\Mav2PT_v2.46\Mav2PT_v2.46
15:43:46.115 -> Target Board is ESP32 / Variant is Dev Module
15:43:46.115 -> Air Mode
15:43:46.115 -> Battery_mAh_Source = 3 - Define battery capacities in the LUA script
15:43:46.115 -> Using Serial_1 for S.Port
15:43:46.115 -> RSSI Automatic Select
15:43:46.115 -> Mavlink Serial In
15:43:46.115 -> Mavlink WiFi Out - Protocol is UDP
15:43:46.149 -> Waiting for telemetry
15:43:46.149 -> ESP32 S.Port pins NOT inverted for Air or Relay Modes. Must use a converter
15:43:46.149 -> AutoBaud - Sensing FC_Mav_rxPin 16 
15:43:46.251 -> AP IP address: 192.168.4.1  SSID: Mav2Passthru
15:43:46.285 -> UDP started, listening on IP 192.168.4.1, UDP port 14555
15:43:46.628 -> Telem found at 57600 b/s
15:43:47.180 -> hb_count=1
15:43:48.174 -> hb_count=2
15:43:49.167 -> hb_count=3
15:43:49.167 -> mavgood=true

zs6buj, 2019-12-03:
Hi Wolfgang

I just ran through an identical test here, and I get:

```
Starting .... C:\Users\Richard\Documents\GitHub\-esp32_wip\Mav2PT_TwoWay_v2.46g\Mav2PT_TwoWay_v2.46g
Target Board is ESP32 / Variant is Wemos® LOLIN ESP32-WROOM-32
Ground Mode
Battery_mAh_Source = 3 - Define battery capacities in the LUA script
Using Serial_1 for S.Port
RSSI Automatic Select
Mavlink Serial In
Mavlink WiFi Out - Protocol is UDP
Waiting for telemetry
ESP32 S.Port pins inverted for Ground Mode
AutoBaud - Sensing FC_Mav_rxPin 25 
AP IP address: 192.168.4.1  SSID: Mav2Passthru
UDP started, listening on IP 192.168.4.1, UDP port 14555
Telem found at 57600 b/s
hb_count=1
hb_count=2
hb_count=3
mavgood=true
dhcps: send_offer>>udp_sendto result 0
Client connected: Remote UDP IP: 192.168.4.2  Remote  UDP port: 14550
```
So, I'm wondering what you mean by 

 **It happens immediately I connect to the WLAN**

Do you mean when you connect to the Mav2Passthrough access point? So instead of this 

```
 dhcps: send_offer>>udp_sendto result 0
Client connected: Remote UDP IP: 192.168.4.2  Remote  UDP port: 14550
```

you get this 

```
15:43:38.322 -> E (70986) wifi: pp.c 3470
15:43:38.322 ->
15:43:43.385 -> E (75985) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
15:43:43.385 -> E (75985) task_wdt: - IDLE0 (CPU 0)
15:43:43.385 -> E (75985) task_wdt: Tasks currently running:
15:43:43.385 -> E (75985) task_wdt: CPU 0: wifi
15:43:43.385 -> E (75985) task_wdt: CPU 1: IDLE1
15:43:43.385 -> E (75985) task_wdt: Aborting.
```
Remember, the ESP32 is running RTOS in the background. I looks like one of the multi-threaded tasks is problematic. I have not seen this before. I'll do some Googling later on

Seeelefant, 2019-12-03:
Hi Eric,

thank you for your effort. With "It happens immediately I connect to the WLAN" I mean I am trying to connect my Windows laptop WLAN to "your" hotspot 
[image]. I was also Googling a little for the watchdog problem--seems quite common actually-and found things like https://github.com/espressif/esp-idf/issues/1646 .
For the inverter problem this link https://quadmeup.com/smartport-inverter-for-f4-flight-controllers/ helped me. You can add this reference for the guys who do not want to wait for the delivery from Ali.

zs6buj, 2019-12-03:
Wolfgang, thank you for the information.

This issue of crashing has been on my mind and I'm wondering if you have a power supply problem. The board seems to crash when it fired up the wifi RF. USB 2 ports are rated at 500mA max, and judging from the discussion below, the ESP32 sees peak current draw of 700mA or more for short durations. Many USB ports do not supply the requisite full 5v, and current capability is well less than 500 mA. USB 3 (and 4) standards are higher.

I suggest you try to power the board from a proper 2A or great BEC and see what happens.

[image]

Seeelefant, 2019-12-04:
Hi Eric,
I attached a big capacitor to it and now it is working on with non-Windows systems the first time. After disconnecting and trying to connect again there are still some problems. I have still crashes when I try to connect with Windows.

zs6buj, 2019-12-05:
Ok, progress !

Try a second, smaller capacitor across the PSU.  In the 10nF to 100nF range. This filter higher frequencies and help with stray RF noise getting into the works. Mount the near the board if you ca, and keep wiring as short as possible.

zs6buj, 2019-12-10:
Wolfgang, since I haven't heard from you for a while I'll take that as a good sign and close this issue :)

closed 2019-12-10 by zs6buj