zs6buj/mav2pt #26: Declaration of Battery in Utilities and instantiation of bat1 and bat2

dragospascale, 2020-03-09:
Declaration of struct Battery {
  float    mAh;
  float    tot_mAh;
  float    avg_dA;
  float    avg_mV;
  uint32_t prv_millis;
  uint32_t tot_volts;      // sum of all samples
  uint32_t tot_mW;
  uint32_t samples;
  bool ft;
  }; has 9 members.
Instantiation of bat1 and bat2 
struct Battery bat1     = {
  0, 0, 0, 0, 0, 0, 0, true};   has only 8 values

Declaration of Battery should be done in config.h and instantiation of bat1 and bat2 in MavToPass_v2.56.4.ino otherwise you will get an undeclared bat1 in this scope.

Staring at line 1721 in MavToPass_v2.56.4.ino 
if (set.wfproto == tcp)  { // TCP  
          bool sentOK = Send_TCP(&R2Gmsg);  // to GCS
          #ifdef  Debug_GCS_Down
            Debug.print("Passed down from Ring buffer to GCS by WiFi TCP: sentOk="); Debug.println(sentOk);
You have bool sentOK (uppercase k) and then in Debug.println you have sentOk (lowercase k)

zs6buj, 2020-03-18:
Thanks for this feedback. I'll look at it more closely tomorrow.

zs6buj, 2020-03-20:
The modules definitely needed cleaning up. I moved all global variable declarations into global_variables.h. Including instantiation. 

global_variables.h is included at the top of the main module. Could you let me know if your compiler now approves.  :)

Which IDE are you using? If it's Arduino, then your version or settings are different from mine

closed 2020-03-21 by zs6buj