zs6buj/mav2pt #27: Platform IO, esp12e compilation fails.

giacomo892, 2020-03-27:
Plaftormio.ini :

```
[env:esp12e]  ; Node MCU variant
; Remember to set the variant in config.h
platform = espressif8266
board = esp12e
framework = arduino
upload_speed = 115200
monitor_speed = 115200
upload_port = COM(40)

```
config.h:

```
//================================================================================================= 
//================================================================================================= 
//
//                                    C O N F I G U R A T I O N 
// 
//================================================================================================= 
//================================================================================================= 

/*
Complete change log and debugging options are at the bottom of this tab
           
v2.58.4 2020-03-25 RPM fixed (library path).           
*/
//===========================================================================================
//
//                   PLEASE SELECT YOUR DEFAULT OPTIONS BELOW BEFORE COMPILING
//
//===========================================================================================

#define webSupport                      // ESP only. Enable wifi web support, including OTA firmware updating. Browse to IP.
#define webPassword      "changeme!"    // Web password 
//#define Reset_Web_Defaults            // Settings in eeprom. Do this if eeprom settings are corrupt or wrong.

//#define SD_Support                    // Activate if you have an SD card reader attached
#define OLED_Support                  // Activate if you have an OLED display attached

//#define AutoBaud                      // Auto detect Mavlink serial-in baud rate
                                        // NOTE: Set mvBaudFC = 57600 for Dragonlink and RFD900X
#define mvBaudFC               57600    // Mavlink to/from the flight controller - max 921600 - must match FC or long range radio
#define frBaud                 57600    // S.Port baud setting - default 57600 

// Do not enable for FlightDeck
#define PlusVersion  // Added support for 0x5009 Mission WPs, 0x50F1 Servo_Channels, 0x50F2 VFR_Hud

//=================================================================================================
//           D E F A U L T   T R A N S L A T I O N   M O D E   S E T T I N G S   
//=================================================================================================       
// Choose only one of these three translation modes
#define Ground_Mode          // Translator between Taranis et al and LRS transceiver (like Dragonlink, ULRS, RFD900...)
//#define Air_Mode             // Translator between FrSky receiver (like XRS) and Flight Controller (like Pixhawk)
//#define Relay_Mode           // Translator between LRS tranceiver (like Dragonlink) and FrSky receiver (like XRS) in relay box on the ground

//=================================================================================================
//           D E F A U L T   F L I G H T   C O M P U T E R    I / O   S E T T I N G S   
//=================================================================================================
// Choose only one of these default Flight-Controller-side I/O channels 
// How does Mavlink telemetry enter this translator?
#define FC_Mavlink_IO  0    // Serial Port (default)         
//#define FC_Mavlink_IO  1    // BlueTooth Classic - ESP32 only
//#define FC_Mavlink_IO  2    // WiFi - ESP32 or ESP8266 only
//#define FC_Mavlink_IO  3    // SD Card / TF - ESP32 only

//=================================================================================================
//           D E F A U L T   G R O U N D S T A T I O N    I / O   S E T T I N G S   
[... 1207 lines cut ...]
src/main.cpp:5688:17: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     set.btmode1 = "checked";
                 ^
src/main.cpp:5689:17: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     set.btmode2 = "";
                 ^
src/main.cpp:5692:17: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     set.btmode1 = "";
                 ^
src/main.cpp:5693:17: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     set.btmode2 = "checked";
                 ^
Compiling .pio/build/esp12e/libb8b/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
In file included from src/main.cpp:142:0:
include/config.h: At global scope:
include/config.h:571:23: warning: 'dt_tm' defined but not used [-Wunused-variable]
     static DateTime_t dt_tm; 
                       ^
*** [.pio/build/esp12e/src/main.cpp.o] Error 1
================================================ [FAILED] Took 7.62 seconds

```
If you need more infos, I'm here.

Thanks for taking a look

zs6buj, 2020-03-27:
Hello Giacomo

In the config.h

Find these lines:

```
//=================================================================================================                             
//                          S E L E C T   E S P   B O A R D   V A R I A N T   
//=================================================================================================

//#define ESP32_Variant     1    //  ESP32 Dev Module - Use Partition Scheme: "Minimal SPIFFS(1.9MB APP...)"
//#define ESP32_Variant     2    //  Wemos® LOLIN ESP32-WROOM-32_OLED_Dual_26p
//#define ESP32_Variant     3    //  Dragonlink V3 slim with internal ESP32 - contributed by Noircogi
#define ESP32_Variant     4    //  Heltec Wifi Kit 32 - Use Partition Scheme: "Minimal SPIFFS(Large APPS ith OTA)" - contributed by Noircogi

//#define ESP8266_Variant   1   // NodeMCU ESP 12F - choose "NodeMCU 1.0(ESP-12E)" board in the IDE
#define ESP8266_Variant   2   // ESP-F Use me for RFD900X TX-MOD - use generic ESP8266 board on IDE
```

Now for NodeMCU  ESP-12F change these two lines like this:

```
#define ESP8266_Variant   1   // NodeMCU ESP 12F - choose "NodeMCU 1.0(ESP-12E)" board in the IDE
//#define ESP8266_Variant   2   // ESP-F Use me for RFD900X TX-MOD - use generic ESP8266 board on IDE
```

giacomo892, 2020-03-27:
Thanks for the reply!

zs6buj, 2020-03-27:
Ok, there are a few things you need to do, because t

1) I updated the code on GitHub with a few exclusions because you don't need the SD functionality.

2) There are some files that come from the ESP32-IDF for the OTA functionality. See if you can install the ESP32-IDF.

[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/#step-2-get-esp-idf](url)

giacomo892, 2020-03-27:
Hi.
I've added esp-idf to my path when compiling within platformio:

```
echo $PATH
/Users/giacomo/esp-idf/components/esptool_py/esptool:/Users/giacomo/esp-idf/components/espcoredump:/Users/giacomo/esp-idf/components/partition_table/:/Users/giacomo/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:/Users/giacomo/.espressif/tools/xtensa-esp32s2-elf/esp-2019r2-8.2.0/xtensa-esp32s2-elf/bin:/Users/giacomo/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/giacomo/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/giacomo/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200309/openocd-esp32/bin:/Users/giacomo/.espressif/python_env/idf4.2_py2.7_env/bin:/Users/giacomo/esp-idf/tools:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
```
I still get the same errors are it wasn't there:

```
Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 2.4.0 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20603.200130 (2.6.3) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <My_Patched_ESP32_SD_Library>
|   |-- <SPI> 1.0
|-- <CircularBuffer> 1.2.0
|-- <SPI> 1.0
|-- <c_library_v2-master>
|-- <Wire> 1.0
|-- <EEPROM> 1.0
|-- <Adafruit SSD1306> 2.2.0
|   |-- <Adafruit GFX Library> 1.5.6
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <EspSoftwareSerial> 6.7.1
|-- <Update> 1.0
Building in release mode
Compiling .pio/build/esp12e/libaf9/My_Patched_ESP32_SD_Library/SD.cpp.o
Compiling .pio/build/esp12e/libaf9/My_Patched_ESP32_SD_Library/sd_diskio.cpp.o
lib/My_Patched_ESP32_SD_Library/src/SD.cpp:15:21: fatal error: vfs_api.h: No such file or directory

*****************************************************************
* Looking for vfs_api.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:vfs_api.h"
* Web  > https://platformio.org/lib/search?query=header:vfs_api.h
*
*****************************************************************

 #include "vfs_api.h"
                     ^
compilation terminated.
lib/My_Patched_ESP32_SD_Library/src/sd_diskio.cpp:16:24: fatal error: diskio.h: No such file or directory

****************************************************************
* Looking for diskio.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:diskio.h"
* Web  > https://platformio.org/lib/search?query=header:diskio.h
*
****************************************************************

     #include "diskio.h"
                        ^
compilation terminated.
Compiling .pio/build/esp12e/lib542/Update/Updater.cpp.o
*** [.pio/build/esp12e/libaf9/My_Patched_ESP32_SD_Library/SD.cpp.o] Error 1
Compiling .pio/build/esp12e/FrameworkArduino/Esp-frag.cpp.o
*** [.pio/build/esp12e/libaf9/My_Patched_ESP32_SD_Library/sd_diskio.cpp.o] Error 1
In file included from lib/Update/src/Updater.cpp:1:0:
lib/Update/src/Update.h:7:27: fatal error: esp_partition.h: No such file or directory

***********************************************************************
* Looking for esp_partition.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_partition.h"
* Web  > https://platformio.org/lib/search?query=header:esp_partition.h
*
***********************************************************************

 #include "esp_partition.h"
                           ^
compilation terminated.
*** [.pio/build/esp12e/lib542/Update/Updater.cpp.o] Error 1
================================================ [FAILED] Took 3.33 seconds =============================
```

giacomo892, 2020-03-31:
Just to keep you updated, i've tried your last commit and still I am not able to build the project.
Thanks for your work

zs6buj, 2020-03-31:
Hi Giacomo, I compiled a long reply when I posted the fix on GitHub, but I see it is missing from this thread. Strange. Anyway...

I have tightened up on the exclusions on SD functionality. The issue is that other people need the libraries, and PIO always compiles all libraries in the lib folder. So please use the latest version (2.58.5), but remove these libraries

My_Patched_ESP32_SD_Library - remove for esp8266 on PIO
Update - remove for esp8266 on PIO

The ESP8266 uses "Updater.h", and it is on the PIO path for the ESP8266 boards.

Hope this helps.

giacomo892, 2020-03-31:
Hi! Thanks!
It actually compiles now!
Uploading it and I get:

```
esptool.py --port /dev/tty.usbserial-A50285BI write_flash 0x0000 firmware.bin
esptool.py v2.5.1-dev
Serial port /dev/tty.usbserial-A50285BI
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: bc:dd:c2:25:aa:22
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 369440 bytes to 261347...
Wrote 369440 bytes (261347 compressed) at 0x00000000 in 36.6 seconds (effective 80.7 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
```
The board doesn't boot and I don't get any WiFi AP.
Do you have a clue?

zs6buj, 2020-03-31:
The binary image was completely flashed. Looks like a serial issue.  Try reducing upload speed. On the Tools pulldown.

zs6buj, 2020-03-31:
was NOT completely flashed.

giacomo892, 2020-03-31:
Thanks. Still no way I can finish the flash and I'm pretty sure this ESP works fine. 
I'll try again later.

zs6buj, 2020-03-31:
I should have said: In the platformio.ini

Try changing "upload_speed = 115200" to 57600

zs6buj, 2020-03-31:
Also make sure you are designating the correct COM

upload_port = COM(40)

Use Device Manager in Windows to check. 

(Excuse me if you know this already :)  )

giacomo892, 2020-03-31:
Picked another ESP from a box... and voilà it works!
Flashed and running. 
Do you have a connection diagram for the ESP12E handy? Basically where do I have to connect passthrough out.. since mavlink in goes into the only serial available.
Then i'll take care of updating the wiki, since lot of stuff have been discussed here

zs6buj, 2020-03-31:
It's the same as the 12F

```
#if (ESP8266_Variant == 1)        // NodeMCU 12F board
  
    #define MavStatusLed  D4        // D4 Board LED - Mav Status LED inverted logic - use 99 while debug
    #define InvertMavLed true      
    #define BufStatusLed  99        // None     
   //                     D4        // TXD1 - Serial1 debug log out SHARED WITH BOARD LED                         
    #define FC_Mav_rxPin  D9        // RXD0 default  
    #define FC_Mav_txPin  D10       // TXD0 default    
    #define Fr_rxPin      D5        // GP10 SPort - Not used in single wire mode
    #define Fr_txPin      D6        // GPIO SPort - Use me in single wire mode
    #if (defined SD_Support) || (defined OLED_Support)  
      #define SCL           D1        // I2C OLED board   
      #define SDA           D2        // I2C OLED board
      #define i2cAddr      0x3C       // I2C OLED board
    #endif     
    int16_t wifi_rssi;   
    uint8_t startWiFiPin = D3;      
    uint8_t WiFiPinState = 0;
  #endif
  
  #if (ESP8266_Variant == 2)        // ESP-F RFD900X TX-MOD - use generic esp8266

    //                         GPIO as per node mcu
    static const uint8_t D0   = 16;   // SCL - optional
    static const uint8_t D1   = 5;    // SDA - optional
    static const uint8_t D2   = 4;    // SPort half-duplex
    static const uint8_t D3   = 0;    // Flash
    static const uint8_t D4   = 2;    // BoardLED & TXD1 optional debug out
    static const uint8_t D5   = 14;   // SPort rx (unused in half-duplex)
    static const uint8_t D6   = 12;   // P2-3 exposed dual row of pins
    static const uint8_t D7   = 13;   // CTS
    static const uint8_t D8   = 15;   // RTS
    static const uint8_t D9   = 3;    // RXD0
    static const uint8_t D10  = 1;    // TXD0
    
    #define MavStatusLed  D4        // D4 Board LED - Mav Status LED inverted logic - use 99 while debug
    #define InvertMavLed true    
    #define BufStatusLed  99        // None
    //                    D4        // TXD1 - Serial1 default debug log out SHARED WITH BOARD LED                           
    #define FC_Mav_rxPin  D9        // RXD0 default  
    #define FC_Mav_txPin  D10       // TXD0 default    
    #define Fr_rxPin      D5        // SPort - Not used in single wire mode
    #define Fr_txPin      D2        // SPort half-duplex inverted - Use me in single wire mode
    #if (defined SD_Support) || (defined OLED_Support)
      #define SCL           D0        // I2C OLED board   
      #define SDA           D1        // I2C OLED board
      #define i2cAddr      0x3C       // I2C OLED board
    #endif       
    int16_t wifi_rssi;   
    uint8_t startWiFiPin = D8;      
    uint8_t WiFiPinState = 0;
  #endif  
```

zs6buj, 2020-03-31:
[image]

zs6buj, 2020-03-31:
static const uint8_t D2   = 4;    // = gpio4 SPort half-duplex

giacomo892, 2020-03-31:
I'm using a barebone ESP12F now:
[image]

I'll use MavlinkToPassthru with QLRS. Anything I should be aware of in the configuration?

zs6buj, 2020-03-31:
I don't know much about QLRS. Check to see that RSSI is coming through, otherwise the LUA script will not connect.

zs6buj, 2020-03-31:
Thanks for offering to update the wiki with this info :)

giacomo892, 2020-03-31:
Those the messages sent out by the QRLS transmitter on the serial port:

MAVLINK_MSG_ID_RC_CHANNELS_RAW
MAVLINK_MSG_ID_HEARTBEAT
MAVLINK_MSG_ID_SYS_STATUS
MAVLINK_MSG_ID_ATTITUDE
MAVLINK_MSG_ID_GLOBAL_POSITION_INT
MAVLINK_MSG_ID_GPS_RAW_INT
MAVLINK_MSG_ID_VFR_HUD
MAVLINK_MSG_ID_GPS_GLOBAL_ORIGIN
MAVLINK_MSG_ID_HOME_POSITION
MAVLINK_MSG_ID_COMMAND_INT
MAVLINK_MSG_ID_RADIO_STATUS

I guess I can get the RSSI from a channel. Have I to configure which channel to use?

zs6buj, 2020-03-31:
Yes, in Mission Planner  parameters, rssi parameter

giacomo892, 2020-03-31:
All good then, it is already set. Working fine on the OSD :)

zs6buj, 2020-03-31:
Cool

giacomo892, 2020-03-31:
I'm still confused on which pin to use for the passthrough to the radio.
[image]
Found this on your repo and seems like it doesn't match what have you said here
can you clarify?

zs6buj, 2020-03-31:
IO4

If you look at the code, your board will be the same as this variant.:
```
 #if (ESP8266_Variant == 2)        // ESP-F RFD900X TX-MOD - use generic esp8266

    //                         GPIO as per node mcu
    static const uint8_t D0   = 16;   // SCL - optional
    static const uint8_t D1   = 5;    // SDA - optional
    static const uint8_t D2   = 4;    // SPort half-duplex
    static const uint8_t D3   = 0;    // Flash
    static const uint8_t D4   = 2;    // BoardLED & TXD1 optional debug out
    static const uint8_t D5   = 14;   // SPort rx (unused in half-duplex)
    static const uint8_t D6   = 12;   // P2-3 exposed dual row of pins
    static const uint8_t D7   = 13;   // CTS
    static const uint8_t D8   = 15;   // RTS
    static const uint8_t D9   = 3;    // RXD0
    static const uint8_t D10  = 1;    // TXD0
    
    #define MavStatusLed  D4        // D4 Board LED - Mav Status LED inverted logic - use 99 while debug
    #define InvertMavLed true    
    #define BufStatusLed  99        // None
    //                    D4        // TXD1 - Serial1 default debug log out SHARED WITH BOARD LED                           
    #define FC_Mav_rxPin  D9        // RXD0 default  
    #define FC_Mav_txPin  D10       // TXD0 default    
    #define Fr_rxPin      D5        // SPort - Not used in single wire mode
    #define Fr_txPin      D2        // SPort half-duplex inverted - Use me in single wire mode
    #if (defined SD_Support) || (defined OLED_Support)
      #define SCL           D0        // I2C OLED board   
      #define SDA           D1        // I2C OLED board
      #define i2cAddr      0x3C       // I2C OLED board
    #endif       
    int16_t wifi_rssi;   
    uint8_t startWiFiPin = D8;      
    uint8_t WiFiPinState = 0;
  #endif  
```

giacomo892, 2020-03-31:
Oh. So for ESP12F I have to build the firmware using variant 2 :) 
Asking questions so the wiki will be more precise. ESP12F boards are so cheap and useful!

zs6buj, 2020-03-31:
You can select it in the config.h here:

```
//=================================================================================================                             
//                          S E L E C T   E S P   B O A R D   V A R I A N T   
//=================================================================================================

//#define ESP32_Variant     1    //  ESP32 Dev Module - Use Partition Scheme: "Minimal SPIFFS(1.9MB APP...)"
//#define ESP32_Variant     2    //  Wemos® LOLIN ESP32-WROOM-32_OLED_Dual_26p
//#define ESP32_Variant     3    //  Dragonlink V3 slim with internal ESP32 - contributed by Noircogi
#define ESP32_Variant     4    //  Heltec Wifi Kit 32 - Use Partition Scheme: "Minimal SPIFFS(Large APPS ith OTA)" - contributed by Noircogi

//#define ESP8266_Variant   1   // NodeMCU ESP 12F - choose "NodeMCU 1.0(ESP-12E)" board in the IDE
#define ESP8266_Variant   2   // ESP-F Use me for RFD900X TX-MOD - use generic ESP8266 board on IDE
```
Then the LED and everything will work.

Or copy this variant, call it ESP8266_Variant   3, and name it  Giacomo's 12-E blah blah

giacomo892, 2020-03-31:
:) Just compiled the variant 2 and flashed it via OTA.
Later will try the soldering job and i'll report if the telemetry works
many thanks for the moment

zs6buj, 2020-03-31:
glad to help

giacomo892, 2020-03-31:
I got it working. I can use QGC on the desktop and YAAPU on the radio.

Only thing I have an issue with RSSI. It shows a not correct one. Do you have an idea?

giacomo892, 2020-04-01:
I don't have write access to wiki. can you amend that so I can add a little bit of guide?

zs6buj, 2020-04-01:
Ok, I have amended the edit restriction on the WiKi for you. Thanks.

Is your rssi coming out at 59% See this patch by Hasi123. It basically inserts a fixed rssi if no other rssi is received.

`v2.51 2020-01-18 Make default rssi 69% for SiK radios, like RFD900x. PR by Hasi123 ` 

Most long-range radios use the SiK firmware and  insert a Mavlink message type #109 into the telemetry stream. However, there are two other RSSI record types, #35 and #65. Look them up [here](url). Use Ctl/F and search for #35 etc.

You can debug RSSI by enabling #define Debug_Rssi in config.h, however your 12E barebones needs a trick to view serial debug messages. Select the debug port:

[image]

and connect a USB/TTL bridge RX to IO2 / TxD1 on the 12E, and also ground, obviously. Then use Putty or any other serial terminal to view the debug output.

giacomo892, 2020-04-01:
I've sorted it out. Check the PR i've made and if you want you can integrate those changes in the next version.

zs6buj, 2020-04-01:
Hey looks good! See my comment, if you want to tag your lines?

giacomo892, 2020-04-01:
https://github.com/zs6buj/MavlinkToPassthru/wiki/ESP12F-and-PlatformIO
Made a first stub. Can you review it?

zs6buj, 2020-04-01:
Eccellente amico mio

zs6buj, 2020-04-01:
With your permission I will make a small change to your PR and publish to v2.57. I must include your changes in the Arduino code and I also want to publish some style and functional improvements to the web interface. Probably tomorrow.

giacomo892, 2020-04-01:
You are the project owner :) Do whatever it takes to make this firmware even better!

Thanks for your support and I guess we can close the issue now.

closed 2020-04-02 by zs6buj