zs6buj/mav2pt #31: configuration problem

Mogway28, 2020-05-27:
Eric, nice piece of work. Could You help with Your code. I have transceiver box with esp d1 mini and now it's working like uart to wifi passthrough. I want to use Your code to  get Yaapu work but when I upload it with proper configuration it won't change AP ssid  or any AP settings.
I've copy all project to Platformio and sets config  
ESP8266_Variant   1
WiFi_Mode   1

it's compile and it seems to  uploading  but AP setting are not update , also debug mode #define Debug_Web_Settings won't work 
on com is:
load 0x4010f000, len 3456, room 16 
                                     
tail 0
                                                                        
chksum 0x84
                                                                   
csum 0x84
                                                                     
va5432625
                                                                     
~ld

zs6buj, 2020-05-27:
Ok, no problem   I'll be able to look at it tomorrow.🙂

Mogway28, 2020-05-28:
Many thanks,

Debug mode show me

/Debug Read WiFi Settings from EEPROM:
validity_check = DC
translator mode = 3
fc_io = 0
gcs_io = 2
gcs_sd = 0
wifi mode = 1
wifi protocol = 1
baud = 57600
wifi channel = 9
apSSID = testowe
apPw = password
staSSID = OmegaOffice
staPw = changeme
Host = test
tcp_localPort = 9876
udp_localPort = 14555
udp_remotePort = 14550
bt mode = 2
SlaveConnect To Name = Crossfire 0277

Target Board is ESP8266 / Variant is Lonlin Node MCU 12F
Relay Mode
Battery_mAh_Source = 3 - Define battery capacities in the LUA script
RSSI Automatic Select
Mavlink Serial In
Mavlink WiFi Out - Protocol is TCP/IP
ISR not in IRAM!

User exception (panic/abort/assert)
Abort called/

But physically (or mentally :) wifi settings is like it was sets in 
previously uploaded software (different one)

Best Regards,

Mariusz

W dniu 27.05.2020 o 21:47, Eric Stockenstrom pisze:

Hasi123, 2020-05-28:
Try in config.h

#define Reset_Web_Defaults

zs6buj, 2020-05-28:
Ok, I found out the WEMOS ESP-12F Mini looks like this:

[image]

and the Arduino pin map looks like this:

[image]

So I created a new ESP8266 variant for you to try:

```
  #if (ESP8266_Variant == 3)   // ESP-12F, WEMOS D1 Mini - use Generic ESP8266 on Arduino IDE
    //  Pin Map as per C:\Users\<user>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\variants\d1_mini\pins_arduino.h

    #define LED_BUILTIN 2

    static const uint8_t D0   = 16;  // unused
    static const uint8_t D1   = 5;   // SCL - optional
    static const uint8_t D2   = 4;   // SDA - optional
    static const uint8_t D3   = 0;   // Flash - reserved
    static const uint8_t D4   = 2;   // LED_BUILTIN & TXD1 optional debug out
    static const uint8_t D5   = 14;  // SPort rx (unused in half-duplex)
    static const uint8_t D6   = 12;  // SPort tx - Use me in single wire mode
    static const uint8_t D7   = 13;  // CTS 
    static const uint8_t D8   = 15;  // RTS
    static const uint8_t D9   = 3;   // RXD0
    static const uint8_t D10  = 1;   // TCD0 
    
    #define MavStatusLed  D4        // D4 Board LED - Mav Status LED inverted logic - use 99 while debug
    #define InvertMavLed true    
    #define BufStatusLed  99        // None
    //                    D4        // TXD1 - Serial1 default debug log out SHARED WITH LED_BUILTIN BOARD LED                           
    #define FC_Mav_rxPin  D9        // RXD0 - Serial(0) 
    #define FC_Mav_txPin  D10       // TXD0 - Serial(0) 
    #define Fr_rxPin      D5        // SPort - Not used in single wire mode
    #define Fr_txPin      D6        // SPort - inverted - Use me in single wire mode
    #define startWiFiPin  D16       // Trigger WiFi startup       
    #if (defined SD_Support) || (defined OLED_Support)
      #define SCL           D1        // I2C OLED board   
      #define SDA           D2        // I2C OLED board
      #define i2cAddr      0x3C       // I2C OLED board
    #endif 
 
  #endif // end of this ESP8266 variant 
```

You might also need to #define Reset_Web_Defaults, but do this only 1 time.

Try v2.60.3 in PlatformIO folder
Let me know if you make progress

Mogway28, 2020-05-28:
There is some compile errors. Same error when I choose  ESP8266_Variant2. Only  ESP8266_Variant 1 can be compile 

Compiling .pio\build\esp12e\src\main.cpp.o
In file included from src\main.cpp:145:0:
include/config.h:463:26: error: redefinition of 'const uint8_t D0'
     static const uint8_t D0   = 16;  // unused
                          ^
In file included from C:\Users\Mariusz\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:296:0,
                 from src\main.cpp:1:
C:\Users\Mariusz\.platformio\packages\framework-arduinoespressif8266\variants\nodemcu/pins_arduino.h:40:22: error: 'const uint8_t D0' previously defined here
 static const uint8_t D0   = 16;
                      ^
In file included from src\main.cpp:145:0:
include/config.h:464:26: error: redefinition of 'const uint8_t D1'
     static const uint8_t D1   = 5;   // SCL - optional

zs6buj, 2020-05-28:
Did you use my platformio.ini ?

zs6buj, 2020-05-28:
Simply open the whole folder in VsCode, then build

Mogway28, 2020-05-29:
Yes,
I've just copy it and configure . I've checked and only ESP8266_Variant 1 (for esp8266) can be compile - else there is seme error:
error: 'const uint8_t D0' previously defined here
static const uint8_t D0 = 16;
^
and it is for all D* variable

zs6buj, 2020-05-29:
"and configure"

Could you explain what you configured?  :)

It's compiling clean here

[image]

[image]

zs6buj, 2020-05-29:
[image]

Mogway28, 2020-05-29:
OK, probably Is something  with board configuration and and platformio ini.
When i live platformio.ini  like You describe it compile but it can't connect to board. When i change to [env:esp12e] it compile only in variant 1 and can upload to board

zs6buj, 2020-05-29:
It is already inverted. I picked D6 but can't test as I don't have a board.

Mogway28, 2020-05-29:
Status for now is : (board is D1mini 4MB)
1.can compile Variant 3 on platformio and env:esp01_1m but can't upload
2.uploading via nodemcu (firmware.bin from platfomio)
3.S.port won't work

will try with other D* and let You know

zs6buj, 2020-05-29:
No, you must stick with the generic esp01_1m because it does not map the pins. I map them in variant 3.

Could you show me your platformio.ini for modemcu

Mogway28, 2020-05-29:
ad2.  *nodemcu firmware

I've tryed  those boards  and all works with variant 1 (uploading)

[env:d1_mini]
platform = espressif8266
board = d1_mini

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2

[env:esp12e]  
platform = espressif8266
board = esp12e

Everywhere is
framework = arduino
upload_speed = 115200
monitor_speed = 115200

zs6buj, 2020-05-29:
Did you change the COM(40) to your usb upload com? Look in device manager.

platform = espressif8266
board = esp01_1m
framework = arduino
upload_speed = 115200
monitor_speed = 115200
upload_port = COM(40)

Mogway28, 2020-06-01:
Yes, Com port is found but can't connect. But I can manage to upload compile *.bin via ESP8266Flasher . Everything work except SPort via D6 - Yappu script do not receive any telemetry.

zs6buj, 2020-06-01:
Could you try a different pin for sport. Just change in config.h variant 3 fir esp8266

Mogway28, 2020-06-01:
changed to Every available pins - same behavior.
#define ESP_One Should be defined in this case ?
Could You point me hardware which works - maybe will be better when I switch to it.

zs6buj, 2020-06-01:
Progress!  Good work. 

No,  do NOT enable ESP_Onewire. It is experimental for now.

I doubt that the board is your problem. Have you setup for debug text out on serial1?

Can you show me the monitor output text?

Are you taking S.Port pin straight to bottom pin on the Taranis? You can put a 4.7k resistor there for safety, but it's not essential.

Don't forget the ground wire to the next pin up on the Taranis.

If you have another board, I can give you a tool which reads SPort and prints.

Mogway28, 2020-06-01:
[image]

zs6buj, 2020-06-01:
It should be inverted. Let me check why.

zs6buj, 2020-06-01:
You have selected Relay Mode".

My understanding is you need Ground Mode.

Mogway28, 2020-06-01:
Yes,
Because like You described in config.h I should use relay mode. On the
gound I have rfd connected to ESP8266 via uart and same rfd connected to RX
frsky (Sbus).
Taranis is connected to via 2,4GHz to RX. Mission Planner is connected to
mavlink though ESP to RFD

zs6buj, 2020-06-01:
"On the
gound I have rfd connected to ESP8266 via uart and same rfd connected to RX
frsky (Sbus)."

Don't you mean ESP8266 connected to receiver SPort?

In that case, yes, relay is correct, but now you need a bi-directional two-wire to 2-wire converter+ inverter.

The reason is the ESP cannot (yet) do 1-wire properly. I can give you some links

zs6buj, 2020-06-01:
https://drive.google.com/file/d/1OVyoxh7nvX6gQY1c9NJ2G9jQsqtVGrak/view?usp=drivesdk

zs6buj, 2020-06-01:
https://drive.google.com/file/d/0BzoCg38LmT22bXFjTVVlMkJJWkk/view?usp=drivesdk

zs6buj, 2020-06-01:
https://photos.app.goo.gl/bqD9noDPijEaPzek7

zs6buj, 2020-06-01:
FrSky Yaapu Telemetry Converter Cable Pixhawk to Taranis X9D Plus QX7 Jumper T16 Smart Port R9 Slim+ R9 X8R XSR R9M X4R Receiver - AliExpress Mobile

Mogway28, 2020-06-02:
I've made converter on MAX3232 - and it works :) 
Thank You very much for your time and again nice piece of code

zs6buj, 2020-06-02:
Glad it's all working. Well done yourself.

closed 2020-06-02 by zs6buj