zs6buj/mav2pt #33: ESP12F Variant 1 code error

drnayeem, 2020-06-05:
The Variant 1 option compiler on this board returns with an error 
of Undeclared Variable 
in line 425 ...

- ``

zs6buj, 2020-06-06:
More information please

drnayeem, 2020-06-09:
@zs6buj
Thanks for the reply  
Am trying to compile the firmware for ESP 12F
========================
//================================================================================================= 
//================================================================================================= 
//
//                                    C O N F I G U R A T I O N 
// 

/*
=================================================================================================== 
                             M o s t    R e c e n t   C h a n g e s
=================================================================================================== 

Complete change log and debugging options are at the bottom of this tab
                                    
v2.60.3 2020-05-25 Added new variant for ESP8266 - ESP-12F WEMOS D1 Mini    
        2020-05-30 Added some support for PitLab flight stack 
v2.60.4 2020-06-01 Half-duplex Mavlite on S.Port - phase 1                                                                
*/

//===========================================================================================
//
//                   PLEASE SELECT YOUR DEFAULT OPTIONS BELOW BEFORE COMPILING
//
//===========================================================================================

#define webSupport                      // ESP only. Enable wifi web support, including OTA firmware updating. Browse to IP.
#define webPassword      "changeme!"    // Web password 
//#define Reset_Web_Defaults            // Reset settings in eeprom. Do this if eeprom settings are corrupt or wrong.

//#define SD_Support                    // Enable if you have an SD card reader attached
#define OLED_Support                  // Enable if you have an OLED display attached

//#define AutoBaud                      // Auto detect Mavlink serial-in baud rate
                                        // NOTE: Set mvBaudFC = 57600 for Dragonlink and RFD900X
#define mvBaudFC               57600    // Mavlink to/from the flight controller - max 921600 - must match FC or long range radio
#define frBaud                 57600    // S.Port baud setting - default 57600 

// Do not enable for FlightDeck
#define PlusVersion  // Added support for 0x5009 Mission WPs, 0x50F1 Servo_Channels, 0x50F2 VFR_Hud

//=================================================================================================
//           D E F A U L T   T R A N S L A T I O N   M O D E   S E T T I N G S   
//=================================================================================================       
// Choose only one of these three translation modes
//#define Ground_Mode          // Translator between Taranis et al and LRS transceiver (like Dragonlink, ULRS, RFD900...)
#define Air_Mode             // Translator between FrSky receiver (like XRS) and Flight Controller (like Pixhawk)
//#define Relay_Mode           // Translator between LRS tranceiver (like Dragonlink) and FrSky receiver (like XRS) in relay box on the ground

//=================================================================================================
//           D E F A U L T   F L I G H T   C O M P U T E R    I / O   S E T T I N G S   
//=================================================================================================
// Choose only one of these default Flight-Controller-side I/O channels 
// How does Mavlink telemetry enter this translator?
#define FC_Mavlink_IO  0    // Serial Port (default)         
//#define FC_Mavlink_IO  1    // BlueTooth Classic - ESP32 only
//#define FC_Mavlink_IO  2    // WiFi - ESP32 or ESP8266 only
//#define FC_Mavlink_IO  3    // SD Card / TF - ESP32 only

//=================================================================================================
//           D E F A U L T   G R O U N D S T A T I O N    I / O   S E T T I N G S   
//=================================================================================================
// Choose only one of these default GCS-side I/O channels
// How does Mavlink telemetry leave this translator?
// These are optional, and in addition to the S.Port telemetry output
//#define GCS_Mavlink_IO  0    // Serial Port  - Only Teensy 3.x and Maple Mini have Serial3     
//#define GCS_Mavlink_IO  1    // BlueTooth Classic - ESP32 only
#define GCS_Mavlink_IO  2    // WiFi - ESP32 or ESP8266 only - auto selects on ESP8266
[... 1142 lines cut ...]
                  ^
src\main.cpp:6025:18: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     set.wfproto2 = "checked";
                  ^
src\main.cpp:6029:17: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     set.btmode1 = "checked";
                 ^
src\main.cpp:6030:17: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     set.btmode2 = "";
                 ^
src\main.cpp:6033:17: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     set.btmode1 = "";
                 ^
src\main.cpp:6034:17: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     set.btmode2 = "checked";
                 ^
*** [.pio\build\esp01_1m\src\main.cpp.o] Error 1
==================================================== [FAILED] Took 9.74 seconds ====================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
===========================

Kindly Help 
Regards

zs6buj, 2020-06-09:
You have selected the NodeMCU board in MavToPass config.h, but your PlatformIO.ini has selected generic ESP8266, so the pins are defined twice and there is a conflict.

```
#define ESP8266_Variant 1 // NodeMCU ESP 12F - choose "NodeMCU 1.0(ESP-12E)" board in the IDE
//#define ESP8266_Variant 2 // ESP-12E, ESP-F barebones boards. RFD900X TX-MOD, QLRS et al - use Generic ESP8266 on IDE
```
Rather try this:

```
//#define ESP8266_Variant 1 // NodeMCU ESP 12F - choose "NodeMCU 1.0(ESP-12E)" board in the IDE
#define ESP8266_Variant 2 // ESP-12E, ESP-F barebones boards. RFD900X TX-MOD, QLRS et al - use Generic ESP8266 on IDE
```

closed 2020-07-08 by zs6buj