zs6buj/mav2pt #34: WeMos D1 Mini no S.port telemetry

hunterbeach, 2020-07-07:
I was able to flash the device with ESP8266 variant 2 in air mode with OneWire disabled and I could connect via UDP but I wasn't receiving anything on my Jumper T16 with the Yaapu scripts. I was receiving the normal telemetry such as RSSI but nothing more. In the debug console I was getting

`S.PORT NOT INVERTED! Hw inverter to 1-wire required. S.Port on ESP is`

and that's all.

I enabled OneWire after looking at the code and this is the debug log.

`Starting main .....
EEPROM initialised successfully
Target Board is ESP8266 / Variant is ESP-F - RFD900X TX-MOD
Air Mode
Battery_mAh_Source = 3 - Define battery capacities in the LUA script
RSSI Automatic Select
Mavlink Serial In
Mavlink WiFi Out - Protocol is UDP
No Bluetooth options selected, BT support not compiled in
Mavlink s
`

Though sometimes it will reset itself and send this instead

`Starting main .....
EEPROM initialised successfully
Target Board is ESP8266 / Variant is ESP-F - RFD900X TX-MOD
Air Mode
Battery_mAh_Source = 3 - Define battery capacities in the LUA script
RSSI Automatic Select
Mavlink Serial In
Mavlink WiFi Out - Protocol is UDP
No Bluetooth options selected, BT support not compiled in
Mavlink 3
`

With OneWire enabled I also can't seem to connect by UDP anymore. I also tried touching the S.port output from the D1 mini to the uninverted S.port pad on the RSXR both with and without OneWire enabled to no effect. If you have any idea why OneWire might not be working so that I don't have to get a conversion board, that would be great.

zs6buj, 2020-07-08:
hunterbeach:

OneWire is listed under "Experimental", and unfortunately the class libraries do not work for me. Yet. I'm hoping that we will eventually get OneWire working in this application but I don't have much time to spend on this right now.

So you should not enable OneWire, and must use a two-wire to one-wire bi-directional (half-duplex) inverter. The wiki talks about this a bit, but Google should also help.

In Air mode,  the MavToPass translator must READ the S.Port (on the X receiver) in order to find the appropriate point at which to inject a packet (WRITE), so half-duplex is necessary.

hunterbeach, 2020-07-08:
Is it not possible to READ into the ESP from the normal S.port pin and WRITE to the separate uninverted S.port pin on the R-XSR or vice versa? The ESP8266 can do both inverted and non-inverted from what I understand so it should be able to communicate with both pins at the same time to read and write simultaneously.

hunterbeach, 2020-07-08:
I tried to find on Google the difference between one-wire and two-wire but couldn't find much. 

What's the status of the blind injection experimental option? If the packets don't have to be interleaved then it shouldn't be necessary to have bidirectional communication.

Lastly, I assume ground mode won't work since the only module I have in the T16 is the internal module.

If it is indeed the case that the ESP8266 requires an external converter to work, I would suggest changing the following on the home page of the Wiki under ESP8266 to make it more clear.

> Nevertheless, they perform more than adequately using SoftwareSerial, with the added advantage that single wire inverted S.Port is supported, so no external converter is required.

zs6buj, 2020-07-08:
`Nevertheless, they perform more than adequately using SoftwareSerial, with the added advantage that single wire inverted S.Port is supported, so no external converter is required.`

Yes, I should correct that statement. At the time the experiments were looking good and I had time on my hands.

You can try blind injection. It worked when I tested it. It only works because asynch packets on the S.Port have relatively huge spaces of time between them. If you throw a packet in at an arbitrary time, chances are you will miss an outgoing packet, and then just keep sending with a regular fixed period into the gap. You still need to invert the ESP output pin, OR use the "uninverted" R-XSR pin.

Ground Mode works in simplex mode (one way) since no interleaving is necessary into the back of the Taranis et al. I can't say for sure with the Jumper T16, but if you are using OpenTX I'm guessing it will work. Actually, Alex has one, and my impression is that it works.

zs6buj, 2020-07-08:
(https://github.com/zs6buj/MavlinkToPassthru/wiki/When-Do-I-Need-and-Inverter-and-or-Converter%3F)

hunterbeach, 2020-07-08:
I will give blind injection a try and report back.

The reason I assumed ground mode wouldn't work on the T16 is because the default module is internal and from looking at this picture from [here](https://discuss.ardupilot.org/t/rfd900x-txmod-jumpert16-f405wing-in-arduboat/47208/3) it looks like the Mav2PT needs to sit between the module and OpenTX to work.

[image]

hunterbeach, 2020-07-08:
Also, you said 

> You still need to invert the ESP output pin, OR use the "uninverted" R-XSR pin.

But I thought the ESP8266 has the ability to invert by itself, just not convert to OneWire? From the wiki page you linked:

> The ESP32 variants have the capability of inverting the signal, but not converting to single wire (as far as I know at this point). So we must still do the two-wire to single-wire conversion when talking to an XSR, because we need to slot in our packets

If the ESP32 varients have the capability of inverting the signal, shouldn't the ESP8266?

zs6buj, 2020-07-08:
0) HardwareSerial on the ESPs does not support serial pin inversion. At least I can't find any examples of it being done. The Teensy 3.x, on the other hand can, and does.

1) SoftwareSerial supports pin inversion, as the pulses are bit-banged.  They don't use a hardware UART.

2) The ESP32 and the ESP8266 are both capable of using SoftwareSerial. 

3) The ESP8266 does not have enough UARTs, so MUST always use SoftwareSerial in this application.

4) The ESP32 has 3 UARTs, so default is HardwareSerial, but SoftwareSerial is an OPTION. With SoftwareSerial selected, it can do serial pin inversion.

I hope that makes it clearer for you. :)

hunterbeach, 2020-07-08:
Ah, alright. So SoftwareSerial needs to be enabled when it's compiled so that it can invert it.

I think I'm just going to get a Mavlink to S.port converter to go directly from the Pixhawk to the R-XSR though. None of the experimental options seemed to change anything and now I can't even get a UDP connection to work. It seems every time there's a different problem like having to reboot to get it to connect and go in STA mode instead of falling back to AP mode or the web username being case sensitive or not being able to connect to the AP even though the password was 100% correct. I've spent probably cumulative 10 hours trying to get it to work so I think just getting a Mavlink to S.port converter will be easier than trying to implement ground mode on the T16 haha

closed 2020-08-06 by zs6buj