zs6buj/mav2pt #36: "No web support!" on OLED despite web support enabled in config

aaaaaaudrey, 2020-07-10:
Hardware is Heltec Wifi Kit 32

Exact steps taken:

1. download mav2pass 2.61.0 as zip file
2. copy libraries folder to arduino libraries folder
3. open MavToPass_v2.61.0.ino with arduino ide
4. comment out both Ground_Mode and Air_Mode #defines and uncomment Relay_Mode define
5. check that webSupport define is uncommented
6. flash to board

[image]

[image]

Commenting out the webSupport define makes a wifi network appear, but breaks the oled screen, removing all newlines. 

[image]

zs6buj, 2020-07-10:
Uncomment #define Reset_Web_Defaults

The first time you fire the board up after the first flash, all the default settings are read into eeprom. After that, if you need to change them, do it through the web interface, or use the option described above to reset them.

aaaaaaudrey, 2020-07-10:
[image]

I tried the original, default password and password "12345678" but I get this error no matter what I do.

aaaaaaudrey, 2020-07-10:
Reflashed 3x more, and now my android will connect with the password "password" but my windows 10 machines all will not.

zs6buj, 2020-07-11:
Can't help you. Sorry.

aaaaaaudrey, 2020-07-11:
I'm certain that this isn't an issue with m2p. If it works with macs and android it should work with windows, and nobody else has this issue. Thank you for the help with getting it working :)

closed 2020-07-11 by aaaaaaudrey