zs6buj/mav2pt #38: Flight Mode Confliction with multiple MAV IDs

mike-bfd, 2020-09-01:
When attaching a Gremsy gimbal to the pixhawk via Mavlink, I noticed that the flight mode constantly changes on my Yaapu Script. When this gimbal is connected to the aircraft, it appears as a second vehicle, with it's own data stream. (In mission planner, you can select the aircraft or the gimbal in the vehicle drop-down next to the connect button). I originally posted this to Yaapu, but it seems that the error has to do with the underlying MavToPassthrough script. This is with using the RFDesign TX MOD, and I'm not sure if they are using a fork of your script or not, but I figure I'd post here just in case. 

I have attached a video and tlog to hopefully help explain.
https://drive.google.com/drive/folders/1ZPfU7rXaT2E5CgQCuOD5_RAK1FszSysk?usp=sharing

The architecture:
Cube Orange FC:
Air Radio: RFD900ux
Ground Radio: RFD900 TX MOD - Wifi (udp) to Mission Planner.

zs6buj, 2020-09-02:
Hi Mike, 
I guess I know what's happening. The firmware (mav2pt) gets MAV_TYPE regularly for any heartbeat on the wire (or air).  For mav_type we determine how to deal with this type of craft. For example, at multirotor will have different modes than a rover. When we get a heartbeat from unknown new type, like your gimbal apparently is, it should be excluded for our purposes.

Around line 1620 of the main module, you will find this line:

```
          if (ap_type_tmp == 5 || ap_type_tmp == 6 || ap_type_tmp == 18 || ap_type_tmp == 27) break;      
          // Ignore heartbeats from GCS (6) or Ant Trackers(5) or Onboard_Controllers(18) or ADSB (27))
```

We need to determine the mav_type on the heartbeats from your gimbal, and exclude them here.

If you could give me a T_Log I could do it for you.

The guys at rfd900 / txmod would have to make the mod (lol) to their fork.

mike-bfd, 2020-09-02:
Thanks! I have a tlog in the drive folder in the original post. I'll follow up with Gremsy to see if all of their gimbal products have the same mav type, and point RFDesign to this post too

mike-bfd, 2020-09-02:
Looks like this gimbal is type 26
[image]

kentmartin99, 2020-09-03:
Hi Eric, I'm not sure what you mean by  rfd900/txmod would have to make this change in their fork?
The RFD900 does not do anything to mav packets or ignore any of them. It is a straight pass through of mav data with the addition of RSSI information only(as a separate packet). The RFD900 does not use mav to pass through code, ( for that matter the only message we look for is a heartbeat, and we still pass it through)
Why would we start ignoring data as we are a transport layer?

mike-bfd, 2020-09-03:
@kentmartin99 This is in reference to the TX MOD's new feature that is within the ESP8266 inside the transmitter module. The new feature allows for telemetry data to be interpreted on its way to the GCS, and translated by the ESP to the FrSky S.Port passthrough protocol. It is only after this translation layer where it's beneficial to exclude those packets. 

https://files.rfdesign.com.au/Files/documents/TXMOD%20-%20Adding%20S.PORT%20Support.pdf

mike-bfd, 2020-09-03:
Just got a confirmation email from one of the firmware developers at Gremsy. All of their products share the same type ID of 26.

> --> Yes, it is the same MAV_TYPE_ID for all our gimbal. It is set as default  MAV_TYPE_GIMBAL=26, /* Gimbal | */

matheusrfdesign, 2020-09-03:
I've just made an update on our firmware addressing the issue on the TXMOD.  @mike-bfd would you check if the issue is resolved? I don't have a gimbal with me right now to test it. 
http://files.rfdesign.com.au/Files/firmware/firmware1451.bin

zs6buj, 2020-09-03:
Ok, all under control then. :) I'll add the exclusion of mav_type 26 into my master.

zs6buj, 2020-09-11:
Hello Mike Thanks for keeping me in the loop. This is the patch that I applied to my mav2pt firmware:           if (ap_type_tmp == 5 || ap_type_tmp == 6 || ap_type_tmp == 18 || ap_type_tmp == 26 || ap_type_tmp == 27) break;                // Ignore heartbeats from GCS (6) or Ant Trackers(5) or Onboard_Controllers(18) or Gremsy Gimbal(26) or ADSB (27)) Line 1641 of my most recent update on GitHub here:  https://github.com/zs6buj/MavlinkToPassthru/tree/master/MavToPass_v2.62.1 I exclude heartbeats from Gremsy gimbals right up front when the arrive from Mavlink before they influence decisions based on mav_type (frame type). Your architecture in TX Mod may well be different.Could you be more specific about your comment “It is only after this translation layer where it's beneficial to exclude those packets.” ?I have not yet had the opportunity to flash your latest TX Mod firmware. I’ll see if I can try it later today.  With kind regardsEric Stockenstrom  Sent from Mail for Windows 10 From: mike-bfdSent: Thursday, 10 September 2020 6:57 pmTo: zs6buj/MavlinkToPassthruCc: Eric Stockenstrom; CommentSubject: Re: [zs6buj/MavlinkToPassthru] Flight Mode Confliction with multiple MAV IDs (#38) @kentmartin99 This is in reference to the TX MOD's new feature that is within the ESP8266 inside the transmitter module. The new feature allows for telemetry data to be interpreted on its way to the GCS, and translated by the ESP to the FrSky S.Port passthrough protocol. It is only after this translation layer where it's beneficial to exclude those packets.https://files.rfdesign.com.au/Files/documents/TXMOD%20-%20Adding%20S.PORT%20Support.pdf

closed 2020-10-12 by zs6buj