zs6buj/mav2pt #40: Pixhawk Mini to FRsky R9 Slim+

leol14, 2020-11-08:
Need MiniMOSD (Mavlink) and Flightdeck telemetry (Passthrough) with only one telem port on the mini. Mavlink output to Minimosd works well. Same output to teensy 3.2 wired to S-port of FRsky R9 not working yet, solid light, but no output to FlightDeck on Taranis 9+. Flightdeck not initializing, keeps waiting for flight controller. What .hex version should I be using with the teesny to interface Pixhawk Mini - Teensy 3.2 - R9 - Taranis - Flight deck? Thanks in advance, 

[image]

zs6buj, 2020-11-09:
Hi Leo, you need Air Mode, and the default .hex is Ground Mode. Stand by one and I'll create one for you to try.

zs6buj, 2020-11-09:
Ok, I've added an Air Mode .hex teensy 3.2 in the v2.63.2 folder on GitHub.

leol14, 2020-11-09:
Eric,

Thanks much! I’m stuck working for a living but will upload and give it try a bit later. Should it work with FlightDeck?

Cheers,

Leo

Leo Lichodziejewski PMP
Program Manager II

Airborne Systems - CA, US

________________________________
Phone: +1 657-859-3132
Mobile: +1 714-552-3865
Fax: 714-668-0446

This e-mail is intended solely for use of the individual or entity to which it is addressed and may contain information that is proprietary, privileged and exempt from disclosure under applicable law. The contents of this e-mail should not be disclosed to any other person.This communication may also contain data subject to U.S. export laws, not releasable to Foreign Persons unless authorized by 22 CFR 120-130 or 15.CFR 730-774. If so, that data subject to the International Traffic in Arms Regulations cannot be disseminated, distributed or copied to foreign nationals, residing in the U.S. or abroad, absent the express prior approval of the U.S. Department of State. If you have received this communication in error, please notify the sender by reply e-mail and destroy the e-mail message and any physical copies made of the communication.

External Mail

zs6buj, 2020-11-09:
Yeah, it should work with Flightdeck Leo.

leol14, 2020-11-10:
Clearly I'm doing something wrong, but the teensy uploader says the MavToPass_v2.63.2.ino.TEENSY32_Air_Mode.hex file is unreadable. I suspect the file is too large? It loads into the uploader, but the download button is greyed out and won't let me transfer to teensy (after pushing the button of course).  I can load smaller hex files and run them just fine (of course they don't actually work for my application). Any ideas, I'm dying to give your air mode a try. Thanks again.

zs6buj, 2020-11-10:
Leo, I compiled and flashed it again here, and it's working, but I noticed a 1 byte difference in size

[image]

This is the new one: **MavToPass_v2.63.2.ino.TEENSY32.hex** and it is in Air Mode

Which FC and firmware stack you are using?

leol14, 2020-11-11:
Working! I was unzipping incorrectly, once I got that pinheadedness behind me your .hex worked like a champ. I have Minimosd and FrSky Passthrough from the same telem port on my Pixhawk Mini. Utlimately both FPV with OSD and FlightDeck simultaneously, very cool! Thanks much for your innovation and personalized support!

zs6buj, 2020-11-11:
Happy to help Leo.

closed 2020-11-24 by zs6buj