zs6buj/mav2pt #41: Compile error when compiling v2.63.4 for dragonlink.

zhaodelong, 2020-11-24:
Try to complie v2.63.4 for dragonlink. Only changed 
//#define displaySupport 
#define ESP32_Variant     3    //  Dragonlink V3 slim with internal ESP32 - contributed by Noircogi

And try to verify. But it hit #error "Unsupported board type!". It seems ESP32 is not defined. But I did select ESP32_Variant     3.

Error message

`Arduino: 1.8.13 (Windows 10), Board: "Arduino Uno"
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Delong\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Delong\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Delong\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10813 -build-path C:\Users\Delong\AppData\Local\Temp\arduino_build_374109 -warnings=none -build-cache C:\Users\Delong\AppData\Local\Temp\arduino_cache_790209 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Delong\Downloads\MavlinkToPassthru-master\MavlinkToPassthru-master\MavToPass_v2.63.4\MavToPass_v2.63.4.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Delong\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Delong\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Delong\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10813 -build-path C:\Users\Delong\AppData\Local\Temp\arduino_build_374109 -warnings=none -build-cache C:\Users\Delong\AppData\Local\Temp\arduino_cache_790209 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Delong\Downloads\MavlinkToPassthru-master\MavlinkToPassthru-master\MavToPass_v2.63.4\MavToPass_v2.63.4.ino

Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Detecting libraries used...

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\Delong\\AppData\\Local\\Temp\\arduino_build_374109\\sketch\\MavToPass_v2.63.4.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for CircularBuffer.h: [CircularBuffer-master@1.2.0]

ResolveLibrary(CircularBuffer.h)

  -> candidates: [CircularBuffer-master@1.2.0]

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Delong\\Documents\\Arduino\\libraries\\CircularBuffer-master" "C:\\Users\\Delong\\AppData\\Local\\Temp\\arduino_build_374109\\sketch\\MavToPass_v2.63.4.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for mavlink_types.h: [c_library_v2]

ResolveLibrary(mavlink_types.h)

  -> candidates: [c_library_v2]

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Delong\\Documents\\Arduino\\libraries\\CircularBuffer-master" "-IC:\\Users\\Delong\\Documents\\Arduino\\libraries\\c_library_v2" "C:\\Users\\Delong\\AppData\\Local\\Temp\\arduino_build_374109\\sketch\\MavToPass_v2.63.4.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Error while detecting libraries included by C:\Users\Delong\AppData\Local\Temp\arduino_build_374109\sketch\MavToPass_v2.63.4.ino.cpp

Generating function prototypes...

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Delong\\Documents\\Arduino\\libraries\\CircularBuffer-master" "-IC:\\Users\\Delong\\Documents\\Arduino\\libraries\\c_library_v2" "C:\\Users\\Delong\\AppData\\Local\\Temp\\arduino_build_374109\\sketch\\MavToPass_v2.63.4.ino.cpp" -o "C:\\Users\\Delong\\AppData\\Local\\Temp\\arduino_build_374109\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE

In file included from C:\Users\Delong\Downloads\MavlinkToPassthru-master\MavlinkToPassthru-master\MavToPass_v2.63.4\MavToPass_v2.63.4.ino:142:0:

config.h:213:4: error: #error "Unsupported board type!"

   #error "Unsupported board type!"

    ^~~~~

config.h:283:8: error: #error Bluetooth works only on an ESP32 board

       #error Bluetooth works only on an ESP32 board

        ^~~~~

Using library CircularBuffer-master at version 1.2.0 in folder: C:\Users\Delong\Documents\Arduino\libraries\CircularBuffer-master 

Using library c_library_v2 in folder: C:\Users\Delong\Documents\Arduino\libraries\c_library_v2 (legacy)

exit status 1

#error "Unsupported board type!"

`

config.h

`//================================================================================================= 
[... 1251 lines cut ...]
v2.61.9 2020-09-04  Tidy up serial downlink capability. Add outgoing TCP client capability. 
                    Improve web setup data vetting. 
                    Show last line properly & reverse scroll buttons on ST7789.
                    Ignore heartbeats from or Gremsy Gimbal(26).       
v2.62.0 2020-09-08  Add support for multiple incoming tcp clients (GCSs, trackers..)
                    Always broadcast UDP on the /24 subnet. 
v2.62.1 2020-09-10  Restore #defined UDP_broadcast option :)    
v2.62.2 2020-09-15  Add support for multi targeted UDP clients.
                    Flush UDP buffer after send. 
                    Always broadcast heartbeat for UDP.   
v2.62.3 2020-09-17  AP channel change fixed.
                    Display remote IP fixed. 
v2.62.4 2020-09-16  Fix BT slave name truncated by 1 chr
                    Improve when BT disabled to free up SRAM for web support, also #undef btBuiltin     
v2.62.8 2020-10-07  Improve display scrolling
[... cut ...]
v2.63.2 2020-11-02  Add switchable flight info page on display                         
        2020-10-31  For mark home include GPS 3D Fix                                        
v2.63.3 2020-11-04  Enumerate info page pins for more variants    
        2020-11-11  Info page working for all supported display types                                                                                                                                                                                                                     
*/`

zs6buj, 2020-11-24:
Hello my friend

From this line:

"Arduino: 1.8.13 (Windows 10), Board: "Arduino Uno" it appears that you have chosen the Arduino Uno board in Tools/Board ? You should select ESP32 Dev Board like this:

[image]

zs6buj, 2020-11-24:
I also added these binaries on GitHub for you:

MavToPass_v2.63.4.ino.esp32_Dragonlink_Ground_Mode.bin

MavToPass_v2.63.4.ino.esp32_Dragonlink_Air_Mode.bin

zhaodelong, 2020-11-24:
Awesome, thanks.

closed 2020-11-24 by zhaodelong