zs6buj/mav2pt #42: Integration of this project with another

vincemic, 2020-11-30:
Thank you for your work on this project, As flight control systems get more and more complicated the limited number of serial ports available on flight controllers become a real problem.  I have run out of serial ports and I would like to incorporate this library into my project [Restraining Bolt](https://github.com/bitdog-io/restraining_bolt). It also uses MAVLink protocol. It attempts to prevent runaways of large rovers that can occur if onboard systems fail. If I can incorporate MavlinkToPassthru, then it would eliminate the need for an additional serial port from the flight controller.  What pieces would be required from this library at a minimum?

zs6buj, 2020-12-01:
@vincemic:

Nice work on your project, and a valuable safety feature.

Hmmm. Which part of mav2pt? 

The original raison d'être for mav2pt was to translate Mavlink into the FrSky DIY protocol, using serial telemetry in and out, for display on FrSky transmitter screens, but later I added WiFi, BT and SD card uplink and downlink.  I'm assuming that you need the multi medium / multi-protocol features, and don't need the translation? 

It could be tricky to extract all the wifi input, output and control code. I also support displays and web configuration, with OTA updating. Ironically, it is relatively easy to lose the Mavlink/FrSky translation. Simply remove the FrSky_Ports.h library, compile, and then remove all references to the FrSky_Ports class. You will be left with a bi-directional "proxy" that supports wifi and BT.

It then might be easier to strategically insert your functionality. Alternatively, take a careful look at my main block, and locate the uplink and downlink loops.

FC implies uplink and GCS means downlink.

`Read_FC_To_RingBuffer();  ` which eventually writes to GCS

and

```
  Read_From_GCS();  

  if (GCS_available) {
    Decode_GCS_To_FC();
    Write_To_FC(G2Fmsg.msgid);  
    GCS_available = false;                      
   }
```

You would need to include the config blocks and setup blocks for WiFi and BT, and a bunch of global variables, but you could work through it.

vincemic, 2020-12-01:
I only need the Mavlink/FrSky translation at this point. This would allow a single FC serial port and a companion computer to handle both the safety features and forward frsky telementry.  The bit of code you provided is helpful as it gives me a bit of insight in how the code is architected. I am now beginning to wonder if I should just adopt the whole feature set by offering to make my feature a plugin to yours or yours to mine.

zs6buj, 2020-12-02:
Vincent, If you simply need the Mavlink/FrSky translation just include the FrSky_Ports class, and look at these public member functions

```
    // Member function declarations
  public:    
    void          initialise();
    void          HandleTraffic();
    void          PushMessage(uint16_t msg_id, uint8_t sub_id); 
    void          ReportFrPortOnlineStatus(); 
```  

initialise()  self explanatory
HandleTraffic()   just include this unchanged in the main loop
PushMessages()  populate and call one for each FrSky message type.  copy my code in the Mavlink loop.
ReportFrPortOnlineStatus()   optional status report for Fport or SPort

zs6buj, 2021-01-18:
Stale ?

closed 2021-01-18 by zs6buj