zs6buj/mav2pt #43: crossfire micro TX V2 as MavLink input

kniuk, 2021-01-10:
Hi is there an option to setup MavlinkToPassthru to connect to Crossfire Micro TX V2 wifi module as MavLink input?
I was encouraged to do some tests because I found DOWNLINK can be set to WiFi on a setup page of the MavlinkToPassthru (please see attached screenshoot).

[image]

However I failed to set it up. Crossfire WiFi default settings is UDP:8888. I have both connected to my home WiFi network, both assigned IPs from the same subnet.
My setup is  an ESP8266 D1 mini board with an OLED 128x64 display.
My goal is to have a MavLink link between an Ardupilot FC and a GCS and be able to use the Yaapu's script on my Jumper T16 via S.Port. I am planning to use Crossfire Tx module in PPM mode and feed S.Port input from MavlinkToPassthru (PPM mode would be neccessary because in CRSF mode S.Port is repurposed to CRSF protocol).

P.S. I also tried to listen to the serial transmission between Crossfire Tx PCB and builtin wifi module, but haven't found right baudrate or the data is encapsulated somehow (will need to look at the signal on the oscilloscope to time the baudrate). I know I have to look for 0xFD packet start "magic" marker in a stream

zs6buj, 2021-01-10:
Hi kniuk

Hmm. Some routers don't allow UDP broadcast by default, and we briefly need a broadcast to identify the downlink remote station. Maybe that is the issue. You could check your router settings. 

By the way, I have changed the downlink/uplink convention today in v2.63.11. Uplink is now to the FC.

Most of my AP_STA mode (WiFi uplink and downlink) testing was using TCP/IP. I'll see if I can test your setup in the next day or two.

I found this [UDP Monitor](https://play.google.com/store/apps/details?id=com.sandersoft.udpmonitor&hl=en_ZA&gl=US) useful while testing

zs6buj, 2021-01-17:
@kniuk:

It took me a few days, but as Siri says, "this is what I found":

As you know, for simultaneous STA and AP connection to the ESP32, we need to invoke WIF_AP_STA mode. In this mode and using UDP, we mast read and write UDP through the STA , AND read and write UDP through the AP. When we call UDP.begin(port_number) of the previously instantiated UDP object, for the STA side read we initialise the object with the receiving port number. But we need a different port number for the AP side read. If the port numbers are the same, both reads will received both data streams. (I tested this, and it is true).

If we call UDP.stop(), and then UDP.begin(port_number) the data stream is interrupted and time is wasted. So I now instantiate separate UDP objects for STA and AP, for use only in WIFI_AP_STA mode, and keep a small table (of 2) of the object pointers. Now we instantiate with the different read ports, and select the appropriate object pointer for reading from STA or AP. Write ports can be assigned at write() time, and local(receive) and remote(send) IPs must obviously match at both ends. I ran out of time yesterday, but I'll try to get it all tested next chance I get.

kniuk, 2021-01-17:
Great news, thank you for your time! I am eagerly waiting for further info.

zs6buj, 2021-01-22:
@kniuk 

I've posted MavToPass_v2.63.12, which should do UDP uplink and downlink for you. It might need a bit more testing, so let me know how it goes.

zs6buj, 2021-01-22:
The ports are quite tricky. You can use this example to match send and receive ports:

[image]

kniuk, 2021-01-22:
thanks!, I will test it this weekend

kniuk, 2021-01-25:
I didn't manage to test it last weekend, I will try in following days!
Regards,
Jacek

zs6buj, 2021-02-12:
You need to enter the local port (read port)  in MP or Crossfire.  I believe MP expects14555 for remote (sending) port.

[image]

zs6buj, 2021-02-27:
Stale? If not, please reopen

closed 2021-02-27 by zs6buj