zs6buj/mav2pt #46: pt_imu_temp uint8_t

rotorman, 2021-01-17:
pt_imu_temp is defined as uint8_t.
The code assigns it a value from ap26_temp without a cast: `pt_imu_temp = ap26_temp;`
ap26_temp is int16_t.

1) 16-bits vs. 8 bits and a signed value to unsigned value. Please check/revise.

2) In case only raw IMU values are sent on MAVLink, then ap27_temp could also be considered as source for pt_imu_temp.

zs6buj, 2021-01-18:
Hmm. The assignment will simply truncate the msd bits if the number is always positive, but I guess in extreme weather it could be negative. But there is another problem here. 0x5001 only allows 6 bits (#63 max) with no sign flag, and the unit is reported to be cdegC (0.01C). 

I remember we cobbled this field on as a request for someone, and there are implications for yaapu. I'll discuss with him.

rotorman, 2021-01-18:
Well a quick solution without implications to @yaapu would be to have a lower limit bound of, but there needs to be code to check this.

Also correct cast to uint8_t needs to be taken care of for values > 255.

Here is my suggestion:
```
if (ap26_temp < 0)
  pt_imu_temp = 0;
else
 if (ap26_temp > 255)
  pt_imu_temp = 255;
 else
  pt_imu_temp = ap26_temp;
```

closed 2021-01-22 by zs6buj