zs6buj/mav2pt #48: satellites_visible should handle special case 255

rotorman, 2021-01-17:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/MavToPass_v2.63.11.ino#L1944

and 

https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/FrSky_Ports.h#L1781-L1784

do not handle the special case of 255, where [MAVLink GPS_RAW_INT #24](https://mavlink.io/en/messages/common.html#GPS_RAW_INT) uses a value of 255 for unknown satellite count.

Best, additionally to check if mavlink_msg_gps_raw_int_get_satellites_visible() == 255 and set the ap24_sat_visible to 0 if so in order not to output wrongly 15 over FrSky.

zs6buj, 2021-01-18:
I made this change:

```
 if (ap24_sat_visible > 15)   // for special case stats_visible == 255   unknown satellite count
   pt_numsats = 0;               // to avoid false sense of security 
 else 
   pt_numsats = ap24_sat_visible; 
```

closed 2021-01-18 by zs6buj

rotorman, 2021-01-18:
Please re-open this case, as this change wrongly outputs for satellite count 16-254 zero.

Please consider the following instead:
 if (ap24_sat_visible > 15) 
{
  if (ap24_sat_visible == 255)
    pt_numsats = 0; // for special case stats_visible == 255   unknown satellite count
  else
    pt_numsats = 15; // limit to 15 due to only 4 bits available
}
 else 
   pt_numsats = ap24_sat_visible;