zs6buj/mav2pt #49: GPS status bits not correctly split into pt_gps_status and pt_gps_adv_status

rotorman, 2021-01-17:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/FrSky_Ports.h#L1788-L1789
and
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/FrSky_Ports.h#L1794-L1795

do not correctly split the 2+2 bits, as this leaves the lower bits to 11 for ALL values higher than 3 and does not allow all 8 MAVLink GPS_FIX_TYPE enum possibilities to be passed (which is no problem with 4 bits, as we can send up to 16 enum values with it, just not with the present funky implementation):

0;  GPS_FIX_TYPE_NO_GPS; pt_gps_status=0; pt_gps_adv_status =0
1; GPS_FIX_TYPE_NO_FIX;  pt_gps_status=1; pt_gps_adv_status =0
2; GPS_FIX_TYPE_2D_FIX;  pt_gps_status=2; pt_gps_adv_status =0
3; GPS_FIX_TYPE_3D_FIX;  pt_gps_status=3; pt_gps_adv_status =0
4; GPS_FIX_TYPE_DGP;  pt_gps_status=**3**; pt_gps_adv_status =1
5; GPS_FIX_TYPE_RTK_FLOAT;  pt_gps_status=**3**; pt_gps_adv_status =2
6; GPS_FIX_TYPE_RTK_FIXED;  pt_gps_status=**3**; pt_gps_adv_status =3
7; GPS_FIX_TYPE_STATIC;  pt_gps_status=**3**; pt_gps_adv_status =**4**
8; GPS_FIX_TYPE_PPP;  pt_gps_status=**3**; pt_gps_adv_status =**5**

Also, as pt_gps_adv_status has only 2 bits, it should not go above value 3!

Better would be to handle the split like this:
pt_gps_status = ap24_fixtype & 0x03;
pt_gps_adv_status = (ap24_fixtype >> 2) & 0x03;

This gives:
0;  GPS_FIX_TYPE_NO_GPS; pt_gps_status=0; pt_gps_adv_status =0
1; GPS_FIX_TYPE_NO_FIX;  pt_gps_status=1; pt_gps_adv_status =0
2; GPS_FIX_TYPE_2D_FIX;  pt_gps_status=2; pt_gps_adv_status =0
3; GPS_FIX_TYPE_3D_FIX;  pt_gps_status=3; pt_gps_adv_status =0
4; GPS_FIX_TYPE_DGP;  pt_gps_status=0; pt_gps_adv_status =1
5; GPS_FIX_TYPE_RTK_FLOAT;  pt_gps_status=1; pt_gps_adv_status =1
6; GPS_FIX_TYPE_RTK_FIXED;  pt_gps_status=2; pt_gps_adv_status =1
7; GPS_FIX_TYPE_STATIC;  pt_gps_status=3; pt_gps_adv_status =1
8; GPS_FIX_TYPE_PPP;  pt_gps_status=0; pt_gps_adv_status =2

@yaapu might need to fix the telemetry script to match this as well:
https://github.com/yaapu/FrskyTelemetryScript/blob/82b2245639e209b875b994cc9dd0ae2e9f4e3f0e/HORUS/SOURCES/SRC/WIDGETS/Yaapu/main.lua#L854-L856

zs6buj, 2021-01-18:
Nice catch again! I'll talk to Alex if he has not seen this.

closed 2021-01-18 by zs6buj

yaapu, 2021-01-18:
Hi, Eric's code is correct since ardupilot implementation only uses 2+2 bits (was not my choice) an there's no room for extra ones.
I suggest Eric's code sticks to the specs and keeps the 2+2 scheme, finally my script decodes 2+2 as per specs, it's a feature not a bug :-)

rotorman, 2021-01-18:
Thank you Alex!
Just read about the issue post here: [https://github.com/ArduPilot/ardupilot/issues/5437](https://github.com/ArduPilot/ardupilot/issues/5437)
Maybe we could add a comment about this peculiarity in the code?

yaapu, 2021-01-18:
@rotorman not at the gps driver level but rather at the frsky library level check here https://github.com/ArduPilot/ardupilot/blob/210b0e7c5fb2b08d19bdac50dca9c5474257d5a1/libraries/AP_Frsky_Telem/AP_Frsky_SPort_Passthrough.cpp#L384-L400