zs6buj/mav2pt #50: pt_hdop should be uint16_t instead of uint8_t

rotorman, 2021-01-17:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/FrSky_Ports.h#L1808

as prep_number returns uint16_t.

The sign of the return value is in the 8th bit, thus in MSByte. With 2,1 parameter, prep_number() returns following limits:
0 1111 1110 = +127
1 1111 1110 = -127

0 1111 1111 = +1270
1 1111 1111 = -1270

zs6buj, 2021-01-18:
agree, done

zs6buj, 2021-01-18:
Let me use this space to say thank you for your diligent code review. Nice work, and I have credited the patches to @rortorman.

I'm busy with some other changes, but the patches will be published soon.

Kind regards
Eric

closed 2021-01-18 by zs6buj

rotorman, 2021-01-18:
More than happy to help! And thank you for the great Mav2PT!
Best,
Risto