zs6buj/mav2pt #55: pt_imu_temp needs further conversion (division with 100 and offset -19) to match @yaapu telemetry

rotorman, 2021-01-18:
mavlink_msg_scaled_imu_get_temperature() outputs int16_t in 0.01°C units. Here the value is read into int16_t ap26_temp:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/MavToPass_v2.63.11.ino#L2018

and here it is directly passed to pt_imu_temp, which is a uint8_t (problem: data length and a sign):
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/FrSky_Ports.h#L1748

and here sent out to FrSky msg 0x5001 using 6 bit value:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/FrSky_Ports.h#L1757

@yaapu is handling it in telemetry script as a °C value with -19°C offset:
https://github.com/yaapu/FrskyTelemetryScript/blob/82b2245639e209b875b994cc9dd0ae2e9f4e3f0e/HORUS/SOURCES/SRC/WIDGETS/Yaapu/main.lua#L850-L851

Either Mav2PT or Yaapu script needs a fix for them to work correctly together.

For Mav2PT, the fix could look like:
```
if ((ap26_temp / 100) - 19 < 0)
  pt_imu_temp = 0; // shows +19°C in yaapu, the lowest value possible
else
{
  if ((ap26_temp / 100) - 19 > 63)
    pt_imu_temp = 63; // shows +82°C in yaapu, the highest value possible
  else
    pt_imu_temp = (ap26_temp / 100) - 19; // shows the true value in 1°C resolution
}

```

yaapu, 2021-01-18:
is not me handling it that way, it's way the ardupilot library encodes it and the library predates my involvment by a couple years, I suggest we stick with the "master" of the encoding specs which is ardupilot

rotorman, 2021-01-18:
OK, then Mav2PT could get following fix as suggested:
```
if (((ap26_temp / 100) - 19) < 0)
  pt_imu_temp = 0; // shows +19°C in yaapu, the lowest value possible
else
{
  if (((ap26_temp / 100) - 19) > 63)
    pt_imu_temp = 63; // shows +82°C in yaapu, the highest value possible
  else
    pt_imu_temp = (ap26_temp / 100) - 19; // shows the true value in 1°C resolution
}
```

closed 2021-01-22 by zs6buj