zs6buj/mav2pt #58: better pt_hdop (GPS horizontal dilution) handling

rotorman, 2021-01-19:
Currently, input for GPS horizontal dilution is received with:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/MavToPass_v2.63.11.ino#L1949
where the output is uint16_t and 0xFFFF is used for unknown state (MAVLink specs).

The info is passed to FrSky passthrough message 0x5002, where it has 7 value bits and a 1-bit multiplication factor.
Thus the usable range is 0 to +1270.

Dilution is unitless, where lower values denote better GPS positioning accuracy, so my suggestion is to limit the output to +1270 (this automatically handles the MAVLink unknown case 0xFFFF correctly as outputting the worst value of +1270) and NOT divide it with 10, which in my opinion, is not necessary:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/FrSky_Ports.h#L1792

Fine dilution resolution between 1 and 10 is more interesting for grading the GPS signal than having more range of the dilution (thus for high values). Thus IMHO the division with 10 is disadvantageous. 

I suggest the following instead of line 1792 above (assuming my previous comment about changing pt_hdop from uint8_t to uint16_t is already done):
if (ap24_eph > 1270)
   pt_hdop =1270;
else
  pt_hdop = ap24_eph;

zs6buj, 2021-01-25:
@rotorman:

Nice argument, but by changing the unit we would need to make changes all the way though to APM, and including Flight Deck and other others. I'm going to give this one a miss.

closed 2021-01-25 by zs6buj

rotorman, 2021-01-25:
OK for changing the unit, but please consider then the following:
```
if (ap24_eph > 1270)
   pt_hdop = 1270;   // unknown state 0xFFFF from MAVLink is also relayed as 1270
else
  pt_hdop = ap24_eph / 10;
```

zs6buj, 2021-01-25:
Agreed.