zs6buj/mav2pt #60: Improved battery voltage handling plus a discussion about a special case for 12 cells in FrSky

rotorman, 2021-01-19:
I would like to propose a better battery voltage handling and also like to discuss about a special case for 12 cells that is presently  implemented in @yaapu telemetry and how to match this with Mav2PT.

In Mav2PT for battery 1, the voltage is read in via:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/MavToPass_v2.63.11.ino#L1746
where the output of mavlink_msg_sys_status_get_voltage_battery() is uint16_t and the unit used is mV. A special case of 0xFFFF needs to be handled.
The range is 0 to 65.534V, and 65.535 (=0xFFFF) is the special case.

Get_Volt_Average1() converts it over float to uint32_t:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/global_variables.h#L83
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/Utilities.ino#L1519-L1527

In FrSky_Ports.h it is converted to dV, before sending out as 9 bit value:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/FrSky_Ports.h#L1827
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/FrSky_Ports.h#L1844

The range via FrSky is thus 0 to 511, or in SI-units 0 to 51.1 V.

I propose first to handle the special case 0xFFFF in MavToPass_v2.63.11.ino L1746:
```
if (mavlink_msg_sys_status_get_voltage_battery(&R2Gmsg) == 0xFFFF)
  ap_voltage_battery1= 0; // Voltage not sent by autopilot
else
  ap_voltage_battery1= Get_Volt_Average1(mavlink_msg_sys_status_get_voltage_battery(&R2Gmsg));  
```

Further, I propose to add following code snippet instead of FrSky_Ports.h L1827 to limit the output:
```
if ((ap_voltage_battery1 / 100) > 51100)
  pt_bat1_volts = 511; // upper bound, limit output to 51.1V
else
 pt_bat1_volts = ap_voltage_battery1 / 100; // convert from mV to dV
```

------------------------
I am not sure if the special 12-cell handling stems from FrSky or Alex, but in @yaapu's telemetry 12-cells batteries are handled differently. With 12C batt, the values below 24.0V are assumed to be roll-over values (please ignore the comment typo of 52.3V that should be 51.2V in the following):
https://github.com/yaapu/FrskyTelemetryScript/blob/82b2245639e209b875b994cc9dd0ae2e9f4e3f0e/HORUS/SOURCES/SRC/WIDGETS/Yaapu/main.lua#L860-L866

This results in the FrSky transmitted voltage range of:
a) for battery cellcount 1 to 11: 0 to 51.1V
b) for battery cellcount 12: 51.2V to 75.1V (values 0 to 239) and 24.0V to 51.1V (values 240 to 511).
(as MAVLink input max is 65.534V, values go in reality only from 0 to 143 and 240 to 511, there is an unused gap from 144 to 239).

The implementation in Mav2PT to match this, could be (I took the liberty to handle not exactly only for cellcount 12, but higher cell counts in the code below similarly to cellcount 12 handling as well):
```
if (ap_ccell_count1 < 12)
{
  if ((ap_voltage_battery1 / 100) > 51100)
    pt_bat1_volts = 511; // upper bound, limit output to 51.1V
  else
   pt_bat1_volts = ap_voltage_battery1 / 100; // convert from mV to dV
}
else
{
  // here ap_ccell_count1 is >= 12
  // Upper limit check can be omitted, as due to uint16_t MAVLink limitation, the value will never be > 65534, alas >65.534V
  if ((ap_voltage_battery1 / 100) > 51100)
   // voltage >51.1 and <=65.5V
   pt_bat1_volts = (ap_voltage_battery1 / 100) - 512; // convert from mV to dV and subtract 51.2V offset
 else
 {
   // voltage <= 51.1V
   if ((ap_voltage_battery1 / 100) >= 24000)
     pt_bat1_volts = (ap_voltage_battery1 / 100); convert from mV to dV
  else
     pt_bat1_volts = 24000; // lower bound, limit output to 24.0V to avoid wrapping over to 75.1V
  }
}
```

BUT, this needs further thought how to reliably detect the cell count, as from above 6 cells this is not easily doable just by looking at the (changing battery) voltage.

Mav2PT currently implements cell count detection up to 6 cell batteries:
https://github.com/zs6buj/MavlinkToPassthru/blob/0f469ce22c8c3f8d7d224751a2ccf244df9069f8/MavToPass_v2.63.11/MavToPass_v2.63.11.ino#L1748-L1753

But where to get a reliable value for ap_ccell_count1 for higher cell counts?

MAVLink [BATTERY_STATUS (#147)](https://mavlink.io/en/messages/common.html#BATTERY_STATUS) supports outputting voltages of up to 14 separate cells, but most systems currently used do not use this possibility, so cannot set on this.

yaapu, 2021-01-19:
Hi @rotorman, let's keep this simple I think you are over complicating it, I suggest MavToPT mimics ardupilot code.

This is how my scripts handle ardupilot battery voltages:
For voltages > 24V cellcount is not taken into account and telemetry voltage is assumed real voltage. (so 1-12s, voltage > 24)
For voltages < 24 I check the cellcount because users with LiHV might have a full voltage pack > 51.2 and ardupilot right now "wraps" the voltage, so it transmit whatever is in excess of 51.2. If that's the case than apply the conversion 51.2 + transmitted voltage, for this to work the user has to "force" cellcount in the script config for autodetection will fail for sure.

Just like ardupilot MavToPT should do no conversion but apply the same algo ardupilot does, simply wrap excess > 51.2, it will be telemetry script responsability to decode the value

Finally there is no reliable cellcount source in mavlink, only smart batteries do have the information and the vast majority out there are not smart "at all" :-)

rotorman, 2021-01-19:
So, what you are saying is, to go with the following:

a) handle the special case 0xFFFF in MavToPass_v2.63.11.ino L1746:
```
if (mavlink_msg_sys_status_get_voltage_battery(&R2Gmsg) == 0xFFFF)
  ap_voltage_battery1= 0; // Voltage not sent by autopilot
else
  ap_voltage_battery1= Get_Volt_Average1(mavlink_msg_sys_status_get_voltage_battery(&R2Gmsg));  
```

b) add following code snippet instead of FrSky_Ports.h L1827 to limit the output:
```
if ((ap_voltage_battery1 / 100) > 51100)
  pt_bat1_volts = 511; // upper bound, limit output to 51.1V
else
 pt_bat1_volts = ap_voltage_battery1 / 100; // convert from mV to dV
```

and skip the other suggestion, correct?

yaapu, 2021-01-19:
a cleaner way for the wrapping assuming pt_bat1_volts is >= 16bits

pt_bat1_volts = (((uint16_t)roundf(ap_voltage_battery1*0.01)) & 0x1FF);

example here https://onlinegdb.com/ryWR5FV1_

rotorman, 2021-01-19:
Your solution is great!
(The URL to the example does not work for me though)

closed 2021-01-25 by zs6buj