zs6buj/mav2pt #61: MavlinkToPassthru for Jeti and Core radios

master-of-desaster-88, 2021-01-31:
Hi Eric, I hope I find You well in this times.
Now Your MavlinkToPassthru is also working for the FrSky F.Port. A realy great work from You.
Do You think it is possible to make the "Passthru" also for Jeti and Core radios?
I think about to buy one of them for my large turbine jets. But without the Mavlink telemetrie I will not do that.
Whit kind of telemetrie protokoll informations are needed if it is possible to write the code?
Cheers, Markus

zs6buj, 2021-01-31:
Hi Markus,  it's good to hear from you my friend. Hope you too are well under the circumstances. 

I will have to find more detail on the Jeti and Core radios.  Mavlink is most suitable because the work is already done, but I would be happy to investigate further.  I'll let you know. 

Regards
Eric

master-of-desaster-88, 2021-01-31:
Hi Eric, we are fine here. Testing many FFP2 and FFP3 mask.
At the winter season time to building new planes. 
The new one is a F-18 Super Hornet from Jetlegend with a 160N turbine, so I should use a other kind of RC controler.
For the Jeti is a kind of translator aviable. https://github.com/DevFor8/Mav2Duplex.
It seems to work but it is far away from Your solutions.
The Core Guys only share the telemetry code with a NDA. I asked them. Not the nice way.
So I will more go to the Jeti.

I also have start a small web-page with a build review from my planes.
www.mirage2000.eu

Regards,
Markus

zs6buj, 2021-02-01:
I wish I had the skill and patience to build and fly those awesome jets! Ah well!

I think the github link will be useful. I'll look at it today.

zs6buj, 2021-02-01:
Nice web site, BTW

master-of-desaster-88, 2021-02-01:
I don't build them. I buy them RTF or ARF. I install the engine, RC control and autopilot.
Just start with a jet trainer like the PC21 from FMS. My pilot skill is not so pefect. The most work is done from the autopilot.
To help a little bit - as follow the link to the Jeti telemetrie protokoll: http://www.jetimodel.com/de/Telemetrieprotokoll/

zs6buj, 2021-02-08:
Hi Markus, I'm sorry I have been slow with this. :(  I've been really crazy busy, but I hope to look  at it today

master-of-desaster-88, 2021-02-08:
Hi Alex, no problem at all. The new RC will be delivered by today.
I also have ordered a beginner kit for the Arduino as I like to learn more about the codeing in C and C++

zs6buj, 2021-02-08:
Markus, I see that the Jeti protocol is similar is concept to the FrSky sensor protocol, and quite extensive in nature. It would not be a trivial matter to write code from scratch. However, I found some interesting libraries by Sepp62 on GitHub:

https://github.com/Sepp62/TxJetiEx

https://github.com/Sepp62/JetiExBus

https://github.com/Sepp62/JetiExSensor

and this one

https://github.com/geins/VOT2EX

With these libraries it is likely you could save a lot of time. **So far I could not find a document showing message structures and types.**

master-of-desaster-88, 2021-02-09:
Hi Eric, tanks for the links. I have build one according to https://github.com/DevFor8/Mav2Duplex.
It is working but I still learnig the coding to understood how to change the values.
I got installed Visual Studio with the Arduio IDE from Visual Studio.
It's working - and I am so proud of me :-)
So I will go trough the lessons and hopefully finaly I could create one version who fits to my Jets.

Regards,
Markus

master-of-desaster-88, 2021-02-09:
Sorry, I forgot the link to the Jeti Bus:
http://www.jetimodel.com/en/Telemetry-Protocol/JETI-Telemetry-Communication-Protocol/

zs6buj, 2021-02-10:
Hey Markus, well done my friend ! Excellent! I was expecting this to be quite a big project. 

Thank you for the library link, I will look more closely.

master-of-desaster-88, 2021-02-19:
Hey Eric, I had done some changes on the code from "DevFor8/Mav2Duplex" and store it on my forc folder.
It is working with the Jeti DC24 and the Ardupilot Hardware. Limit is only 15 sensors - but seems enugh for my bussines.

zs6buj, 2021-02-19:
Hey Markus

That's great news my friend. If anyone asks I will know who they can talk to.    :)

Viel Spaß beim Fliegen!

Eric

closed 2021-02-19 by zs6buj