zs6buj/mav2pt #66: impossible to flash into brand new wifi kit 32

jeanbatfr, 2021-04-27:
Hello, I tried to install your program into a heltec wifi kit 32 with the help of this guide : 
https://docs.google.com/document/d/1ePGAmYa-GSwJ9TrpRcMSZ5HjKpIELJfxtvpISxID2Xs/edit

But after the tool said "finish", nothing happen even after several resets.

I tried with arduino IDE but I have a error message : 

Le croquis utilise 1605430 octets (122%) de l'espace de stockage de programmes. Le maximum est de 1310720 octets.
text section exceeds available space in boardLes variables globales utilisent 76104 octets (23%) de mémoire dynamique, ce qui laisse 251576 octets pour les variables locales. Le maximum est de 327680 octets.

(Sketch is too big????)

What should I do please?

jeanbatfr, 2021-04-27:
Ok, "Minimal SPIFFS(Large APPS with OTA)" is not selectionnable if we select Heltec wifi kit 32, we need to choose ESP 32 Dev module...

zs6buj, 2021-04-27:
Bonjour mon ami,

Oui, vous avez raison, la partition de mémoire de l'application a été dépassée. Pour corriger cela, vous devez sélectionner la version minimum de spiffs du schéma de partition. Je vois que vous le savez, mais le kit wifi Heltec 32 n'a pas ce profil. Si je me souviens bien, cette carte heltec a 8 Mo de sram, donc llok pour cette option, ou allez dans les fichiers de variantes dans l'IDE, et modifiez la configuration de la partition en copiant à partir du kit de développement esp32.

Malheureusement, mon équipement n'est actuellement pas disponible pour moi.

Cordialement
Eric

zs6buj, 2021-04-27:
Ok, I logged into the latest Arduino IDE, and I see the problem. Let me see if I can hack the partition file for you.

zs6buj, 2021-04-27:
Try this:

Navigate to the equivalent folder on your windows 10 installation:

C:\Users\erics\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

Save a copy of the boards.txt file

Download this [boards.txt](https://drive.google.com/file/d/14_5cdOHoLoF4t4N_1hXtE8TGDRFGxN1m/view?usp=sharing), and copy it into the above folder

Now open the Arduino IDE and select the Min Spiffs partition scheme

[image]

Now compile, and it should be fine.
[image]

jeanbatfr, 2021-04-27:
Actually, it seems to work by selecting esp 32 dev module, but then I don't know if it's the cause of my next problem with Dragon link connection

zs6buj, 2021-04-27:
The assignment of pins on the dev board and heltec board is different. I assume you selected the heltec board in config.h

Best you use my modified board.txt, and select the helec variant.

jeanbatfr, 2021-04-27:
ok, it's maybe the answer of the second issue. But Can you please give the autorization to download your file? thank you

jeanbatfr, 2021-04-27:
ok, I upload the sktech using your file, but I still have the next issue, I propose to close this one and go to the second.
Thank you

closed 2021-04-27 by jeanbatfr