zs6buj/mav2pt #67: No telemetry on yaapu/taranis

jeanbatfr, 2021-04-27:
Hi again, sorry for posting two issues the same day.

My goal is to have the telemetry from the model's pixhawk through dragonlink, displayed on opentx.
For now I have a x9e, yaapu9 works, but I still get "no telemetry".

On the Heltec Screen I have the good GPS position displayed, so I think everything work until the connection Heltec-Taranis.

I connected the pin 14 to the s.port of the Taranis' JR connector, are we ok with that?
I st up the serial port to "telemetry" and activate "discover new sensors".

What did I miss?
Thank you

zs6buj, 2021-04-28:
It all sounds OK. I assume you also  connected ground to the taranis :)

Could you post the serial monitor log and let's see if we can spot anything.

jeanbatfr, 2021-04-28:
Are you talking about arduino's serial monitor?

zs6buj, 2021-04-28:
The Arduino IDE outputs log messages from the firmware to  serial (0)

zs6buj, 2021-04-28:
Navigate to Tools, serial monitor

jeanbatfr, 2021-04-28:
16:24:35.895 -> 
16:24:35.895 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
16:24:35.895 -> configsip: 0, SPIWP:0xee
16:24:35.895 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:24:35.895 -> mode:DIO, clock div:1
16:24:35.895 -> load:0x3fff0018,len:4
16:24:35.942 -> load:0x3fff001c,len:1044
16:24:35.942 -> load:0x40078000,len:10124
16:24:35.942 -> load:0x40080400,len:5856
16:24:35.942 -> entry 0x400806a8
16:24:38.983 -> 
16:24:38.983 -> Starting MavToPass version:2.64.09
16:24:38.983 -> Display support activated: Landscape 
16:24:38.983 -> 64x128  TEXT_SIZE=1  CHAR_W_PX=6  CHAR_H_PX=8  SCR_H_CH=8  SCR_W_CH=21
16:24:39.031 -> EEPROM initialised successfully
16:24:39.079 -> EEPROM settings read and adopted
16:24:39.079 -> Ground Mode selected
16:24:39.079 -> Battery_mAh_Source = 3 - Define battery capacities in the LUA script
16:24:39.127 -> RSSI Override for testing = 70%
16:24:39.174 -> Target Board is ESP32 / Variant is Heltec Wifi Kit 32
16:24:39.221 -> Mavlink Serial In
16:24:39.268 -> FrSky Serial Out
16:24:39.315 -> Mavlink WiFi+BT Out - WiFi mode is STA>AP
16:24:39.409 -> Protocol is UDP IP-Targeted
16:24:40.076 -> Bluetooth slave mode, host name for pairing is MavToPass
16:24:40.216 -> Mavlink serial on pins rx:27 and tx:17  baud:38400
16:24:40.216 -> Default S.Port on rxpin:13 is IDLE_LOW, inverting rx polarity
16:24:40.216 -> Default S.Port baud:57600 b/s
16:24:40.216 -> FrSky simplex on tx pin = 14
16:24:40.737 -> WiFi mode set to STA sucessfully
16:24:40.737 -> Trying to connect to OmegaOffice....................
16:24:51.311 -> Failed to connect in STA mode
16:24:51.359 -> Failover to AP. Rebooting ....
16:24:52.446 -> ets Jun  8 2016 00:22:57
16:24:52.446 -> 
16:24:52.446 -> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
16:24:52.446 -> configsip: 0, SPIWP:0xee
16:24:52.446 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:24:52.446 -> mode:DIO, clock div:1
16:24:52.446 -> load:0x3fff0018,len:4
16:24:52.446 -> load:0x3fff001c,len:1044
16:24:52.446 -> load:0x40078000,len:10124
16:24:52.446 -> load:0x40080400,len:5856
16:24:52.446 -> entry 0x400806a8
16:24:55.504 -> 
16:24:55.504 -> Starting MavToPass version:2.64.09
16:24:55.504 -> Display support activated: Landscape 
16:24:55.504 -> 64x128  TEXT_SIZE=1  CHAR_W_PX=6  CHAR_H_PX=8  SCR_H_CH=8  SCR_W_CH=21
16:24:55.552 -> EEPROM initialised successfully
16:24:55.600 -> EEPROM settings read and adopted
16:24:55.600 -> Ground Mode selected
16:24:55.648 -> Battery_mAh_Source = 3 - Define battery capacities in the LUA script
16:24:55.648 -> RSSI Override for testing = 70%
16:24:55.696 -> Target Board is ESP32 / Variant is Heltec Wifi Kit 32
16:24:55.743 -> Mavlink Serial In
16:24:55.790 -> FrSky Serial Out
16:24:55.838 -> Mavlink WiFi+BT Out - WiFi mode is STA>AP
16:24:55.980 -> Protocol is UDP IP-Targeted
16:24:56.595 -> Bluetooth slave mode, host name for pairing is MavToPass
16:24:56.691 -> Mavlink serial on pins rx:27 and tx:17  baud:38400
16:24:56.739 -> Default S.Port on rxpin:13 is IDLE_LOW, inverting rx polarity
16:24:56.739 -> Default S.Port baud:57600 b/s
16:24:56.739 -> FrSky simplex on tx pin = 14
16:24:56.739 -> WiFi mode set to WIFI_AP 
16:24:56.739 -> AP IP address: 192.168.4.1  SSID: SR_2021
16:24:56.931 -> Begin UDP using AP UDP object  read port:14550  send port:14555
16:24:56.931 -> UDP for AP started, local 192.168.4.1   remote 192.168.4.255
16:24:57.027 -> Web support active on http://192.168.4.1
16:24:57.075 -> hb_count=1
16:24:57.837 -> hb_count=2
16:24:58.878 -> hb_count=3
16:24:58.878 -> Mavlink good!

zs6buj, 2021-04-28:
Hmm. All looks good. Just to be sure, you connected pin 14 of Heltec to the bottom pin in the back bay of the Taranis? And ground from DL to Heltec, and ground from Heltec to Taranis, second pin from bottom.

Can you beg, borrow or steal an oscilloscope? One of the the cheap usb types would likely be good enough.

jeanbatfr, 2021-04-29:
Yes I verified the ground with a tester.
I have an oscilloscope yes

zs6buj, 2021-04-29:
Probe pin 14 on the heltec, once it's up and running normally with satellite coordinates showing on the oled display. You should see puses rising up from zero to 3.3v, and since their speed is 57600 b/s, a pulse should be about 17mS long. At idle the pin should be low.

zs6buj, 2021-04-29:
#pulses

jeanbatfr, 2021-04-29:
I have a signal with a peak of 3.28V. The baudrate I set with the webbrowser is 38400 because My dragonlink is setted up like this.

zs6buj, 2021-04-29:
That's the problem then. The Taranis expects 57600.

jeanbatfr, 2021-04-29:
but what about the dragonlink? between the pixhawk and the receiver 38400 is mandatory

zs6buj, 2021-04-29:
The two ports on the esp32 can run at different speeds. The Taranis S.Port expects 57600 b/s.

I'm running out of ideas.

zs6buj, 2021-09-11:
I guess this issue has gone stale.

closed 2021-09-11 by zs6buj