zs6buj/mav2pt #71: Battery_maH_Source = 1 doesn't compile

Vabe7, 2021-12-09:
In line 1058 to 1061 in main.cpp the battery capacities are requested but the function Mavlink_Param_Request_Read requires two arguments and only one is given.

zs6buj, 2021-12-10:
Did you resolve this Vabe7? It's compiling here? Maybe Mavlink library version ?

Vabe7, 2021-12-10:
No...

Maybe because I'm working with the files in the PlatformIO folder, which would actually result in using the same Mavlink library, but that shouldn't matter anyways since the failing function is part of main.cpp...
It seems to me that the Source folder is further advanced than the PlatformIO folder. So maybe the issue lies in here.
But in the source folder the function is also called with only one variable (from Source/MavToPass/MavToPass.ino):
>#if (Battery_mAh_Source == 1)  
  // Request battery capacity params 
  if (mavGood) {
    if (!ap_bat_paramsReq) {
      Mavlink_Param_Request_Read(356);    // Request Bat1 capacity   do this twice in case of lost frame
      Mavlink_Param_Request_Read(356);    
      Mavlink_Param_Request_Read(364);    // Request Bat2 capacity
      Mavlink_Param_Request_Read(364);    
      Log.println("Battery capacities requested");
      LogScreenPrintln("Bat mAh from FC");    
      ap_bat_paramsReq = true;
    } else {
      if (ap_bat_paramsRead &&  (!parm_msg_shown)) {
        parm_msg_shown = true; 
        Log.println("Battery params successfully read"); 
        LogScreenPrintln("Bat params read ok"); 
      }
    } 
  }
  #endif

And the definition of Mavlink_Param_Request_Read() requires two parameters:
>void Mavlink_Param_Request_Read(int16_t param_index, char * param_id) {  // #20
  apo_sysid = Device_sysid;                    // From config.h MP is 255, QGC default is 0
  apo_compid = Device_compid;                  // 158 Generic autopilot peripheral component ID. MP is 190
  apo_targsys = 1;                             // APM FC
  apo_targcomp = 1;                            // APM FC
  mavlink_msg_param_request_read_pack(apo_sysid, apo_compid, &G2Fmsg,
                   apo_targsys, apo_targcomp, param_id, param_index);              
  Send_To_FC(20);             
 }

So I don't understand how it works on your side...

zs6buj, 2021-12-11:
Ok, I ported v2.67.09 to PIO and uploaded it.  It compiles clean here. Would you mind trying it?

Vabe7, 2021-12-11:
Strange... I just downloaded the new PIO, changed Battery_maH_Source to 1 and compiled it, but still:
>src\main.cpp: In function 'void loop()':
src\main.cpp:1058:37: error: too few arguments to function 'void Mavlink_Param_Request_Read(int16_t, char*)'
       Mavlink_Param_Request_Read(356);    // Request Bat1 capacity   do this twice in case of lost frame
                                     ^
src\main.cpp:177:6: note: declared here
 void Mavlink_Param_Request_Read(int16_t, char *);
      ^
src\main.cpp:1059:37: error: too few arguments to function 'void Mavlink_Param_Request_Read(int16_t, char*)'
       Mavlink_Param_Request_Read(356);    
                                     ^
src\main.cpp:177:6: note: declared here
 void Mavlink_Param_Request_Read(int16_t, char *);
      ^
src\main.cpp:1060:37: error: too few arguments to function 'void Mavlink_Param_Request_Read(int16_t, char*)'
       Mavlink_Param_Request_Read(364);    // Request Bat2 capacity
                                     ^
src\main.cpp:177:6: note: declared here
 void Mavlink_Param_Request_Read(int16_t, char *);
      ^
src\main.cpp:1061:37: error: too few arguments to function 'void Mavlink_Param_Request_Read(int16_t, char*)'
       Mavlink_Param_Request_Read(364);    
                                     ^
src\main.cpp:177:6: note: declared here
 void Mavlink_Param_Request_Read(int16_t, char *);
      ^
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1

zs6buj, 2021-12-12:
Ok, I see it now. Looks like a Mavlink library version number issue. I'll dig deeper tomorrow.

zs6buj, 2021-12-13:
Vabe7: I patched PIO V2.67.09 to include the param-id.  I never tested (Battery_mAh_Source == 1) since a mavlink lib update :(

I'm not in a position to test for a few days, travelling.

Vabe7, 2021-12-17:
I just tried it out, it compiles successful and runs. I can't try if it runs alright since on my test setup I don't have a current sensor configured but so far it looks good.

zs6buj, 2022-01-12:
Looks like I can close this one. Reopen if you need to.

closed 2022-01-12 by zs6buj