zs6buj/mav2pt #73: FrSky port type ESP32 heltec RX8R receiver

Mogway28, 2021-12-10:
No matter which #define FrSky_Port_Type 3  or #define FrSky_Port_Type 4  FPort2 is detected

zs6buj, 2021-12-11:
@Mogway28

I'm wondering if you have checked the saved settings through the Web interface. The settings saved into eeprom persist and prevail, until you refresh them with the config.h settings. There are a few ways to refresh them, including  making resetEepromPin (defined in config.h for your board variant) high (3.3v) for 10 seconds.  See [here](https://github.com/zs6buj/MavlinkToPassthru/wiki/Settings-in-NVM-EEPROM-and-config.h).

If the problem persists, let me know and I'll take a closer look.  If you could post the monitor log from bootup that would be useful.

Mogway28, 2021-12-12:
This issue is related with EEPROM. At start I've problem to enable reset pin (was 99 - set to pin5) but after compile and upload nothing changed in configuration (even with #define Reset_EEPROM  ). I've compile version 2.60.3 which I've  had used to this day and it's set converter correctly   from first upload - a can change config.h on this version and upload without problem. But anyway after upload 2.60.3 I've upload 2.67.06 with final changes (with enable eEprom pin ). then I can change config.h but always need reboot config by reset pin to change take effect. Thanks again for Your briliant converter.

zs6buj, 2021-12-13:
Thanks for your kind feedback :) I'm traveling during the next few days, but have my laptop and will look at this when I can. Did you hold Pin 5 high for at least 10 seconds?

Mogway28, 2021-12-13:
Yes 3,3V till "reboot" appears on OLED. In near future I will check how it goes on ESP8266 and let You know.

zs6buj, 2022-03-17:
I guess this issue is stale now

closed 2022-03-17 by zs6buj