zs6buj/mav2pt #77: No telemetry on Yaapu Horus X12S with X4R receiver in Relay mode

supachok654, 2022-03-17:
Hello 
Eric
I appreciated on your work.
I want my esp32 board receive telemetry from RFD900x that connected from FC on air side. Sending mavlink to MissionPlanner with TCP and sending to Horus X12S with S.port that connected to a receiver. It works well on my MP but my telemetry on yaapu telemetry script on Horus X12S still "no telemetry data".
I connected pin 13, pin 14 , 5V and GND with Yappu Telemetry cable to receiver and connected pin 25, pin26 and GND to RFD900x on ground side.

What did I miss ? 
Thank you

FYI:
Long Range Telemetry : RFD900x
Receiver : Frsky x4r
ESP32 variant : Wemos® LOLIN ESP32-WROOM-32_OLED_Dual_26p
Web support config:
[image]

Wiring:
[image]
[image]

Serial monitor log : 
> Starting MavToPass version:2.67.12
No display support selected or built-in
EEPROM initialised successfully
EEPROM settings version:2.67.12
EEPROM settings read and adopted
Relay Mode selected
Battery_mAh_Source = 1 - Get battery capacities from the FC
RSSI Override for testing = 70%
Target Board is ESP32 / Variant is Wemos® LOLIN ESP32-WROOM-32
Mavlink Serial In
FrSky Serial Out
Mavlink WiFi Out - WiFi mode is AP
Protocol is TCP/IP
No Bluetooth options selected, BT support not compiled in
Mavlink serial on pins rx:25 and tx:26  baud:57600
S.Port at 57.6kb/s selected on rxpin:13 is IDLE_LOW, inverting rx polarity
FrSky half-duplex on pins rx:13 and tx:14
For 1-wire applications a 2-wire to 1-wire converter or diode is required
WiFi mode set to WIFI_AP 
AP_default_IP:192.168.4.1  AP_gateway:192.168.4.1  AP_mask:255.255.255.0
AP IP address: 192.168.4.1  SSID: Crossfire_APLiLy
mDNS responder started
TCP/IP started, local IP:port 192.168.4.1:5760
Web support active on http://mavtopass.local
hb_count=1
hb_count=2
hb_count=3
Mavlink good!
Battery capacities requested
Battery params successfully read
FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus
Remote STA 1 connected to our AP 

> 

>

zs6buj, 2022-03-21:
Hi @Supachock654:

Sorry for the slow reply. It looks like you are inverting the S.Port signal from the ESP32 twice. Since you have selected S.Port, the firmware tries to "help", and inverts the S.Port signal  as per this line in the logs:

`S.Port at 57.6kb/s selected on rxpin:13 is IDLE_LOW, inverting rx polarity
`
So now you only need to add a diode to convert two-wire to one-wire.

But you are also using a commercial **Yappu Telemetry cable**, which in itself **inverts** and converts to one-wire.

The solution is to either:

1  Don't use the Yaapu telemeter cable, but try using a signal diode like this:  

[image]

  OR

2  Near line 344 in the FrSky_Ports.h tab, find the line

`        frInvert = true;` 

and change it to 

`        frInvert = false;`

rebuild and flash.

supachok654, 2022-03-21:
@zs6buj 
Thank you for your reply.
Can you recommend diode that is suitable for this work ?

zs6buj, 2022-03-21:
I used a 1N5822 Schottky diode, but almost any signal diode will work.

supachok654, 2022-03-22:
Hi @zs6buj 
Your solution is work fine then it stopped working and show "FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus".
I don't have any idea what does happen in my esp32 board.
What does "FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus" and "MavRingBuff full. Ignoring new records
" mean ?
Thank you.

zs6buj, 2022-03-22:
"FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus"

The Mavlink telemetry from the FC is buffered, then translated to FrSky, then the FrSky messages are buffered again in the FrPort scheduler buffer.  Certain messages are sent before others. We read the telemetry from the FrSky receiver, when we get a match on a certain key we pop a message off the scheduler buffer and inject it into the receiver stream.

So, in Relay Mode, messages are being pushed onto the scheduler buffer, and popped off that buffer when we can read from the FrSky receiver and find a match. If we can't read from the FrSky receiver reliably and fast enough, the scheduler buffer fills up and we get the message "FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus".

If it "worked fine" for a while, I would check the wiring of the yaapu converter. I must say I found those converters somewhat unreliable myself, and finally resorted to the simple signal diode solution.

supachok654, 2022-03-25:
It is working now ! Thank you very much Eric. :)

closed 2022-03-25 by supachok654

zs6buj, 2022-03-25:
A pleasure my friend.

ModaFpv, 2022-05-26:
> It is working now ! Thank you very much Eric. :)

HI ,
I got a same problem as yours . How did you fix it ? 
I get nothing on Sport but data on OLED monitor.