zs6buj/mav2pt #78: ESP8285 support

RM-FPV, 2022-03-29:
Thanks a lot for the great project!
Can be ESP8285 support also added? If it's already supported are there any changes in setup of FW needed?

zs6buj, 2022-03-30:
Hi RM-FPV,

I would need to get an ESP8285 board to test, but I guess it would be relatively easy to add it in as a variant.

RM-FPV, 2022-03-30:
Hi zs6buj!
This is a nice module, if you are looking for the one
https://aliexpress.com/item/32877575176.html

zs6buj, 2022-04-05:
Hi, that particular offering on ali wanted $48 for shipping to my home base, so I ordered [this one](https://www.banggood.com/DT-06-Wireless-WiFi-Serial-Transmissions-Module-TTL-to-WiFi-Compatible-HC-06-Bluetooth-External-Antenna-Version-Optional-p-1424145.html?cur_warehouse=CN&ID=566572&rmmds=search](url)) instead. I'm in another country until the end of April, so I'm hoping the board gets to me before I leave for home.

RM-FPV, 2022-04-23:
Hi, the module you found looks even better because of external antenna!

One more question please - when transferring mavlink to wifi, will it be also possible to connect notebook to this module to use some groundstation SW for displaying plane position on a map?

zs6buj, 2022-04-25:
Yes you can connect several wifi clients, like gcs or tracker or whatever else.

RM-FPV, 2022-04-25:
That's really great!
Thanks a lot!
Just waiting for 8285 version.

RM-FPV, 2022-06-11:
Hi zs6buj!
Have you received your modules? Would be nice to know any news about the FW.
Thanks.

zs6buj, 2022-06-12:
Hi RM-FPV, yes I did receive a module, but I unfortunately have very little free time right now. I'll earmark this issue and get back to it as soon as I can.

RM-FPV, 2022-06-12:
Got it! Thank you for reply!