zs6buj/mav2pt #79: No Sport out

ModaFpv, 2022-05-26:
Hi, 
I want to make a relay box which includes Crossfire, xsr recever.  ( between Crossfire and MAVTOPASS is using WIFI to communicate,  and wire pin 14 to XSR Sport)
I want to use  Wemos® LOLIN ESP32-WROOM-32_OLED board to do translate work .
But here i have issues on S.port output ,My X9d Plus doesnt receving anything from xsr 
However, I look at the OLED monitor on Esp32 DEV board, i can see bat voltage and GPS information...etc...
And i had check the settings below:
#define FrSky_Port_Type 4 ( or 3)  // not woking at all
#define Relay_Mode 

And My FC is using INAV software and using CSRF protocol

Now ,i don't know what to do :(
PS:
Sorry for my pool english :)
And wish know how to solve it !
Thanks

ModaFpv, 2022-05-26:
Here is LOG:

Starting MavToPass version:2.67.16
Display support activated: Landscape 
64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
EEPROM initialised successfully
EEPROM settings version:2.67.16
Reset_EEPROM defined. ALL SETTINGS IN EEPROM SET TO COMPILE_TIME DEFAULTS
EEPROM settings read and adopted
Relay Mode selected
Battery_mAh_Source = 1 - Get battery capacities from the FC
RSSI Automatic Select
Target Board is ESP32 / Variant is Wemos® LOLIN ESP32-WROOM-32
Mavlink WiFi In
FrSky Serial Out
WiFi mode is STA
Protocol is TCP/IP
No Bluetooth options selected, BT support not compiled in
FrSky Port: No telem on pin:13. Retrying 5 4 3 2 1 
Auto sensing abandoned. Defaulting to S.Port
Sensed serial port pin 13 is IDLE_LOW, inverting polarity
S.Port at 57.6kb/s selected on rxpin:13 is IDLE_LOW, inverting rx polarity
FrSky half-duplex on pins rx:13 and tx:14
For 1-wire applications a 2-wire to 1-wire converter or diode is required
WiFi mode set to STA sucessfully
Trying to connect to tbs_crossfire_B4E62D2FF1CF.....
WiFi connected!
mDNS responder started
Local IP address: 192.168.4.2  port: 5760
WiFi RSSI:-74 dBm
Local tcp client connected to remote server IP:192.168.4.1 remote Port:5760
TCP server started
Web support active on http://mavtopass.local
hb_count=1
hb_count=2
hb_count=3
Mavlink good!
Battery capacities requested
FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus

zs6buj, 2022-05-26:
Hello my friend,  your English is fine. Mav2PT only take the #define settings once, then keeps them in flash, only to be changed through the web interface. There is a #define to reset the eeprom settings.

ModaFpv, 2022-05-26:
> 

yeah!Thanks.
But how to fix "FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus" ?

ModaFpv, 2022-05-26:
> Hello my friend, your English is fine. Mav2PT only take the #define settings once, then keeps them in flash, only to be changed through the web interface. There is a #define to reset the eeprom settings.

yeah!Thanks.
But how to fix "FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus" ?

zs6buj, 2022-06-02:
```
FrSky Port: No telem on pin:13. Retrying 5 4 3 2 1
Auto sensing abandoned. Defaulting to S.Port
Sensed serial port pin 13 is IDLE_LOW, inverting polarity
S.Port at 57.6kb/s selected on rxpin:13 is IDLE_LOW, inverting rx polarity
FrSky half-duplex on pins rx:13 and tx:14
For 1-wire applications a 2-wire to 1-wire converter or diode is required
```
Hi again ModaFpv:

Try the following:

Select S.Port on Mav2PT. This will cause the FrSky output UART to invert and select 57600 b/s.

So now you need to convert 2 wire uart output to 1 wire. Use a signal diode like this:

[image]

See the white wire and diode that go to pins 13 and 14. Of course, your ESP32 board may be a little different.

ModaFpv, 2022-06-02:
Here is my settings:
[image]

Logs after wiring diode:
[image]

zs6buj, 2022-06-03:
Could you please post the log all the way from the boot. :)

ModaFpv, 2022-06-05:
> 

> Could you please post the log all the way from the boot. :)

Hi zs6buj,
Apology i didn't say more details.
Here is LOG From BOOT UP to end  &  MAVTOPASS settings & My setup wiring~

BOARD INFORMATION:  Wemos® LOLIN ESP32-WROOM-32_OLED_Dual_26p
RECEVER:  XSR/X4R
TRANSMITER: X9D PLUS 2019  + CROSSFIRE FULL MODULE

My basic idea :
I want to make a relay box  which runing Crossfire full module and xsr .  In this case i can send telmeterry back to X9D and use LUA  to see basic flight information and also send Sport signal to U360GTS antenna tracker to do tacker work.

-----------------------------------HERE LOG BEGIN---------------------------------------
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

Starting MavToPass version:2.67.16
Display support activated: Landscape 
64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
EEPROM initialised successfully
EEPROM settings version:2.67.16
Reset_EEPROM defined. ALL SETTINGS IN EEPROM SET TO COMPILE_TIME DEFAULTS
EEPROM settings read and adopted
Relay Mode selected
Battery_mAh_Source = 1 - Get battery capacities from the FC
RSSI Automatic Select
Target Board is ESP32 / Variant is Wemos® LOLIN ESP32-WROOM-32
Mavlink Bluetooth In
FrSky Serial Out
WiFi mode is AP
Protocol is TCP/IP
Bluetooth master mode host MavToPass is trying to connect to slave Crossfire 6059. This can take up to 30s
Bluetooth done
S.Port at 57.6kb/s selected on rxpin:13 is IDLE_LOW, inverting rx polarity
FrSky half-duplex on pins rx:13 and tx:14
For 1-wire applications a 2-wire to 1-wire converter or diode is required
WiFi mode set to WIFI_AP 
AP_default_IP:192.168.4.1  AP_gateway:192.168.4.1  AP_mask:255.255.255.0
AP IP address: 192.168.4.1  SSID: Crossfire_AP
mDNS responder started
TCP/IP started, local IP:port 192.168.4.1:5760
Web support active on http://mavtopass.local
hb_count=1
hb_count=2
hb_count=3
Mavlink good!
Battery capacities requested
FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus
Remote STA 1 connected to our AP

----------------------------------------LOG ENDS HERE-----------------------------------------

----------------------------------------MAVTOPASS SETTINGS---------------------------------
[image]
----------------------------------------END-----------------------------------------------------

----------------------------------------WIRING---------------------------------------------
[image]

zs6buj, 2022-06-06:
Thank you for the additional information. If you look carefully you will see that your S.Port wire goes to pin 14, and it should go to pin 13 as per my image above. Also, the cathode of your diode it attached to pin 14, it should be attached to the pin 13 side. I might also be better to use a  high-speed signal diode.

ModaFpv, 2022-06-06:
> Thank you for the additional information. If you look carefully you will see that your S.Port wire goes to pin 14, and it should go to pin 13 as per my image above. Also, the cathode of your diode [...]

 I have tried wire  Sport wire and the cathode of the diode to Pin 13 as you said ,then  the LOG will show up with" Fport READ OK "and then it will come up with" Fport read Time out".The  X9D doesn't receive any sensor information during this time.
Maybe those  diode doesn't work well in this way.  Do you have any high-speed signal diode for recommand?

zs6buj, 2022-06-06:
I use a high speed Schottky diode 1N5822, but almost any signal diode should work.

It's important that you post the log consistent with your settings and wiring. When you mention FPort being detected, it sounds like the Auto setting.

ModaFpv, 2022-06-07:
> I use a high speed Schottky diode 1N5822, but almost any signal diode should work.
> 
> It's important that you post the log consistent with your settings and wiring. When you mention FPort being detected, it sounds like the Auto setting.

Hi,
I change a new high-speed diode IN4148 on it .But it seems like not working 

-------------------------------LOG------------------------------------

Starting MavToPass version:2.67.16
Display support activated: Landscape 
64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
EEPROM initialised successfully
EEPROM settings version:2.67.16
EEPROM settings read and adopted
Relay Mode selected
Battery_mAh_Source = 1 - Get battery capacities from the FC
RSSI Automatic Select
Target Board is ESP32 / Variant is Wemos® LOLIN ESP32-WROOM-32
Mavlink WiFi In
FrSky Serial Out
WiFi mode is STA
Protocol is TCP/IP
No Bluetooth options selected, BT support not compiled in
S.Port at 57.6kb/s selected on rxpin:13 is IDLE_LOW, inverting rx polarity
FrSky half-duplex on pins rx:13 and tx:14
For 1-wire applications a 2-wire to 1-wire converter or diode is required
WiFi mode set to STA sucessfully
Trying to connect to tbs_crossfire_B4E62D2FF1CF......
WiFi connected!
mDNS responder started
Local IP address: 192.168.4.2  port: 5760
WiFi RSSI:-64 dBm
Local tcp client connected to remote server IP:192.168.4.1 remote Port:5760
TCP server started
Web support active on http://mavtopass.local
hb_count=1
hb_count=2
hb_count=3
Mavlink good!
Battery capacities requested
FrPort read good!
FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus
[image]
[image]

zs6buj, 2022-06-07:
Looks like some progress, as we got "FrPort read good!", which means we got a good read from the XSR.

But then this request was sent to the flight computer:

"Battery capacities requested"

but no answer received.  Normally you would receive this

"Battery params successfully read", but the lack of reply from the FC is not a fatal condition.

This message "FrPort scheduler buffer full. Check FrPort Downlink to Taranis/Horus" means that many messages have been lined up for the XSR, but the correct sensor slots have not been read, and the messages not popped off our buffer and sent.

Can you possibly compile and flash with these debug macros activated. This should show us the byte stream read from the XSR.

#define Debug_FrPort_Stream 
#define Debug_FrPort_Stream_Out

and publish the log again.

ModaFpv, 2022-06-07:
> Looks like some progress, as we got "FrPort read good!", which means we got a good read from the XSR.
>
> But then this request was sent to the flight computer:
>
> "Battery capacities requested"
>
> but no [...]

--------------------------------------LOG-----------------------------------------------
WiFi mode set to STA sucessfully
Trying to connect to tbs_crossfire_B4E62D2FF1CF.....
WiFi connected!
mDNS responder started
Local IP address: 192.168.4.2  port: 5760
WiFi RSSI:-60 dBm
Local tcp client connected to remote server IP:192.168.4.1 remote Port:5760
TCP server started
Web support active on http://mavtopass.local
hb_count=1
hb_count=2
hb_count=3
Mavlink good!
7E<AC<7E<0D<7E<8E<7E<2F<7E<D0<7E<71<7E<F2<7E<53<7E<34<7E<95<7E<16<7E<B7<7E<98<7E<39<7E<BA<7E<1B<10>01>50>02>00>00>00>9C>Battery capacities requested
FrPort read good!
7E<00<7E<A1<7E<22<7E<83<7E<E4<7E<45<7E<C6<7E<67<7E<48<7E<E9<7E<6A<7E<CB<7E<AC<7E<0D<7E<8E<7E<2F<7E<D0<7E<71<7E<F2<7E<53<7E<34<7E<95<7E<16<7E<B7<7E<98<7E<39<7E<BA<7E<1B<10>07>50>02>00>00>01>95>7E<00<7E<A1<7E<22<7E<83<7E<E4<7E<45<7E<C6<7E<67<7E<48<7E<E9<7E<6A<7E<CB<7E<AC<7E<0D<7E<8E<7E<2F<7E<D0<7E<71<7E<F2<7E<53<7E<34<7E<95<7E<16<7E<B7<7E<98<7E<39<7E<BA<7E<1B<10>06>50>5F>33>0E>00>F8>7E<00<7E<A1<7E<22<7E<83<7E<E4<7E<45<7E<C6<7E<67<7E<48<7E<E9<7E<6A<7E<CB<7E<AC<7E<0D<7E<8E<7E<2F<7E<D0<7E<71<7E<F2<7E<53<7E<34<7E<95<7E<16<7E<B7<7E<98<7E<39<7E<BA<7E<1B<10>05>50>00>00>06>0E>86>7E<00<7E<A1<7E<22<7E<83<7E<E4<7E<45<7E<C6<7E<67<7E<48<7E<E9<7E<6A<7E<CB<7E<AC<7E<0D<7E<8E<7E<2F<7E<D0<7E<71<7E<F2<7E<53<7E<34<7E<95<7E<16<7E<B7<7E<98<7E<39<7E<BA<7E<1B<10>01>F1>4A>00>00>00>B2>10<01<50<02<00<00<00<9C<7E<D0<7E<71<7E<D0<7E<F2<7E<D0<7E<53<7E<D0<7E<34<7E<95<7E<16<10<07<50<02<00<00<01<95<7E<16<10<06<50<5F<33<0E<00<F8<7E<B7<10<05<50<00<00<06<9E<59<7E<98<10<01<F1<4A<00<00<00<B2<7E<16<7E<98<7E<39<7E<16<7E<98<7E<BA<7E<16<7E<98<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<98<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<00<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<A1<7E<1B<10>06>50>57>3B>0E>00>F8>10<06<50<57<3B<0E<00<F8<7E<22<7E<1B<10>05>50>00>00>06>0E>86>10<05<50<00<00<06<0E<86<7E<83<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<E4<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<45<7E<1B<10>03>50>00>FE>01>00>9C>10<03<50<00<FE<01<00<9C<7E<C6<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<67<7E<1B<10>02>50>10>00>00>00>8D>10<02<50<10<00<00<00<8D<7E<48<7E<1B<10>04>50>00>00>00>76>25>10<04<50<00<00<00<76<25<7E<E9<7E<1B<10>06>50>51>43>0E>00>F6>10<06<50<51<43<0E<00<F6<7E<6A<7E<1B<10>01>F1>46>00>00>00>B6>10<01<F1<46<00<00<00<B6<7E<CB<7E<1B<10>05>50>00>00>06>0E>86>10<05<50<00<00<06<0E<86<7E<AC<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<0D<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<8E<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<2F<7E<1B<10>06>50>4D>43>0E>00>FA>10<06<50<4D<43<0E<00<FA<7E<D0<7E<1B<10>05>50>00>00>06>0E>86>10<05<50<00<00<06<0E<86<7E<71<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<F2<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<53<7E<1B<10>06>50>4A>4B>0E>00>F5>10<06<50<4A<4B<0E<00<F5<7E<34<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<95<7E<1B<10>05>50>00>00>06>0E>86>10<05<50<00<00<06<0E<86<7E<16<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<B7<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<98<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<39<7E<1B<10>06>50>48>4B>0E>00>F7>10<06<50<48<4B<0E<00<F7<7E<BA<7E<1B<10>01>F1>44>00>00>00>B8>10<01<F1<44<00<00<00<B8<7E<00<7E<1B<10>05>50>00>00>06>0E>86>10<05<50<00<00<06<0E<86<7E<A1<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<22<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<83<7E<1B<10>03>50>00>FE>01>00>9C>10<03<50<00<FE<01<00<9C<7E<E4<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<45<7E<1B<10>04>50>00>00>00>76>25>10<04<50<00<00<00<76<25<7E<C6<7E<1B<10>02>50>10>00>00>00>8D>10<02<50<10<00<00<00<8D<7E<67<7E<1B<10>06>50>46>4B>0E>00>F9>10<06<50<46<4B<0E<00<F9<7E<48<7E<1B<10>01>F1>43>00>00>00>B9>10<01<F1<43<00<00<00<B9<7E<E9<7E<1B<10>01>F1>43>00>00>00>B9>10<01<F1<43<00<00<00<B9<7E<6A<7E<1B<10>05>50>00>00>06>0E>86>10<05<50<00<00<06<0E<86<7E<CB<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<AC<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<0D<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<8E<7E<1B<10>06>50>45>4B>0E>00>FA>10<06<50<45<4B<0E<00<FA<7E<2F<7E<1B<10>05>50>00>00>06>0E>86>10<05<50<00<00<06<0E<86<7E<D0<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<71<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<F2<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<53<7E<1B<10>01>F1>49>00>00>00>B3>10<01<F1<49<00<00<00<B3<7E<34<7E<1B<10>06>50>44>4B>0E>00>FB>10<06<50<44<4B<0E<00<FB<7E<95<7E<1B<10>05>50>00>00>06>0E>86>10<05<50<00<00<06<0E<86<7E<16<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<B7<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<98<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<39<7E<1B<10>06>50>43>4B>0E>00>FC>10<06<50<43<4B<0E<00<FC<7E<BA<7E<1B<10>05>50>00>00>06>0E>86>10<05<50<00<00<06<0E<86<7E<00<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<A1<7E<1B<10>F2>50>00>00>54>01>57>10<F2<50<00<00<54<01<57<7E<22<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<83<7E<1B<10>03>50>00>FE>01>00>9C>10<03<50<00<FE<01<00<9C<7E<E4<7E<1B<10>04>50>00>00>00>76>25>10<04<50<00< [... cut ...] >00>9C>10<01<50<02<00<00<00<9C<7E<2F<7E<1B<10>03>50>A2>00>00>00>F9>10<03<50<A2<00<00<00<F9<7E<D0<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<00<00<01<95<7E<71<7E<1B<10>04>50>00>00>00>76>25>10<04<50<00<00<00<76<25<7E<F2<7E<1B<10>02>50>10>00>00>00>8D>10<02<50<10<00<00<00<8D<7E<53<7E<1B<10>06>50>40>4B>0E>00>00>10<06<50<40<4B<0E<00<00<7E<34<7E<1B<10>01>F1>46>00>00>00>B6>10<01<F1<46<00<00<00<B6<7E<95<7E<1B<10>01>50>02>00>00>00>9C>10<01<50<02<00<00<00<9C<7E<16<7E<1B<10>07>50>02>00>00>01>95>10<07<50<02<0

zs6buj, 2022-06-08:
Well, it looks like you are successfully sending at least some FrSky frames down to the transmitter.

This is the sequence of bytes being received for XSR:

7E<AC<7E<0D<7E<8E<7E<2F<7E<D0<7E<71<7E<F2<7E<53<7E<34<7E<95<7E<16<7E<B7<7E<98<7E<39<7E<BA<7E<1B

When we receive 0x7E1B, we immediately send 

10>01>50>02>00>00>00>9C>

which is messageID 0x5002 GPS Status and 

7E<1B<10>07>50>02>00>00>01>95>
7E<1B<10>06>50>3F>3B>0E>00>11>

and so on

Check that your transmitter has discovered these new sensors.

ModaFpv, 2022-06-08:
> I assume there is no latitude and longitude because your test is taking place indoors, so no satellite signal.
I think there should be a voltage or HDG sensor. But there nothing

ModaFpv, 2022-06-09:
> I assume there is no latitude and longitude because your test is taking place indoors, so no satellite signal.

Try many times,  still only 4 sensors shown on X9D :(