zs6buj/mav2pt #80: Can not get serial output

404-fpv, 2022-09-06:
I have 2x Heltec wifi kit 32 devices; 1 connected to my dragonlink tx. Mavlink in via serial and out via wifi with s.port translation connected to an FrSky rx.
The second wifi kit is connected to my MFD Crossbow mini and receives mavlink from the first wifi kit via wifi and outputs to the crossbow via serial.

No matter what I do, what settings I use or even swapping to a wifikit 8 on the crossbow, I cannot get any output at all, I have tried using s.port instead of serial with no success. If I swap over to a pair of HC05/06 bluetooth modules then all works as expected (proving the crossbow works).

zs6buj, 2022-09-08:
Hello and sorry for the slow response as I have been travelling.

I'm pretty sure you need to check the ports you are using for WiFi. Make sure the local and remote  ports match. For example,  esp32 (1) remote port number must be the same as esp(2) local port number, and visa versa.

404-fpv, 2022-09-17:
I swapped the ports for each node to match, which does work as if I have them wrong then I don't see any data on the screen of the remote node. 
With the correct p[orts I see data on the remote node screen, but simply no data output on serial of the remote node. It is weird, almost as if it is not mirroring/routing that data out over the serial pins.
---
Remote node:
uint16_t  TCP_localPort = 5760;     
uint16_t  TCP_remotePort = 5760;    
uint16_t  UDP_localPort = 14555;    // readPort - (default 14555) remote host (like MP and QGC) expects to send to this port
uint16_t  UDP_remotePort = 14550;   // sendPort - (default 14550) remote host reads on this port

Local node:
uint16_t  TCP_localPort = 5760;     
uint16_t  TCP_remotePort = 5760;    
uint16_t  UDP_localPort = 14550;    // readPort - (default 14555) remote host (like MP and QGC) expects to send to this port
uint16_t  UDP_remotePort = 14555;   // sendPort - (default 14550) remote host reads on this port

zs6buj, 2022-09-18:
Could you please post the second ESP32's log (serial monitor), and also it's config.h. Let's see if we can fix this.